arendst/Tasmota #6081: No more VCC state on V 6.6.0

JackNJack, 2019-07-12:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
_A clear and concise description of what the bug is._
Upon updating to v 6.6.0 I noticed that Vcc Status on State command is no longer available.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

Did I miss something on the change-log? As I read the change log, no mention on Vcc status being changed on the latest version 

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Device used (i.e. Sonoff Basic) : Basic_Dual__and POW__
- [ x] Tasmota binary firmware version number used :6.6.0 ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
22:33:40 MQT: stat/sonoff/pow05/STATUS = {"Status":{"Module":6,"FriendlyName":["Sonoff Pow05"],"Topic":"sonoff/pow05","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"test","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
22:33:40 MQT: stat/sonoff/pow05/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"GsonoffPOW","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:23","StartupUTC":"2019-07-12T14:22:17","Sleep":50,"CfgHolder":4617,"BootCount":129,"SaveCount":5111,"SaveAddress":"F9000"}}
22:33:40 MQT: stat/sonoff/pow05/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:33:40 MQT: stat/sonoff/pow05/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Balay24","camp_courageous"],"TelePeriod":30,"Resolution":"55C780C0","SetOption":["00008029","280500000100060000000000000000000000","00000000"]}}
22:33:40 MQT: stat/sonoff/pow05/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":12,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
22:33:40 MQT: stat/sonoff/pow05/STATUS5 = {"StatusNET":{"Hostname":"sonoff/pow05-2091","IPAddress":"192.168.10.204","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"2C:3A:E8:3B:48:2B","Webserver":2,"WifiConfig":4}}
22:33:40 MQT: stat/sonoff/pow05/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.200","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3B482B","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:33:40 MQT: stat/sonoff/pow05/STATUS7 = {"StatusTIM":{"UTC":"Fri Jul 12 14:33:40 2019","Local":"Fri Jul 12 22:33:40 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+08:00","Sunrise":"05:33","Sunset":"18:28"}}
22:33:40 MQT: stat/sonoff/pow05/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":12000}}
22:33:40 MQT: stat/sonoff/pow05/STATUS10 = {"StatusSNS":{"Time":"2019-07-12T22:33:40","ENERGY":{"TotalStartTime":"2019-07-07T16:28:22","Total":87.522,"Yesterday":9.885,"Today":9.106,"Power":678,"ApparentPower":678,"ReactivePower":0,"Factor":1.00,"Voltage":223.800,"Current":3.031}}}
22:33:41 MQT: stat/sonoff/pow05/STATUS11 = {"StatusSTS":{"Time":"2019-07-12T22:33:40","Uptime":"0T00:11:23","Heap":12,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Balay24","BSSId":"20:0D:B0:90:17:C0","Channel":1,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:03"}}}
22:33:41 MQT: tele/sonoff/pow05/STATE = {"Time":"2019-07-12T22:33:41","Uptime":"0T00:11:24","Heap":13,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":102,"POWER":"ON","Wifi":{"AP":1,"SSId":"Balay24","BSSId":"20:0D:B0:90:17:C0","Channel":1,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:03"}}
22:33:41 MQT: tele/sonoff/pow05/SENSOR = {"Time":"2019-07-12T22:33:41","ENERGY":{"TotalStartTime":"2019-07-07T16:28:22","Total":87.522,"Yesterday":9.885,"Today":9.106,"Period":6,"Power":649,"ApparentPower":719,"ReactivePower":309,"Factor":0.90,"Voltage":223.800,"Current":3.211}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-07-12:
Not a bug.

Now the ADC pin is used for Analog pin.

closed 2019-07-12 by ascillato2

JackNJack, 2019-07-12:
Thank you for the information!

ascillato2, 2019-07-12:
If you want to display VCC instead of the Analog Pin, please, just enable it in my_user_config.h