arendst/Tasmota #6082: IRRecv not working

betyar, 2019-07-12:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X ] Device used (i.e. Sonoff Basic) : _ESP-01____
- [X] Tasmota binary firmware version number used : _6.50___ / (pre-compiled)
- [X ] Development IDE - Compiler / Upload tools used : ____ / ESPTOOL____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```
**(Please, remember to close the issue when the problem has been addressed)**

Hi,

I tried to test my IR remote with Tasmota on an ESP-01 module configured as generic and with the IR receiver connected to GPIO3 with no pullups or pulldowns, just straight connections to VCC, GND, and the GPIO. I tried using a plain sensor (VS1838) first  [https://www.aliexpress.com/item/33002865269.html?spm=a2g0s.9042311.0.0.27424c4dGOyzbQ](https://www.aliexpress.com/item/33002865269.html?spm=a2g0s.9042311.0.0.27424c4dGOyzbQ) and then also tried a HX1838 RC module [https://www.aliexpress.com/item/32853903220.html?spm=a2g0s.9042311.0.0.27424c4dzxu60d](https://www.aliexpress.com/item/32853903220.html?spm=a2g0s.9042311.0.0.27424c4dzxu60d). In both cases when I checked the console there was nothing when a button on the remote was pressed. I tried different remotes and GPIOs but all were negative. The receivers do seem to work because the LED on the RC module flashes when a button is pressed and for the simple sensor I added a LED to the signal input and there also the LED would blink if a button was pressed. Setting up an IR on Tasmota should be quite straightforward but still I am at a loss to get it working. Anyone have any ideas?

ascillato2, 2019-07-12:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-12 by ascillato2

betyar, 2019-07-12:
I am not sure if this isn't a bug because not long ago LIRC had been changed so that previous methods no longer worked. I am not sure which libraries Tasmota uses for IRRecv or if they are in anyway related to LIRC or these changes, that is why I thought I would try here first.

Jason2866, 2019-07-13:
IRRecv does work fine with Tasmota. Your hardware setup is faulty.