arendst/Tasmota #6084: Committee error when using USE_MQTT_TLS

timstanley1985, 2019-07-13:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

Tasmota will not come with USE_MQTT_TLS as I get the following error

`/home/tim/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/sonoff/src/sonoff.ino.cpp.o:(.text._Z8MqttInitv+0xc): undefined reference to `BearSSL::WiFiClientSecure_light::WiFiClientSecure_light(int, int)'
/home/tim/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/sonoff/src/sonoff.ino.cpp.o:(.text._Z8MqttInitv+0x10): undefined reference to `BearSSL::WiFiClientSecure_light::setTrustAnchor(br_x509_trust_anchor const*)'
/home/tim/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/sonoff/src/sonoff.ino.cpp.o: in function `MqttInit()':
sonoff.ino.cpp:(.text._Z8MqttInitv+0x2a): undefined reference to `BearSSL::WiFiClientSecure_light::WiFiClientSecure_light(int, int)'
/home/tim/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sonoff.ino.cpp:(.text._Z8MqttInitv+0x3a): undefined reference to `BearSSL::WiFiClientSecure_light::setTrustAnchor(br_x509_trust_anchor const*)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/sonoff/firmware.elf] Error 1`

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Device used (i.e. Sonoff Basic) : ___sonoff basic__
- [x ] Tasmota binary firmware version number used : __latest github dev__ / (pre-compiled or self-compiled ?)
- [x ] Development IDE - Compiler / Upload tools used : __platformio__ / _n/a___
- [n/a ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

```
- [n/a ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
#define USE_MQTT_TLS

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Successful compilation

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

avrhack, 2019-07-13:
Did you also set the MQTT_LIBRARY_TYPE correctly? In my sonoff/my_user_config.h it shows....

`#define MQTT_LIBRARY_TYPE MQTT_ARDUINOMQTT`

timstanley1985, 2019-07-13:
Thanks.  MQTT_LIBRARY_TYPE wasn't defined in my my user config.  I have added but still get the following error:

``/home/tim/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/sonoff/src/sonoff.ino.cpp.o:(.text._Z8MqttInitv+0x8): undefined reference to `BearSSL::WiFiClientSecure_light::WiFiClientSecure_light(int, int)'
/home/tim/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/sonoff/src/sonoff.ino.cpp.o: in function `MqttInit()':
sonoff.ino.cpp:(.text._Z8MqttInitv+0x22): undefined reference to `BearSSL::WiFiClientSecure_light::WiFiClientSecure_light(int, int)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/sonoff/firmware.elf] Error 1``

Jason2866, 2019-07-13:
@avrhack  This is loooong time outdated and not needed anymore. 
There is just on mqtt library which supports all  -> PubSubClient

Jason2866, 2019-07-13:
Have you done all the needed steps as described?
https://github.com/arendst/Sonoff-Tasmota/wiki/AWS-IoT#step-1-prerequisites
You need to generate and define your keys. Without it will fail!

timstanley1985, 2019-07-13:
> Have you done all the needed steps as described?
> https://github.com/arendst/Sonoff-Tasmota/wiki/AWS-IoT#step-1-prerequisites
> You need to generate and define your keys. Without it will fail!

I am using a local mosquito server and authorising via fingerprint.  Do I still need to follow all those steps?  That is not made clear in the instructions.  What steps need to be followed if using a local mosquito server?

s-hadinger, 2019-07-13:
If you're using Mosquitto, instructions are here: https://github.com/arendst/Sonoff-Tasmota/wiki/TLS

You can save the private key generation part, you will not need it.

Though it doesn't explain your compilation error. It looks like `WiFiClientSecure_light.cpp` did not get compiled.

closed 2019-07-13 by arendst

timstanley1985, 2019-07-13:
@s-hadinger - I followed those instructions but it wouldn't build.  I added use mqtt TLS and added a fingerprint full of zeros.

@Jason2866 - this did not fix this issue.  I am building using core v2.5.2

Jason2866, 2019-07-13:
@arendst  I think you wanted to close #6073 not this one ;-)

docBliny, 2019-07-20:
@timstanley1985 I got the same error and build failure when I had `USE_MQTT_TLS` and `USE_MQTT_TLS_CA_CERT` in `user_config_override.h`, but it built fine when I only had those uncommented in `my_user_config.h` file instead.

e-t-h, 2019-08-23:
I have a similar issue: successfully compiled only if I uncommented both options (USE_MQTT_TLS and USE_MQTT_TLS_CA_CERT) in my_user_config.h  for the TLS version . Non TLS version was compiled successfully with undef in user_config_override, even with the standard also . I am using dev branch 6.6.0.6.

s-hadinger, 2019-08-23:
What do you want to achieve and what is the error message?

e-t-h, 2019-08-24:
I want to compile Tasmota with the TLS encryption using letsencrypt. This is already working with 6.6.0.1.
I changed to the current dev branch. I prepared the user_config_override and platform.io like in the previous project in ATOM. 
When I compile with both options USE_MQTT_TLS and USE_MQTT_TLS_CA_CERT) in user_config_override I get following error messages:

`Linking .pioenvs/sonoff-DE/firmware.elf
/Users/Ekki/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/sonoff-DE/src/sonoff.ino.cpp.o:(.text._
Z8MqttInitv+0xc): undefined reference to `BearSSL::WiFiClientSecure_light::WiFiClientSecure_light(int, int)'
/Users/Ekki/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/sonoff-DE/src/sonoff.ino.cpp.o:(.text._
Z8MqttInitv+0x10): undefined reference to `BearSSL::WiFiClientSecure_light::setTrustAnchor(br_x509_trust_anchor const*)'
/Users/Ekki/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/sonoff-DE/src/sonoff.ino.cpp.o: in func
tion `MqttInit()':
sonoff.ino.cpp:(.text._Z8MqttInitv+0x2a): undefined reference to `BearSSL::WiFiClientSecure_light::WiFiClientSecure_light(int, int)'
/Users/Ekki/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sonoff.ino.cpp:(.text._Z8MqttInitv+0x3a): undefi
ned reference to `BearSSL::WiFiClientSecure_light::setTrustAnchor(br_x509_trust_anchor const*)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/sonoff-DE/firmware.elf] Error 1
========================== [ERROR] Took 35.17 seconds ==========================
Environment sonoff-DE           [FAILED]
==================== 1 failed, 0 succeeded in 35.18 seconds ====================
 `

I just wanted to mention this, @timstanley1985 had the similar issue, found this solution, and this helped for me. 
Thanks.

arendst, 2019-08-24:
Is expected.

See note in user_config_override.h:

```
 ******************************************************************************************************
 * ATTENTION:
 *   - Changes to SECTION1 PARAMETER defines will only override flash settings if you change define CFG_HOLDER.
 *   - Expect compiler warnings when no ifdef/undef/endif sequence is used.
 *   - You still need to update my_user_config.h for major define USE_MQTT_TLS.
 *   - All parameters can be persistent changed online using commands via MQTT, WebConsole or Serial.
\*****************************************************************************************************/
```
TLS defining needs to be done in my_user_config.h as it interferes with other defines in user_config-override.

e-t-h, 2019-08-24:
Ups, okay. 

And thank you.

benzino77, 2019-09-03:
@arendst Could you please clarify whether only `USE_MQTT_TLS` has to be defined in `my_user_config.h` and the rest of MQTT TLS parameters/variants (`USE_MQTT_TLS_CA_CERT` or `USE_MQTT_TLS_FORCE_EC_CIPHER` or `USE_MQTT_AWS_IOT`) can be defined in `user_config_override.h`? 
Or maybe every MQTT TLS ralated parameter has to be defined in `my_user_config.h` file?

s-hadinger, 2019-09-03:
Things changed 3 days ago. Since PR #6330 you can now define `USE_MQTT_TLS` and other options in `user_config_override.h`

benzino77, 2019-09-04:
Thanks Stephan!

PR #6087 (2019-07-13): added "build_dir"
to make Core 2.3.0 compile with PlatformIO V.4.0.
Change has no impact to older versions of PlatformIO

## Description:

**Related issue (if applicable):** fixes #6084
## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on all cores
  - [ ] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md# [... cut]