arendst/Tasmota #6085: switching by main power frequency recognition

beenes, 2019-07-13:
Hi all,

There are several new mini consoled switches similar to Shelly 2.5
All types based on 8266 (LM1 or TYWE3S)

Like shelly the dual switches triggered using live power by the standard mechanical switch.
If turning on the switch, as a result the GPIO input state chances from 3V3 DC to 1.5**AC**(50 or 60HZ depending the main frequency).
Seems it can sense the main power frequency if the switch turned on.

I did try all options in the latest tasmota, but look like the current switch’s configuration doesn’t support the new mini switches.
Is it possible to add new configuration to Tasmota enabling triggering by frequency recognition?

The ESP also connected to one GPIO which receiving the constant reference 1.5AC. maybe the switch IO can be compared with the reference IO, and if equal -> switch is turned on.

Jason2866, 2019-07-13:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

thxthx0, 2019-07-13:
Summary of discord:

- they count 50Hz/60Hz pulses - needed hardware is already there

- you can do this with Counter and Rule, but in fact, it's not a satisfactory solution.

- same problem for that dimmer and even more:
https://github.com/arendst/Sonoff-Tasmota/issues/5737#issuecomment-507675241

I really hope there will be a solution,
maybe @chaosmaster @thirug010 @digiblur ... can help us, plzzzzz :)

- no similarity to the Shelly SW / S1 / S2... inputs.

- define a Counter at that input in Tasmota and see what is happening, 
when the switch is pressed. you'll see the pulses in TasmotaUI

- what we're talking about :smiley: ...
https://www.aliexpress.com/item/33012114855.html

https://www.aliexpress.com/item/33012242667.html

- "If someone with programming skills knowing Tasmota well, have a interest to support this device maybe a PR is made....
So dont await anything."

- I don't think it's only a device specific feature,
but a general and good solution to detect mains switch inputs.

e.g. additional Tasmota switch config (triggered counter / predefined number of pulses) 
would be great...

Jason2866, 2019-07-13:
Since Tasmota has a lot of tasks to do, counting AC frequency pulses is not possible.
Tasmota has already counter function. You can use it via rules. See wiki, commands and rules

thxthx0, 2019-07-13:
It's not the task of counting AC frequency, it is a task of pulse assignment.

"not possible" ... words that coders don't know :)

Jason2866, 2019-07-13:
Just do it and make a PR ;-)

ascillato, 2019-07-13:
Hi, there is no trigger for counters. Counters can be checked only with the teleperiod message.

If adding a new trigger for counters like:
```
On counter1#state do power1 toggle endon
```
will not work either for that usecase due it will be toggling every 50ms while the switch is being pressed. 

A proper solution for that should be a new gpio definition.

At this moment I don't have that device in order to implement that.

thirug010, 2019-07-13:
@thxthx0

Can you post the picture of the circuit if possible

beenes, 2019-07-13:
yes of course.
[image]
[image]

thxthx0, 2019-07-13:
@ascillato and @thirug010 BIG, BIG thanks for your help :) highly appreciated.

@thirug010
I only have the dimmer module (issues/5737), but I think the input is the same.

That's what I found out (also most other parts if needed)
[image]

thxthx0, 2019-07-14:
> If adding a new trigger for counters like:
> 
> ```
> On counter1#state do power1 toggle endon
> ```
> 
> will not work either for that usecase due it will be toggling every 50ms while the switch is being pressed.

@ascillato

I've tried @gemu2015's Script interpreter, where it is possible to set a trigger
```
input=COUNTER#C1 
if chg[input]>0 then ...
```
but in fact, as you mentioned, there is the problem of repeated toggling.

I guess the exact number of pulses isn't relevant / 50 or 60Hz input voltage doesn't matter,
and may work like this:

Are there pulses at the input?
yes -> push button/switch is pressed
no  -> button is not pressed

for the 
- WiFi Switch:
if there is a short period of pulses (e.g. < 1 second) -> then toggle the relay

- WiFi Dimmer:
 if there is a short period of pulses -> then toggle the light
(if light is on ->SerialSend '0x00' / if light is off ->SerialSend previous value )

  if there starts a longer period of pulses -> toggle dim direction and then 
 increase/decrease dim value as long as the button is pressed
 (and the limits aren't reached)...

gemu2015, 2019-07-14:
if you accept a one second delay the following script should work

>D
sw=0
cnt=0
lcnt=0

\>T
cnt=COUNTER#C1

\>S
if cnt!scnt
then
; counter has changed, switch is on
scnt=cnt
sw=1
else
; no change switch is off
sw=0
endif

; set power according to switch
if chg[sw]>0
then
=>power1 sw
endif

thxthx0, 2019-07-15:
Many thanks @gemu2015 for the script and of course especially for 
your great Tasmota Script editor :thumbsup: :thumbsup: :thumbsup:

Yes, I also think the max. 1sec delay wouldn't be a problem at all.

Your script seems to work for switch (not pushbutton),
but if the switch remains in one position, there is also the problem with repeated toggling.

It seems for that reason I'm not able to write a script for the dimmer with 'soft' dimming.

gemu2015, 2019-07-15:
i  edited the script above for toggling power
i don't think repeated toggling should be a problem.
dimming would better be done with 2 gpios

thxthx0, 2019-07-15:
BIG thanks again!

I think that has to be changed:
```
; set power according to switch
if chg[sw]>0
```
to
```
; set power according to switch
if sw>0
```

Yes, two GPIOs would be a solution (as other dimmers do),
but this WiFI Dimmer doesn't provide it an works flawless with orig. firmware.

for dimmer test (only light on/off) your script:
```
>D
sw=0
cnt=0
toggle=0

>T
cnt=COUNTER#C1

>S
if chg[cnt]>0
then
; counter has changed, switch is on
sw=1
else
; no change switch is off
sw=0
endif

; set power according to switch
if sw>0
then
;=>power1 %sw%
toggle^=1
endif

if chg[toggle]>0
then
;=>power1 %toggle%
=#light
endif

#light
if toggle==0
then
=>SerialSend5 FF550005DC0A
else
=>SerialSend5 FF55FF05DC0A
endif

gemu2015, 2019-07-16:
in the next release of scripter i introduce an F section which executes every 100ms
with this option i did a one button dimmer (via websend)
you may adapt this to your hardware

pr coming soon

; expand strings to hold websend
>D 25
sw=0
ws="websend [192.168.178.86]"
timer=0
hold=0
toggle=0

>B
; gpio 5 button input
spinm(5,0)

; fast section 100ms
>F
sw=pin[5]
; 100 ms timer
timer+=1

; 3 seconds long press
; below 0,5 short press
if sw==0
and timer>5
and timer<30
then
; short press
;=>print short press
toggle^=1
=>%ws% power1 %toggle%
endif

if sw>0
then
;pressed
if timer>30
then
; hold
hold=1
;=>print hold=%timer%
if toggle>0
then
=>%ws% dimmer +
else
=>%ws% dimmer - 
endif
endif
else
timer=0
hold=0
endif

gemu2015, 2019-07-16:
did a pr with the new features.
since teleperiod counter messages were to slow i also added access to the irq driven pulse counter
val=pc[x] which updates immediately.

thxthx0, 2019-07-16:
Perfect! Thanks so much @gemu2015

I'm looking forward to testing and will reply...

thxthx0, 2019-07-17:
WoW what should I say - it WORKS :thumbsup: Awesome!!!!!!

No more limits (well, obviously my scripting skills are, 
but I'm trying to get that Wifi Dimmer under control :smile: )

MANY, MANY thanks @gemu2015

@beenes the following script should do the job for your 2-Gang Wifi Switch
(similar to https://www.aliexpress.com/item/33050766518.html?spm=2114)
```
>D
sw1=0
sw2=0
cnt1=0
cnt2=0
timer1=0
timer2=0
toggle1=0
toggle2=0

>B
=>print "WiFi 2-Gang Switch Script"

>F
; Counter1/2 and Relay1/2 configured in template
cnt1=pc[1]
cnt2=pc[2]

if chg[cnt1]>0
then
; counter1 has changed, switch is on
sw1=1
else
; no change switch is off
sw1=0
endif

if chg[cnt2]>0
then
; counter2 has changed, switch is on
sw2=1
else
; no change switch is off
sw2=0
endif

; 100 ms timer
timer1+=1
timer2+=1

if sw1==0
and timer1>2
and timer1<30
then
;=>print short press1
toggle1^=1
=>Power1 %toggle1%
endif

if sw1==0
then
timer1=0
endif

if sw2==0
and timer2>2
and timer2<30
then
;=>print short press2
toggle2^=1
=>Power2 %toggle2%
endif

if sw2==0
then
timer2=0
endif
```

beenes, 2019-07-19:
Thanks guys,

Sorry @thxthx0  for my clueless question, but at the moment I have no idea how to use this script code.
Definitely would like to learn. Is there a tutorial for this?

meingraham, 2019-07-19:
@beenes https://github.com/arendst/Sonoff-Tasmota/blob/8ab60add17b1fe94340348ec4970d261c66092cd/scripter.md

dbrb2, 2019-07-19:
So perhaps dimming control would be something roughly like the below? (very quick and largely missing any detail...)

`;Long press....
if timer>30
then
	; held for > 3 seconds - start dimming
	hold=1
	
	;=>print held for=%timer%
	if toggle>0
	then
		<increment dimmer variable as hex string> 
		=>SerialSend5 FF55 <dimmer> 05DC0A
	else
		<decrement dimmer variable as hex string> 
		=>SerialSend5 FF55 <dimmer> 05DC0A
	endif
endif

;Short press...
if toggle==0
then
	=>SerialSend5 FF550005DC0A
else
	=>SerialSend5 FF55 <dimmer> 05DC0A
endif`

beenes, 2019-07-19:
@meingraham , @thxthx0  thanks for the script tutorial.
which files in the release version controlling the switch configuration?
the mention @thxthx0's script need to be added in a specific file before compilation?

meingraham, 2019-07-19:
You enter the script at runtime with the device online. The script is not "compiled in". This is the same as Tasmota rules.

Third line in the scripting documentation:
`In submenu Configuration => edit script`

Develop your script in a file and paste it into the console. Then, if you have to reset the device, you'll have the script backed up.

Mike

thxthx0, 2019-07-19:
and you have to compile recent Tasmota dev version with enabled Scripts @beenes
```
//#define USE_RULES
#define USE_SCRIPT
```  
[image]

thxthx0, 2019-07-19:
@dbrb2 
I've already managed to write a working Dimmer script :)
with almost all functions of the original firmware.

one thing still missing -> a dimmer slider

@gemu2015 You got any tips for me?? :)
is there anything like pc[x] ... to read Tasmota PWM "Dimmer" value or another solution...

gemu2015, 2019-07-20:
@thxthx0 
you have to define a fake pwm channel in tasmota on an unused pin
you will get a slider and an on/off button then

in section >E you will get the event for this power and dimmer values

dimmer=Dimmer
power=pwr[x]

according to a tutorial for scripter:
i have been told to be a bad teacher for beginners.
would be nice if someone with better skills in this field would write one
i would certainly help if needed

dbrb2, 2019-07-20:
Oh cool @thxthx0 would you be happy to share it?

Jason2866, 2019-07-20:
@thxthx0  It would be highly appreciated if you add your findings to this devices in a new template.
I know it is NOT a template. ALL devices are expained there. It is the place where user search...
Thx

thxthx0, 2019-07-20:
@gemu2015
Thanks man :thumbsup: I didn't get it,
maybe sometimes the fault of the teacher, mostly of the lazy student ;))

@dbrb2 
Of course - let me do a few more tests and maybe some changes, then I'll post it soon.

@Jason2866
Yes, I will try it (never done it before) once there is feedback that it works.

meingraham, 2019-07-20:
The device can be added as a "miscellaneous" type in the templates repository. Since the device is an ESP device with Tasmota loaded and has certain components assigned to GPIO (real and phantom), then a template is actually in order.

In addition, information on the frequency detection circuit (i.e., the schematic above), how to connect it, the script (with copious comments) can be part of the "flashing" guide of the template.

The detailed "flashing guide" is probably best added in a second step. First create a new template in the repository with "basic" information in the notes field. After the template is merged into the repository, you can edit it (there is an "Edit on GitHub" link on the template's page) and add the details using the full markdown syntax and create a pull request to have these details merged onto the template page.

Cheers!

Mike

beenes, 2019-07-21:
@meingraham is there any way to enable the script option?
at the moment i don't have the edit script option in my submenu.

meingraham, 2019-07-21:
@beenes https://github.com/arendst/Sonoff-Tasmota/issues/6085#issuecomment-513204013

thxthx0, 2019-07-21:
or @beenes https://github.com/arendst/Sonoff-Tasmota/issues/6085#issuecomment-513369748 ;)

@meingraham 
thanks, yes I also think there is a lot of information here and under #5737
so two steps is a good idea.
I think it also would fit under "Switches and Dimmers" in the templates repository.

thxthx0, 2019-07-22:
Hi, **Script** and config to test:
**[WiFi-Dimmer-Script-v0.1.txt](https://github.com/arendst/Sonoff-Tasmota/files/3418862/WiFi-Dimmer-Script-v0.1.txt)**

for **Wifi Smart Dimmer Module 220V-240V 150W**
e.g. https://www.aliexpress.com/item/33010332202.html?spm=2114
[image]
(Wi-Fi Dimmer Smart Life/Tuya APP MoesHouse and many other sellers)

recent Tasmota dev version with **enabled Scripts** is needed:
https://github.com/arendst/Sonoff-Tasmota/tree/development
`#undef  USE_RULES`
`#define USE_SCRIPT`

Generic config:
`GPIO1  -> Serial Tx (148)`
`GPIO3  -> Serial Rx (149)`
`GPIO13 -> Counter1 (42)`
`GPIO14 -> PWM1 (37)` (or any unused GPIO)

or Template:
`{"NAME":"WiFi Dimmer","GPIO":[0,148,0,149,0,0,0,0,0,42,37,0,0],"FLAG":0,"BASE":18}`

meingraham, 2019-07-22:
@thxthx0 

When you create the entry in the templates repository, please call the device `MoesHouse MS-105 Wi-Fi Dimmer`

Jason2866, 2019-07-23:
Does this dimmer has Zero Cross Detection? Or is it a classic Triac dimmer?

thxthx0, 2019-07-23:
@Jason2866
It's a Trailing-Edge Phase Dimmer and yes, has 
Zero Crossing Detection done by STM8S003F3 MCU
[image]

Although there is the misleading module name 
QS-WIFI-D01-TRIAC in the instruction manual, 
it is definitely Power MOSFET driven (SVF12N65F) 
for dimmable LED light bulbs / dimmable power supplies...

In my opinion the quality is high and there are many 
sellers who already offer it e.g. at Ali, Amazon...
registered shipping from China only took about 10 days.

So maybe worth a try before Shelly presents a clone 😄

chaosmaster, 2019-07-23:
@thirug010 Interesting, it seems similar in hardware design as the [Tuya Touch Dimmer](https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer#touch-eu-and-us---multiple-manufacturers) which also uses an STM8S to handle zero-crossing detection but uses the tuya-protocol to communicate with the ESP.
I have actually written a replacement-firmware for the MCU some time ago:
https://github.com/chaosmaster/tuya_mcu/tree/master/stm8_tuya

thxthx0, 2019-07-23:
@chaosmaster
Would be GREAT, if you could write a replacement FW here, too.
I'm not familiar with STM8 MCUs, but the orig. FW seems to be protected.

The PWM output for dimming is PC3 (Pin 13) @ STM8S003F3 MCU
and the serial protocol is very simple, only one command with changing dim values.

chaosmaster, 2019-07-23:
@thxthx0 The Tuya touch uses [PC7](https://github.com/chaosmaster/tuya_mcu/blob/master/stm8_tuya/src/gpio.h#L11) for the PWM.
I never really proceeded with the project, as i was hitting size constraints with using SDCC.
I reverted back to using the original firmware that was on the MCU, but it does mostly work.
I'm sure it could be easily modified to work with your dimmer.

meingraham, 2019-07-28:
@thxthx0 

Did you create an entry in the template repository? If so, did you include the frequency detection circuit and an explanation of what it does, why it's needed, and what you're doing with it?

I have created a new article for the Scripter and a Cookbook to go along with that. I pulled several examples that @gemu2015 and users have posted.
https://github.com/arendst/Sonoff-Tasmota/wiki/Tasmota-Scripting-Language
https://github.com/arendst/Sonoff-Tasmota/wiki/Script-Cookbook

Can you provide an abstract of what your script does; why it's required? I can add your script to the Cookbook but I'd like put an explanation with it.

Thanks!

Mike

thxthx0, 2019-07-29:
@meingraham

I took the first step last week, but no detailed explanations yet: 
https://blakadder.github.io/templates/qs-wifi_D01_dimmer.html

Spent some time finding a suitable name, 
maybe Moes or MoesHouse in front would be a better choice, as you suggested.

Yes, I'll provide more explanations for the template repository and
the Cookbook (not only the short comments in the script due max script size),
but there still seem to be some issues with Alexa, HA, Domoticz...

dbrb2, 2019-07-29:
Are there any binaries available with scripting already enabled (I realise that sounds very lazy!)

thxthx0, 2019-07-29:
I don't think so, you may use Gitpod
https://gitpod.io/#https://github.com/benzino77/tasmocompiler

with "Custom parameters"
`#undef USE_RULES`
`#define USE_SCRIPT`

meingraham, 2019-07-29:
There are no plans to provide pre-compiled binaries with Scripting enabled. As suggested, Gitpod makes it super easy to get binaries for "special" use cases. You can still OTA, but you have to do it via file upload instead of pointing the web OTA to `thehackbox`. Still super easy. The only thing to be aware of with file upload is that you will have to perform the intermediate `minimal` upload yourself in order to make room for the final binary if the binary is larger than 502KB.

We have added the scripting feature to the wiki so it's easier to find information. It's in its early stages so much to be enhanced in terms of details... such as how to compile, upload, etc. But now that it's in the wiki, the user community (i.e., you guys) can start contributing your knowledge and experiences.

Cheers!

Mike

ondoteam, 2019-07-29:
Hello! I will continue https://github.com/arendst/Sonoff-Tasmota/issues/5737#issuecomment-515796060 here. I found the bug in the script and fixed it. "Insert code" is not working very good with scripts so I will paste plain.

**Change:**

>E
slider=Dimmer
; slider change
if chg[slider]>0
then
; dim according slider
dimval=slider
=#senddim(dimval)
endif

**To:**

>E
slider=Dimmer
; slider change
if chg[slider]>0
then
; dim according slider
if slider>0
then
dimval=slider
=#senddim(dimval)
else
powert=0
=#senddim(0)
endif
endif

thxthx0, 2019-07-29:
THX @ondoteam
txt updated above

xbmcnut, 2019-07-30:
I compiled 6.6.0.3 and enabled the above script after flashing by soldering all wires to the module (with RE and GPIO0 at GND). Working a treat thank you except the switch (momentary) connected to 'S' is not working. Is there a trick to that?

ondoteam, 2019-07-30:
@xbmcnut hello! Did you restart the device after save the script? If not, just try. By the way double check the script, because sometimes the devices just restart when you tap "Save" button due to an exception and do not save anything.

About the slider: it is working for me, but I am using HA not Tasmota. By the way I tested it yesterday using the Tasmota WebUI and it works... the issue is from OFF to ON using the slider?

Sorry about my english :see_no_evil:

xbmcnut, 2019-07-30:
@ondoteam Yes, hard reset after flashing. I've gone back into script editor and the V2 text is still there so assuming it's stuck. Switch was working on stock firmware but as I'm new to Platformio, I'll double check my settings there but if scripts are available, I assume I got it right?

ondoteam, 2019-07-30:
@xbmcnut I mean restart (red button in Tasmota WebUI) after press "Save" button in "Configuration > Edit script".

xbmcnut, 2019-07-31:
@ondoteam Yep, did that too. Script has stuck, see below.
[image]

This is what I did in `my_user_config.h`

```
// Select none or only one of the below defines
#undef USE_RULES                                // Add support for rules (+8k code)
#define USE_SCRIPT                               // Add support for script (+17k code)
#define USE_SCRIPT_FATFS 4 
``` 

Button/switch still not working.

thxthx0, 2019-08-01:
@meingraham 

Template Repo has been updated now:
**[Moes QS-WiFi-D01 Dimmer 150W](https://blakadder.github.io/templates/qs-wifi_D01_dimmer.html)**

Please feel free to make corrections, 
and if there are too few details, I will add them.

Some format errors I'll fix with the next changes.

meingraham, 2019-08-01:
@thxthx0 

Great entry!

I did some re-arranging/re-formatting. You'll have to wait for my PR to be merged before reviewing. Let me know if I screwed anything up... and take corrective action to edit as needed :wink: 

Mike

thxthx0, 2019-08-02:
Thanks a lot! :thumbsup: @meingraham

meingraham, 2019-08-02:
@beenes 

Has this community effort provided the solution you needed? If so, please close this issue. If not, what symptoms are you encountering?

Mike

thxthx0, 2019-08-04:
Before this thread gets closed :),
there's still an issue that dimming does not work with Alexa and the script.

my test config:
Tasmota  6.6.0.3 with Hue Emulation
Echo Dot 2nd gen

everything works fine
- on/off and dimming via slider, button, MQTT, http
- on/off with Alexa

but not dimming with Alexa (e.g. "Alexa, light to 80")
- although the "Dimmer" value and the slider are set accordingly,
console message:
`MQT: stat/light/RESULT = {"POWER":"ON","Dimmer":80}` 
- the script seems not to be triggered 

maybe @gemu2015 would you be so kind to look at the script above
https://github.com/arendst/Sonoff-Tasmota/files/3444155/WiFi-Dimmer-Script-v0.2.txt
and what could be the error in >E ❔

meingraham, 2019-08-04:
@thxthx0 

Try adding some debug print statements to see if the script starts at all?

=\> print prints to info log for debugging

and set logging levels for more information - loglvl = loglevel of script cmds

gemu2015, 2019-08-04:
@thxthx0 

MQTT RESULT calls the E section, you should get an update there.

try the following code in E to debug 

slider=Dimmer
power=POWER

if upd[slider]>0
then
=>print slider updated %slider%
endif

if upd[power]>0
then
=>print power updated %power%
endif

gemu2015, 2019-08-04:
@thxthx0 

in case dimmer doesn't call E insert the following code in file xdrv_20_hue.ino

identify the following lines
LightPreparePower();
if (LST_COLDWARM <= light_subtype) {
   MqttPublishPrefixTopic_P(RESULT_OR_STAT, PSTR(D_CMND_COLOR));
 } else {
    MqttPublishPrefixTopic_P(RESULT_OR_STAT, PSTR(D_CMND_DIMMER));
  }

and insert this code =>XdrvRulesProcess(); in the next line

[62 more replies]

closed 2019-10-05 by ascillato2