arendst/Tasmota #6086: Firmware 6.6.0 AND Amazon Echo Dot Gen. 2 AND WeMo Emulation = no new devices found

M4APger, 2019-07-13:
Release 6.5.0 works fine

Release 6.6.0:
- Emulation: Belkin WeMo 
- 2x Amazon Echo Dot Gen. 2
no new device found - (old devices works fine)

test environment:
AVM FritzBox 7590 - 7.11
2x Amazon Echo Dot Gen. 2
3x Sonoff S20
1x Sonoff Touch
1x Sonoff Pow
1x Sonoff TH16
... and more

no working:
- latest release 6.6.0 (core 2.3.0 / 2.4.2 / 2.5.2)
- development release 6.6.0.1 (core 2.3.0 / 2.4.2 / 2.5.2) Build 2019-07-12T10:10:06

flashing back up 6.5.0 works fine !!!

Sorry ... my english is not the yellow of the egg!

More infos in German:
https://forum.creationx.de/forum/index.php?thread/2013-tasmota-release-6-6-0/&postID=25076#post25076

----------------------------------------------------------------------------------------------------------

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : _____
- [ ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ihavenonick, 2019-07-13:
same here, with Dot 3 no Problem, but with Dot 2 no new detection after removing the device

Jason2866, 2019-07-13:
WeMo Emu is known not working reliable since a while. Use HueEmulation instead.
HueEmulation is better anyway because there is feedbaxk back to sender that command is done.

ihavenonick, 2019-07-13:
No good news, because i don't like the slider and color bar in Alexa if i use only a simple switch and the problems with HUE EMU in the past was not funy to.
Also i had never a Problem with WeMo EMU without a feedback to sender.

I think i will use Node-Red WeMo for connection to Alexa.

M4APger, 2019-07-13:
That's really not good news!
the fact is:
WeMo is for switches
Hue is for lights

s-hadinger, 2019-07-13:
Ha ha! I finally understand why we had mixed reports about WeMo emulation. It has always worked perfectly for me but I have multiple Echo device at home.

With a single Echo Dot 2nd gen, I do confirm WeMo emulation does not work (6.6.0). Log below:

```
13:02:45 HTP: Console
13:02:48 UDP: Packet (122)
13:02:48 UDP: Packet (122)
13:02:48 UDP: Packet (107)
13:02:49 UDP: Packet (107)
13:02:50 UPP: WeMo Type 1, Response sent to 192.168.1.25:50000
13:02:51 WIF: Checking connection…
```

Now as soon as I switch on my Echo Spot, it suddenly works. Log:

```
13:05:51 WIF: Connected
13:05:59 UDP: Packet (94)
13:05:59 UDP: Packet (101)
13:05:59 UDP: Packet (94)
13:05:59 UDP: Packet (101)
13:05:59 UDP: Packet (94)
13:05:59 UDP: Packet (101)
13:05:59 UDP: Packet (107)
13:05:59 UDP: Packet (107)
13:05:59 UDP: Packet (122)
13:05:59 UDP: Packet (122)
13:06:00 UPP: WeMo Type 2, Response sent to 192.168.1.43:50000
13:06:01 HTP: WeMo setup
13:06:01 HTP: WeMo basic event
13:06:01 HTP: WeMo basic event
13:06:03 HTP: WeMo basic event
```

I'll try to dig about the difference between WeMo Type 1 and WeMo Type 2. What's still surprising is that I'm not aware of any significant change in WeMo code between 6.5.0 and 6.6.0.

Time for debug! Thanks @M4APger for reporting this bug.

s-hadinger, 2019-07-13:
I found a difference between 6.5.0 and 6.6.0:

6.6.0:
```
const char URN_BELKIN_DEVICE[] PROGMEM = "urn:belkin:device:**";
```

6.5.0:
```
response.replace("{r3", F("urn:Belkin:device:**"));
```

So it looks like a capitalization error and should be: 
```
const char URN_BELKIN_DEVICE[] PROGMEM = "urn:Belkin:device:**";
```

I don't have time to test now...

arendst, 2019-07-13:
I will push an update with this fix now.

s-hadinger, 2019-07-13:
We need to test it before. This string is used in two locations. So we may need two different strings, one with small b one with Capital B. 

And it's maybe not the problem at all.

M4APger, 2019-07-13:
My test socket is ready. Is the bug fix in the latest development version?
20190712 14:02 GMT + 2
I could test these then!

arendst, 2019-07-13:
Nope. Wait another hour for it to be available.

s-hadinger, 2019-07-13:
Confirmed to work, but we do need 2 strings.
I'm pushing the PR right now.

s-hadinger, 2019-07-13:
@M4APger Fix is here if you want to test before it is merged into development: 02ee16c3dcec012bd374d3b8faea60d50a9de675

ascillato2, 2019-07-13:
Closing this issue as it is solved.

Thanks everyone for working on this :+1:

closed 2019-07-13 by ascillato2

s-hadinger, 2019-07-13:
There one glitch you need to know, if the WeMo device was discovered by a newer echo device, it cannot be controlled via an echo 2G. You need to remove the device and redo a discovery.

But if all you have are echo dot 2G, then there's no issue.

M4APger, 2019-07-13:
Program Version | 6.6.0.1(7e20579-classic)
Build Date & Time | 2019-07-13T19:04:10
Core/SDK Version | 2_3_0/1.5.3(aec24ac9)

Test Ok ... 
Development BIN 20190713 19:04 GMT+2

thx

oywino, 2019-09-18:
Hi all, 
I have flashed my Sonoff Basic successfully with Tasmota 6.6.0 and got it up'n'running on my WLAN.
Then I changed its configuration to; Disable MQTT and enable WeMo emaulation, hpoing that I could see the switch in my Home Assistant 0.98.2
The integration shows up ok, but no device is added/visible.
Then I changed to Hue Bridge emulation, but that resulted in nothing at all.
I both cases, I manually added the host IP to _configuration.yaml_
What am I missing here?

Jason2866, 2019-09-18:
@oywino  HueEmulation in Tasmota is designed for the use with Alexa.
It CAN work for other systems.

oywino, 2019-09-18:
@Jason2866  This is the error message I get from HA ;

`Log Details (ERROR)
Wed Sep 18 2019 08:46:11 GMT+0200 (Central European Summer Time)
Unknown error connecting with Hue bridge at 192.168.0.21
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hue/config_flow.py", line 207, in async_step_import
    bridge = await get_bridge(self.hass, host, username)
  File "/usr/src/homeassistant/homeassistant/components/hue/bridge.py", line 171, in get_bridge
    await bridge.initialize()
  File "/usr/local/lib/python3.7/site-packages/aiohue/bridge.py", line 45, in initialize
    self.config = Config(result['config'], self.request)
TypeError: list indices must be integers or slices, not str`

Jason2866, 2019-09-18:
As i said before, no support for other devices'/ software. 
HueEmulation is ONLY for Alexa.

s-hadinger, 2019-09-18:
This is an HA issue. Please direct it to the HA project. If solving it requires simple changes in Tasmota we may do it. But diagnosis must be done by HA first.

oywino, 2019-09-18:
@Jason2866 & @s-hadinger  OK guys - thanks for taking the time to reply.
One last question: Where can I find a user guide that will explain in detail all the various settings, configs and template options in the Tasmota Web front end?

meingraham, 2019-09-18:
@oywino Click the `Wiki` tab at the top of the page.

Makk1983, 2019-10-13:
Hi, I'm starting to experiment with Alexa now and I wanted to integrate my sonoff tasmota 6.6.0, I enabled the Belkin WeMo emulation, I tried to add a new device to alexa but found nothing, I tried to change emulation again in Hue Bridge but nothing, then reading on this forum I installed version 6.5.0 but nothing does not work, may be due to the fact that I have no device (DOT, Echo etc.) but only the app on the smartphone?

s-hadinger, 2019-10-13:
Alexa Hue integration requires a physical Alexa device. The app alone is not enough.

PR #6088 (2019-07-13): Fix WeMo emulation for 1G echo and 2G echo dot (#6086)
## Description:

**Related issue (if applicable):** fixes #6086
## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2 and 2.5.2
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).