arendst/Tasmota #6089: Adding support for PCA9555 IO-expander?

zaprodk, 2019-07-13:
Hi.

I have some ready-made boards with 8 relays and a single PCA9555 that controls them.
The 8 relays are wired on outputs 9-15 on the chip. No input is needed, no need for interrupts.

I looked into the MCP23017 support library but with no amount of modifying i could get it working with the PCA9555.

So can anyone, smarter than me try adding experimental support for the PCA9555?
Regards,
Per.

ascillato2, 2019-07-13:
Hi,

By now it is not supported. Added a label to see if someone is interested on working on that.

ascillato2, 2019-07-29:
Adding label Feature request (hold over) for future reference if someone wants to develop a driver for this device.

closed 2019-07-29 by ascillato2