arendst/Tasmota #6090: Sonoff POW - Connection Problems

riotm0de, 2019-07-13:
Hi Guys,

i already checked the [Troubleshootings](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk) and erased my flash. But i still have connection issues as long as i use it via 230v.

When the sonoff is powered by COM Port its working like it should. But when using it with 230V, i cant reach the IP and cant do anything with it. Just the Button on the device is controlling the relais. Sometime i can reach the device via IP and after 2x on/off its gone and not responsive any longer.... 

Please help - Thank you!

Jason2866, 2019-07-13:
Your device has a hardware issue with the power supply.
It is not Tasmota related.

riotm0de, 2019-07-13:
> Your device has a hardware issue with the power supply.
> It is not Tasmota related.

so the device is broken ?   i mean just L/N cables + EU Plug on it for testing...

ascillato2, 2019-07-13:
Closing this issue as it is a hardware problem. You should use the warranty and ask for another device. Sorry. 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-13 by ascillato2