arendst/Tasmota #6091: Xiaomi Philips brightness changes color since 6.6.0

chrisnharvey, 2019-07-13:
## BUG DESCRIPTION

When a Xiaomi Philips module's color is set to "cool white", the brightness slider acts as a color temperature slider, increasing the color temperature value instead of the brightness value. Looking at the output of the console, it would appear that Tasmota is aware of the changing color.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Xiaomi Philips E14
- [x] Tasmota binary firmware version number used : 6.6.0 / (pre-compiled)
- [x] Development IDE - Compiler / Upload tools used : OTA update from 6.5.0 binary
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

22:04:51 CMD: Status 0
22:04:51 MQT: stat/living-room-lamp/STATUS = {"Status":{"Module":48,"FriendlyName":["Living Room Lamp"],"Topic":"living-room-lamp","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:04:51 MQT: stat/living-room-lamp/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:13:57","StartupUTC":"2019-07-13T20:50:54","Sleep":50,"CfgHolder":4617,"BootCount":15,"SaveCount":345,"SaveAddress":"F8000"}}
22:04:51 MQT: stat/living-room-lamp/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-basic)","BuildDateTime":"2019-07-06T13:11:28","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:04:51 MQT: stat/living-room-lamp/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Skynet 2.4GHz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
22:04:51 MQT: stat/living-room-lamp/STATUS4 = {"StatusMEM":{"ProgramSize":422,"Free":580,"Heap":18,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","0F082382","001083A0","20021706","010001C0"]}}
22:04:51 MQT: stat/living-room-lamp/STATUS5 = {"StatusNET":{"Hostname":"living-room-lamp-0928","IPAddress":"192.168.10.169","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.10","Mac":"78:11:DC:54:E3:A0","Webserver":2,"WifiConfig":4}}
22:04:51 MQT: stat/living-room-lamp/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E654E3A0","MqttUser":"homeassistant","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:04:51 MQT: stat/living-room-lamp/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 13 21:04:51 2019","Local":"Sat Jul 13 22:04:51 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:00","Sunset":"20:51"}}
22:04:51 MQT: stat/living-room-lamp/STATUS10 = {"StatusSNS":{"Time":"2019-07-13T22:04:51"}}
22:04:51 MQT: stat/living-room-lamp/STATUS11 = {"StatusSTS":{"Time":"2019-07-13T22:04:51","Uptime":"0T00:13:57","Vcc":2.100,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":18,"Color":"2E00","HSBColor":"0,0,0","Channel":[18,0],"CT":153,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Skynet 2.4GHz","BSSId":"BC:EE:7B:30:0D:B2","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```

22:06:27 CMD: Dimmer 5
22:06:27 SRC: WebConsole from 192.168.10.101
22:06:27 RSL: Received Topic /Dimmer, Data Size 1, Data 5
22:06:27 RSL: Group 0, Index 1, Command DIMMER, Data 5
22:06:27 MQT: stat/living-room-light-3/RESULT = {"POWER":"ON","Dimmer":5,"Color":"13,0","HSBColor":"0,0,0","Channel":[5,0],"CT":153}
22:06:27 CFG: Saved to flash at F9, Count 1296, Bytes 3584
22:06:29 CMD: Dimmer 10
22:06:29 SRC: WebConsole from 192.168.10.101
22:06:29 RSL: Received Topic /Dimmer, Data Size 2, Data 10
22:06:29 RSL: Group 0, Index 1, Command DIMMER, Data 10
22:06:29 MQT: stat/living-room-light-3/RESULT = {"POWER":"ON","Dimmer":10,"Color":"26,0","HSBColor":"0,0,0","Channel":[10,0],"CT":153}
22:06:29 CFG: Saved to flash at F8, Count 1297, Bytes 3584
22:06:30 WIF: Checking connection...
22:06:30 WIF: Connected
22:06:35 CMD: Dimmer 50
22:06:35 SRC: WebConsole from 192.168.10.101
22:06:35 RSL: Received Topic /Dimmer, Data Size 2, Data 50
22:06:35 RSL: Group 0, Index 1, Command DIMMER, Data 50
22:06:35 MQT: stat/living-room-light-3/RESULT = {"POWER":"ON","Dimmer":50,"Color":"128,0","HSBColor":"0,0,0","Channel":[50,0],"CT":153}
22:06:35 CFG: Saved to flash at F7, Count 1298, Bytes 3584
22:06:37 CMD: Dimmer 100
22:06:37 SRC: WebConsole from 192.168.10.101
22:06:37 RSL: Received Topic /Dimmer, Data Size 3, Data 100
22:06:37 RSL: Group 0, Index 1, Command DIMMER, Data 100
22:06:37 MQT: stat/living-room-light-3/RESULT = {"POWER":"ON","Dimmer":100,"Color":"255,0","HSBColor":"0,0,0","Channel":[100,0],"CT":153}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Set the color temperature to "Cool" then move the brightness slider

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Moving the brightness slider should change the brightness, not the color temperature

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

chrisnharvey, 2019-07-13:
I've tried downgrading these bulbs back to 6.5.0, but the problem now seems to persist. Even after a reset in the console.

s-hadinger, 2019-07-13:
I don't own a Philips Xiaomi bulb so it's not easy for me to test, but I think I know what's wrong.

I will look at it tomorrow.

chrisnharvey, 2019-07-14:
Great! Thank you.

If you need me to test anything on one of my bulbs then just let me know.

s-hadinger, 2019-07-14:
Can you please test PR #6092 ?

closed 2019-07-14 by arendst

chrisnharvey, 2019-07-14:
Just tested your PR. Unfortunately, the issue still persists on my Xiaomi Philips E14 bulb.

chrisnharvey, 2019-07-15:
Weirdly, I've flashed some older versions of Tasmota which were previously working but don't anymore.

I've tried a settings reset in Tasmota, but this doesn't seem to help either?

Is it possible that the 6.6.0 release has changed something that wouldn't be overwritten by an OTA flash?

Is there any way I can fully erase and re-flash the firmware without soldering? Opening these bulbs up is a pain.

s-hadinger, 2019-07-15:
This is weird. I can't think about anything changed by 6.6.0 that could affect 6.5.0. 

Could you try to set the module as 'Sonoff B1', then select 'color 0000000000' and set back to Xiaomi. I want to make sure you don't have a residual color in the settings in Flash.

chrisnharvey, 2019-07-19:
Just tried your suggestion. Still not working.

I would put this down to a hardware issue, but I have 6 of these bulbs and they’re experiencing the same issue.

chrisnharvey, 2019-07-19:
Can this issue be reopened?

s-hadinger, 2019-07-20:
You flashed all 6 to 6.6.0?

Normally doing an erase using 6.5.0 should not have residual effect from 6.6.0

Can you try different combinations of 'color 00ff', 'color ff00' and others, and report what it does to the bulb?

s-hadinger, 2019-07-20:
Can you send the exact reference of the bulbs?

chrisnharvey, 2019-08-21:
Sorry for the delay in getting back, this is still an issue for me, but I've kind-of resolved it using the generic configuration and mapping GPIO12 to PWM2 and GPIO15 to PWM1. This resolves the problem, but the Cool and Warm slider is inverted, so Cool=Warm and Warm=Cool.

Weirdly, I tried creating a template based on the Xiaomi Philips module, which actually maps the GPIO pins as I did using the generic module, but this doesn't work if I select Xiaomi Philips as the configuration to base the template on, I have to select "Generic" and map the GPIO pins manually.

Hope this makes sense.

chrisnharvey, 2019-08-21:
@arendst Can this issue be re-opened?

s-hadinger, 2019-08-22:
If `PWM2` is inverted, you can define it as `PWM2i`. The trailing `i` means it is inverted.

chrisnharvey, 2019-08-23:
That’s what I thought, but doing that produces weird behaviour, similar to if I use the default Xiaomi module type.

There is something really weird going on with these bulbs.

PR #6092 (2019-07-14): Xiaomi Philips brightness (#6091)
## Description:

I don't own a Philips Xiaomi bulb, but simulated on a led device. This should be back to 6.5.0 behavior.

**Related issue (if applicable):** fixes #6091 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2 and 2.5.2
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tas [... cut]