arendst/Tasmota #6098: Issue in Timers

Adityakush, 2019-07-15:
I have a question about timers. When I arm any timer to toggle output and set the time, it does not trigger if todays weekday is not checked, however i have not turned on repeat and as far as i know, there is no need to check weekdays if you want to trigger alarm just once. 
To be sure I checked my mobile phone alarm clock also and found the working to be different from how timers work in tasmota.
Is this something delibrately done in tasmota or am I wrong somewhere or missing something.

Adityakush, 2019-07-15:
In a normal alarm clock without repeat mode turned on, the alarm will trigger once irrespective of any weekdays set or not. So weekdays are usable with the repeat mode only. However in tasmota, even though repeat mode is off, and all weekdays unchecked leads to not triggering of the alarm at the set time. Only works if you set todays weekday on.

ascillato2, 2019-07-15:
Hi

Sorry, it is not a bug. It is how timers works in Tasmota. 

If you need further assistant, please find us in the Tasmota support chat. 

Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-15 by ascillato2