arendst/Tasmota #6100: Irsend doesn't work for pioneer

Oprtyo, 2019-07-15:
I have YTF IR Bridge with latest Tasmota and reciever Pioneer.
When I press turn on on remote control i recieve:
22:16:01 IRR: Echo 0, RawLen 68, Overflow 0, Bits 32, Value 0xA55A38C7, Decode 3
22:16:01 RSL: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0xA55A38C7","RawData":[8522,4252,536,1586,534,538,532,1586,534,540,532,542,528,1588,532,542,528,1588,532,542,532,1586,534,538,506,1614,508,1610,532,540,532,1590,532,542,524,544,536,538,532,1586,534,1584,534,1588,534,536,534,540,532,538,534,1588,532,1586,534,540,532,540,532,538,534,1588,532,1586,534,1586,530],"RawDataInfo":[67,67,0]}}
When I do:
IrSend {"Protocol":"NEC","Bits":32,"Data":"0xA55A38C7"} - nothing
Irsend 0,8522,4252,536,1586,534,538,532,1586,534,540,532,542,528,1588,532,542,528,1588,532,542,532,1586,534,538,506,1614,508,1610,532,540,532,1590,532,542,524,544,536,538,532,1586,534,1584,534,1588,534,536,534,540,532,538,534,1588,532,1586,534,540,532,540,532,538,534,1588,532,1586,534,1586,530 - nothing
I've found Hex for my device in the web:
0000 0068 0000 0022 0168 00b4 0016 0043 0016 0016 0016 0043 0016 0016 0016 0016 0016 0043 0016 0016 0016 0043 0016 0016 0016 0043 0016 0016 0016 0043 0016 0043 0016 0016 0016 0043 0016 0016 0016 0016 0016 0016 0016 0043 0016 0043 0016 0043 0016 0016 0016 0016 0016 0016 0016 0043 0016 0043 0016 0016 0016 0016 0016 0016 0016 0043 0016 0043 0016 0043 0016 06d9
decoded it in IrScrutinizer and got:
9032,4516,552,1681,552,552,552,1681,552,552,552,552,552,1681,552,552,552,1681,552,552,552,1681,552,552,552,1681,552,1681,552,552,552,1681,552,552,552,552,552,552,552,1681,552,1681,552,1681,552,552,552,552,552,552,552,1681,552,1681,552,552,552,552,552,552,552,1681,552,1681,552,1681,552,43982 - which is pretty similar to received data in console.
tried irsend 0, 9032,4516,552,1681,552,552,552,1681,552,552,552,552,552,1681,552,552,552,1681,552,552,552,1681,552,552,552,1681,552,1681,552,552,552,1681,552,552,552,552,552,552,552,1681,552,1681,552,1681,552,552,552,552,552,552,552,1681,552,1681,552,552,552,552,552,552,552,1681,552,1681,552,1681,552 - nothing
probably the problem in frequency, because my device use protocol Pioneer (not nec)...
what should i put instead of 0 (38kHz) to get 40?
Maybe there is smth. else I should do?

Oprtyo, 2019-07-15:
I can control my TV LG via YTF IR Bridge without any problems btw.

Jason2866, 2019-07-16:
Try instead of 0 the value of your needed frequency.
Maybe 39857 for pioneer

Oprtyo, 2019-07-16:
Here is the log:
 10:27:49 CMD: irsend 39857,9032,4516,552,1681,552,552,552,1681,552,552,552,552,552,1681,552,552,552,1681,552,552,552,1681,552,552,552,1681,552,1681,552,552,552,1681,552,552,552,552,552,552,552,1681,552,1681,552,1681,552,552,552,552,552,552,552,1681,552,1681,552,552,552,552,552,552,552,1681,552,1681,552,1681,552
10:27:49 SRC: WebConsole from 192.168.0.100
10:27:49 RSL: Group 0, Index 1, Command IRSEND, Data 39857,9032,4516,552,1681,552,552,552,1681,552,552,552,552,552,1681,552,552,552,1681,552,552,552,1681,552,552,552,1681,552,1681,552,552,552,1681,552,552,552,552,552,552,552,1681,552,1681,552,1681,552,552,552,552,552,552,552,1681,552,1681,552,552,552,552,552,552,552,1681,552,1681,552,1681,552
10:27:49 RSL: stat/sonoff/RESULT = {"IRSend":"Done"}
10:27:49 IRR: Echo 1, RawLen 68, Overflow 0, Bits 32, Value 0xA55A38C7, Decode 3
but pioneer didn't responded

Oprtyo, 2019-07-17:
So, I'll add some information:
I've downloaded and installed version 6.6.0.2 with new option Irsend "repeat"
IrSend {"Protocol":"NEC","Bits":32,"Data":"0xA55A38C7","repeat":3} also no success.
I can see how leds are blinking but my device can't receive the signal.

S-hadinger tried to help me, he has pioneer VSX-c300 which uses same ir codes as my pioneer VSX-820.
IrSend {"Protocol":"NEC","Bits":32,"Data":"0xA55A38C7"} works for him, not for my pioneer.
We have one and the same device YTF IR Bridge and same firmware. I can't understand what is the problem connected with.
Maybe my device is more sensitive to the signal....
pleae help

s-hadinger, 2019-07-20:
After lots of trial and error it appears that ir timing on pioneer is 5% shorter than Nec, probably due to a frequency of 40kHz instead of 38kHz. Duty cycle is also different which means it can't be done with current code. 

I'll prepare a lightweight PR next week to solve this.

arendst, 2019-07-20:
Why all the trouble to make pioneer work using the NEC protocol?

Irremote supports pioneer as a seperate protocol so that will probably work just fine. All it needs is enabling it (optionally) in IRremoteESP8266.h and make an entry in Tasmota. It isn't supported yet because I hate the way this can be implemented as it will increase tasmota code with every protocol enabled while at the same time I see no easy way in making protocols configurable at compile time.

Also see this: https://github.com/arendst/Sonoff-Tasmota/issues/5853#issuecomment-513452905

crankyoldgit, 2019-07-20:
Pioneer is a 64 bit protocol. It's basically two NEC 32bit (typically both different) messages back to back.
A `repeat` of a simple NEC message will not help you. Also, as far has been reported to me, it is a 38kHz protocol, and the duty cycle really won't make a difference typically. i.e. You are barking up the wrong tree. ;-)

See https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Pioneer.cpp#L31 if you want to learn more.

s-hadinger, 2019-07-20:
I understand your concerns. This was my first option but it doesn't work. Irremote is wrong, it takes the assumption that nec and Pioneer have the same timings but they don't. Most devices still work but some don't. 

Please look at my PR first, it's basically 2 lines of code, calling sendGeneric with different timings. I'll measure the code increase.

s-hadinger, 2019-07-20:
@crankyoldgit believe me it does make a difference. I know about the 64 bits difference, most simple commands like Vol+ are still 32 bits. 

I have the proof that the same sendNec() command with NEC timings or Pioneer timings - the first one fails the second works. 

Most Pioneer equipment don't care but some do.

s-hadinger, 2019-07-20:
Pioneer detailed timings : http://www.adrian-kingston.com/IRFormatPioneer.htm

crankyoldgit, 2019-07-20:
@s-hadinger I'm happy to add a 40kHz sending variant with different timing etc to [IRremoteESP8266](https://github.com/crankyoldgit/IRremoteESP8266) if you want. Just log an issue etc.

I'm hesitant to change the FM of the existing `sendPioneer()` as so far it's working for everyone else. But if you have a wide range of equipment to verify against. I could easily have my arm twisted to make it the default.

crankyoldgit, 2019-07-20:
> Pioneer detailed timings : http://www.adrian-kingston.com/IRFormatPioneer.htm

FYI, the duty cycle on the Oscilloscope (per link) is pretty darn close to 33%, which is what the NEC spec says it is, and is what is used by `sendNEC()`. The FM is typically gated at/by the hardware demodulator. It could be the problem. However, it's typically is with in tolerance.

FYI, the micro-second timings difference for 38kHz@33% vs 40kHz@33% during a "mark" signal are:
[38kHz@33%](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/test/IRsend_test.cpp#L252): `[On]6usecs[Off]15usecs[On]6usecs[Off]15usecs`
 [40kHz@33%](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/test/IRsend_test.cpp#L306): `[On]6usecs[Off]14usecs[On]6usecs[Off]14usecs`

Sure, there is a difference, not arguing that, but it's 1 micro-second. But hey, your hardware and experience says it matters. I believe you. Happy to help/try it out for you etc. Due to the limitations of the ESP8266 hardware, the library uses software PWM (there is no hardware PWM at that resolution  on the platform, except for the serial line), and thus the practical limit of resolution is 1usec granularity. i.e. fractional usecs are completely out. I draw the line at counting cpu cycles to go to that level of precision. ;-)

If was to "bet" where the timing problem was based on experience with all the issues reported to me with the library and expected-device compatibility, 90+% of the time it's a total length of a "mark" / "space" component of the signal, not the underlying FM (<10%) and so far only once the underlying duty cycle.
In almost all cases, the problems were "intermittent" reception of a signal. i.e. It was not 100% reliable at 38kHz, but at 40kHz it was 100%.

crankyoldgit, 2019-07-20:
Oh, before you ask, the above reported timings *do not* include the 5usec delay in switching the GPIO on and off.

e.g. 6(on) + 14(off) + 5(overhead) == 25usec wavelength == 40kHz.
i.e. it takes about 2.5usecs per transition. 2.5us(to turn on) + wait 6us + 2.5us(to turn off) + wait 14us == 25us.

s-hadinger, 2019-07-20:
Thanks for the detailed answer.

I changed 3 things to make it work: timing, freq and duty cycle. Let me check now which one is important.

s-hadinger, 2019-07-20:
After more tests, it's not the frequency nor the duty cycle, just the timing. Changing the tick from 560 us to 530 us makes it work.

s-hadinger, 2019-07-22:
@crankyoldgit I didn't realize your are the author of IRRemoteESP8266. Awesome work! I love it!

I submitted a PR to IRRemoteESP8266 to adjust both frequency and timing. As you mentioned, duty cycle has no impact.

@Oprtyo confirmed it works with VSX-820 and it works also with my VSX-C300 (that worked also with sendNEC).

PR #6119 (2019-07-22): Fix send IRSend for Pioneer devices (#6100)
## Description:

After spending the ween-end to remote debug the issue from #6100 and with the help of @oprtyo

It appears that Pioneer timings are 5% shorter than NEC. If you're lucky, some devices will tolerate the difference, but some will not. This fixes aligns timing to actual Pioneer IR. For example, the NEC preamble is 9ms + 4.5ms, whereas the Pioneer timing is 8.5ms and 4.25ms. Frequency is also 40KHz instead of 38KHz.

I also introduced an easier way to add IR codes above SHARP (14). Pioneer code is 50 in IRRemoteESP8266, which would require [... cut]
PR #6145 (2019-07-29): Upgrade library IRRemoteEsp8266 to 2.6.4, now using sendPioneer()
## Description:

Upgraded IRRemoteEsp8266 to 2.6.4 which now includes patched Pioneer protocol.
Specifically updated IRremoteESP8266.h according the previous Tasmota configuration.
Changed xdrv_05_irremote.ino to now use sendPioneer().

Code impact: 32 bytes.

**Related issue (if applicable):** fixes #6100

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2 [... cut]
closed 2019-07-22 by arendst