arendst/Tasmota #6102: Flashed Basoc R3 with OTA in DIY mode

auriux, 2019-07-16:
Hi, could someone help me.. I messed up. Added the jumper to activate DIY mode and used DIY Tool to flash tasmota minimal.bin, tried normal version of tasmota, but it say the filesize was too big. After the flash complete my R3 is not responding, no lights flashing, nothing. No sign that it works. Did I made a big mistake? Or I just could order serial to USB adapter and try to flash it in the old way?

arendst, 2019-07-16:
The DIY ota procedure won't work with tasmota directly as tasmota expects execution from address 0. DIY ota expects espressif images starting at either 1000 or 8100.

You now can revive your hardware simply by flashing the  old way.

auriux, 2019-07-16:
Ok, will wait for the serial to USB adapter.
Should i leave the DIY mode jumper on before flashing?
Thank you.

arendst, 2019-07-16:
I think the jumper should be removed. 

I will check it's functionality the next days as it would be great to get tasmota on it by OTA considering the fact that the Sonoff Mini already comes with the jumper installed.

I expect a solution like Tuya-convert where an interim firmware is loaded first at 8100 which in turn loads tasmota at 0. Perhaps it's already possible with this tool if the DIY parameters are set correctly. I'll have to investigate.

auriux, 2019-07-16:
That sounds great. Easier to flash tasmota the better. I think there are lot of people who don't know how to use soldering iron. This method would help a lot.

nicgord, 2019-07-17:
Have a few of these arriving very soon so would be great to avoid soldering and use native upload capability...

auriux, 2019-07-18:
Dont get it.. I see this video: https://youtu.be/RT_s2R39i9o?t=3300
And id does the same I did, uploaded the firmware, rebooted and tasmota is running..

arendst, 2019-07-18:
Wow! (old man cheering). That worked right out of the box!!

First I looked at your video from where you suggested and noticed that DrZZ managed to get Tasmota running. Then I tried to see the video from the beginning but It couldn't keep my attention long enough to see how he did it.

Here is what I did:
- Leave the jumper out and connect to eWelink
- Update firmware to at least 3.1 (mine came with 3.0)
- Power off and install jumper
- Download the Sonoff DIY package from Github at https://github.com/itead/Sonoff_Devices_DIY_Tools
- Modify your laptop (windows 10) configuration to provide a Mobile hotspot with SSID sonofDiy (case sensitive!) and password 20170618sn
- Power on Basic R3 and verify it pops up on the Mobile hotspot Settings page
- Start \tool\tool_01DIY85(v1.0.0).exe
- It should find the Basic R3
- Select it and play with ON and OFF to verify correct connection to Basic R3
- Select Firmware flash
- Select pre-compiled or own compiled binary smaller than 500k. Do NOT SELECT sonoff-minimal as it does not allow to change any setting.
- Select the device in the flash pop-up and select OK
- Tasmota will be uploaded an start running

No continue as if you just installed Tasmota on the device.

To clean up:
- Quit DIY mode tool
- Stop mobile hotspot

closed 2019-07-18 by ascillato2

Jason2866, 2019-07-18:
More infos for this "easy" process: https://github.com/itead/Sonoff_Devices_DIY_Tools/blob/master/other/SONOFF%20DIY%20MODE%20Protocol%20Doc.pdf

auriux, 2019-07-18:
So I messed  up choosing minimal version. Is there any way to flash ota from minimal? Or my only choice with serial connection? Dont really know what minimal includes.

Jason2866, 2019-07-18:
If you have access to the device you can flash a normal sonoff version via OTA

auriux, 2019-07-19:
It should create a Wifi hotspot with minimal version too?

arendst, 2019-07-19:
I don't think so. I removed all code not needed during minimal activity. See release notes:

> sonoff-minimal.bin = The Minimal version allows intermediate OTA uploads to support larger versions and does NOT change any persistent parameter. This version should NOT be used for initial installation.

auriux, 2019-07-19:
Ok, i see. If there is no Wifi hostpot, how can I connect to it to flash via ota?

Jason2866, 2019-07-19:
No way. Needs serial flashing

auriux, 2019-07-19:
Thats what i thought. Will wait for serial adapter and flash it.

fragolinux, 2019-07-21:
> * Modify your laptop (windows 10) configuration to provide a Mobile hotspot with SSID sonofDiy (case sensitive!) and password 20170618sn
> 
> [image]

typo, in text it says "sonofDiy", which should be "sonoffDiy" (in image is correct)

fragolinux, 2019-07-21:
@arendst 

> * Modify your laptop (windows 10) configuration to provide a Mobile hotspot with SSID sonofDiy (case sensitive!) and password 20170618sn
> 
> [image]

typo, in text it says "sonofDiy", which should be "sonoffDiy" (in image is correct)

arendst, 2019-07-21:
Thx. Just updated

arendst, 2019-07-22:
Info added to wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-DIY

titomontes, 2019-07-25:
> Thats what i thought. Will wait for serial adapter and flash it.

@auriux I think you don´t have to flash it. You just need to connect to serial and send commands with Thermite (set SSID and PW) so you can connect to the webserver and upload the sonoff.bin image.

Am I correct?

meingraham, 2019-07-25:
@titomontes @auriux 

Can't hurt to give it a try. You have the serial adapter connected... sending the SSID & password with a terminal app is easy to try. If that gets you to the point where you can then connect and replace the firmware OTA, then off to the races. Otherwise, you have the device connected and you can go through the regular erase and upload with the "right" binary.

Mike

auriux, 2019-07-26:
@titomontes @meingraham Will do that now. I see this pinout:
https://user-images.githubusercontent.com/35382385/59029435-9505cf00-885e-11e9-8837-aeefa060c3c5.jpg
The RX from the picture goes to TX on the serial adapter? Or its marked for serials wires already?

meingraham, 2019-07-26:
Thanks for performing the test. Now we know. Glad you recovered your device.

titomontes, 2019-07-27:
So this can be closed now. Cheers.