arendst/Tasmota #6104: Today

ninhvanthang, 2019-07-17:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : _____
- [ ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-07-17:
Behaviour is as expected. Power Monitoring values are only stored one per day in flash (around midnight)
This needs to be done to reduce flash writes. If flash write would be done more often device would fail with
defect flash. If the values are mandantory for you save it in a database, all Home Automation systems are capable of this.

ninhvanthang, 2019-07-17:
> Behaviour is as expected. Power Monitoring values are only stored once per day in flash (around midnight)
> This needs to be done to reduce flash writes. If flash write would be done more often device would fail with
> defect flash. If the values are mandantory for you save it in a database, all Home Automation systems are capable of this.

Do you mean that total value is stored in RAM, and it's written to flash one time per day ?

Jason2866, 2019-07-17:
yes, all power values are stored in RAM and written one time per day to flash.

ninhvanthang, 2019-07-17:
ok, Thanks for your info

closed 2019-07-17 by ninhvanthang

al13nus, 2022-07-09:
So writting to flash one time per day does make sense but how to I manualy set the total energy consumsion then?

sfromis, 2022-07-09:
See commands here: https://tasmota.github.io/docs/Commands/#power-monitoring
(including to set total energy consumption)

al13nus, 2022-07-09:
> 

Yeah, saving before would have been great, nevermind:

`13:11:57.934 CMD: set EnergyTotal 1033
13:11:57.941 MQT: stat/tasmota_D537DC/RESULT = {"Command":"Unknown"}`

what I'm doing wrong?

brjhaverkamp, 2023-01-04:
I got hit by this issue as well. I understand the background and only writing to flash once a day.
 I had a couple of (unplanned) outages on parts of my home the past weeks and my graphs look like this now:
[image]
But it could be made more intelligent, as the updated value total can never be smaller than the old value, right.
So you could do something like this, in pseudo-code:
```
If new_total > old_total
    write new total to flash
if  new_total <old_total
   write old_total+daily usage to flash
```

This way you only use the amount between the previous write and the power outage, which is the best you can get in this case.

arendst, 2023-01-04:
Try latest dev release. 

It was recently amended with a feature setting the registers after power on like it was midnight the day before.

Might solve your issue.

brjhaverkamp, 2023-01-05:
Nice, as always you are way ahead of me:-)