arendst/Tasmota #6105: SDM120 default baudrate

gajotnt, 2019-07-17:
If the default SDM120 Baudrate is 2400 why is it that in the source code it's set to 9600?

`  #define SDM120_SPEED         9600              // SDM120-Modbus RS485 serial speed (default: 2400 baud)`

Anyone that flashes the binaries and has a SDM120 will have to recompile with the correct settings. Maybe there's an explination for this, maybe it just got under the radar :)

Jason2866, 2019-07-18:
The default baudrate is 2400. See http://www.eastrongroup.com/data/uploads/Eastron_SDM120-Modbus_user_manual_2016_V2_6.pdf

gajotnt, 2019-07-18:
I know... but in the tasmota source code it is set to 9600 by default.

https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/my_user_config.h

Line 407
`#define SDM120_SPEED 9600 // SDM120-Modbus RS485 serial speed (default: 2400 baud)
`

Jason2866, 2019-07-18:
Do a PR to change ;-)

Jason2866, 2019-07-18:
Fixed with PR #6111 
Thx, please close

closed 2019-07-18 by gajotnt