arendst/Tasmota #6106: Security : MQTT broker name without external/server DNS resolution

dgrostoto, 2019-07-18:
**Have you look for this feature in other issues and in the wiki?**
Tasmota rely on either IP or DNS name with mDNS or SD DNS resolution which imply a DNS server.

**Is your feature request related to a problem? Please describe.**
The use case is a public lan. Call to MQTT broker need to be secure. The problem is : TLS is good but imply a hostname for mqtt broker and DNS request for resolving are not secure.
The server certificate could do the job, but mqtt use learning fingerprinting instead of using public server key for authenticating server.

**Describe the solution you'd like**
A good workaround would be to have a client DNS resolution such as the linux file /etc/hosts. So tasmota doesn't have to rely on any name resolution service, which is OK from a secure point of view.

**Describe alternatives you've considered**
The MQTT client could also check server certificate, but it seems a much more complex solution.

**Additional context**

**(Please, remember to close the issue when the problem has been addressed)**

Thanks in advance for your time.

s-hadinger, 2019-07-18:
I don't undestand why you wouldn't just put an IP address as MQTT server. It would not be different to a local /etc/host.

Btw, the current TLS implementation in Tasmota checks either CA or Fingerprint, but does not check that the server matches the name in the certificate. So putting an IP address will work.

dgrostoto, 2019-07-18:
OK to put IP since there is no checking about DN from the server certificate.
Just a point that worry me. I our conversation you mentionned "Tasmota TLS does not check the domain name of the server. It only checks that the certificate is valid. So you don't need to fake a dns server. However IP does not work, that's definitely a bug." 

Can you clarify this bug ?

s-hadinger, 2019-07-18:
It's just that I never tried entering an IP address as TLS MQTT Host. I always tried with DNS name.

It should work seamlessly with IP address.

closed 2019-07-29 by ascillato2