arendst/Tasmota #6107: Can Var and Mem issued without parameters return a dictionary similar to RuleTimer?

jziolkowski, 2019-07-18:
Hi,

I'm working on a tool to help manage/configure Tasmota (TDM) and in the rules editor I'm working on the functionality to watch (by polling) the state of var's, mem's and ruletimers. It works great for the latter, but to get the current values of variables I must send 10 mqtt messages per polling cycle time, which itself is not resource-intensive, however not optimal (and not pretty). Unless there is a reason for this difference, could the behavior be made more consistent and have the "empty" var and mem command return the dictionary with current values?

arendst, 2019-07-24:
Must be doable. I'm looking for the smallest code implementation...

jziolkowski, 2019-07-24:
Thank you so much! Awesome!

closed 2019-07-24 by jziolkowski

arendst, 2019-07-24:
This changes the functionality of commands Var and Mem without an index. Command ``Var`` is now different from ``Var1``. Also ``Mem`` is not equal to ``Mem1``.

``Var`` now returns ``{"Var1":"","Var2":"","Var3":"","Var4":"","Var5":""}`` where ``Var1`` returns ``{"Var1":""}``
``Mem`` now returns ``{"Mem1":"","Mem2":"","Mem3":"","Mem4":"","Mem5":""}``

jziolkowski, 2019-07-24:
Yup, we'll have to update wiki accordingly pointing out this change.

meingraham, 2019-07-24:
wiki - done