arendst/Tasmota #6108: Support for Shelly Plugs S

m0rph2k, 2019-07-18:
Hi

can tasmota get support for Shelly Plug S ?

Jason2866, 2019-07-18:
Have you tried templates from similar looking devices?
For example https://blakadder.github.io/templates/blitzwolf_SHP6.html

m0rph2k, 2019-07-18:
No, I will try it

m0rph2k, 2019-07-18:
mhh i open the Shelly-Plug-S but no RX,TX GND label

Jason2866, 2019-07-18:
Do upload High Res pictures here, maybe we can help

m0rph2k, 2019-07-18:
I found this.I will test it
https://faulty.cloud/blog/shelly-plug-s-pinout

Jason2866, 2019-07-18:
Take a look in this template too, if there is a temp sensor in the Shelly S you can adapt from to try to get it working https://blakadder.github.io/templates/shelly_25.html

ascillato2, 2019-07-23:
>Ok, Shelly S works with Tasmota 6.6.0 and the "BlitzWolf SHP" Template.

Great. Closing this issue as it is solved. Thanks

closed 2019-07-23 by ascillato2

xppx99, 2019-07-29:
Hi there! I've managed to flash Shelly Plug S following this guide, and aplying the "blitzWolf SHP" almost everything works. 
One sugestion though is that for someone to update the wiki with this device, please.

meingraham, 2019-07-30:
@xppx99 

You can submit a new entry in the [template repository](https://blakadder.github.io/templates/). Just name the entry for this device and provide the template you used (similar or same as SHP).