arendst/Tasmota #6109: Intermittent WiFi Connection

avrhack, 2019-07-18:
# ISSUE DESCRIPTION - TROUBLESHOOTING

I've noticed that WiFi drops out regularly with versions of Tasmota later than 6.4.1, specifically I've tried 6.5.0 and 6.6.0.

I appreciate this has been raised multiple times before, but I have a specific reason for bringing it up again which should become clear below so please let's not see this just closed with the same links as before.......

I've spent a lot of time reading the various issues that have been raised on this topic before including 4896, 4899 and as far back as 1824. I've also seen the WiKi references etc. on the fact that some WiFi APs seem to struggle with compatibility with various Arduino Core versions e.g. 2.3.0 works for some people and 2.5.2 for others.

From the issues I've read, the key advice seems to be to go back to 2.3.0 for those people having trouble with later versions, however now that e.g. TLS is no longer supported under 2.3.0 it doesn't seem that that is an option any longer - please correct me if I'm wrong on this - but in any event I'm hoping that by collaborating on what the underlying problem actually is, we can find a solution that works more permanently than different people having to use different versions of Tasmota.

### REQUESTED INFORMATION

I've seen the issue on a Blitzwolf BHP4, several Sonoff S20s, a couple of ESP12E/Fs, a Shelly1 and an ESP01S so there's no common hardware apart from the esp8266 itself of course.

I've also seen the issue on three different APs - my current UniFi AC Lite, a BT HH5 and a TPLink WD9970 so again it's not specific to the AP.

However with all of these, Tasmota v6.4.1 (core 2.3.0) works significantly more reliably than later versions. I'm currently on v6.6.0 (core 2.5.2).

In all cases the devices have been flashed both OTA and also tried completely clean with a serial port erase_flash, program in dout and reconfigure manually which has no effect on the problem.

The one thing I **have** found that **significantly** affects how often the problem occurs is the sleep mode in Tasmota. I've experimented with both dynamic and 'original' sleep and with various settings of both.

In some of the upstream issues raised against Arduino core, sleep was suggested as being the cause of this problem and disabling sleep entirely stops it, however that hasn't been my experience with Tasmota. Changing setoption60 to on and then sleep to 0 disables sleep entirely (I believe) and certainly current consumption suggests that to be the case, but I still get some drops. However I'm not experimenting with going back to dynamic sleep and with a setting of 250 and things seem more stable.

FWIW the output of my status 0 on one of my devices exhibiting this is as follows:

`Received 'stat/kitchen/STATUS', '{"Status":{"Module":8,"FriendlyName":["kitchen"],"Topic":"kitchen","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}' len 286
Received 'stat/kitchen/STATUS1', '{"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://apu:80/sonoff/sonoff.bin","RestartReason":"Power on","Uptime":"4T12:54:41","StartupUTC":"2019-07-13T21:14:28","Sleep":10,"CfgHolder":4617,"BootCount":5,"SaveCount":22,"SaveAddress":"F5000"}}' len 262
Received 'stat/kitchen/STATUS2', '{"StatusFWR":{"Version":"6.6.0-a3a1fbd0(sonoff)","BuildDateTime":"2019-07-10T14:59:15","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}' len 140
Received 'stat/kitchen/STATUS3', '{"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":5,"LogHost":"XXX","LogPort":514,"SSId":["BadgerI",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008008","280500000100060000000000000000000000","00000404"]}}' len 218
Received 'stat/kitchen/STATUS4', '{"StatusMEM":{"ProgramSize":473,"Free":528,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0EAAA346","041403A0","201617CD","010013C0"]}}' len 200
Received 'stat/kitchen/STATUS5', '{"StatusNET":{"Hostname":"sonoff-kitchen","IPAddress":"192.168.151.88","Gateway":"192.168.151.199","Subnetmask":"255.255.255.0","DNSServer":"192.168.151.199","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":5}}' len 214
Received 'stat/kitchen/STATUS6', '{"StatusMQT":{"MqttHost":"apu","MqttPort":8883,"MqttClientMask":"sonoff-kitchen","MqttClient":"sonoff-kitchen","MqttUser":"sonoff","MqttCount":32,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}' len 185
Received 'stat/kitchen/STATUS7', '{"StatusTIM":{"UTC":"Thu Jul 18 10:09:09 2019","Local":"Thu Jul 18 11:09:09 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99}}' len 171
Received 'stat/kitchen/STATUS10', '{"StatusSNS":{"Time":"2019-07-18T11:09:09+01:00"}}' len 50
Received 'stat/kitchen/STATUS11', '{"StatusSTS":{"Time":"2019-07-18T11:09:09+01:00","Uptime":"4T12:54:41","Heap":29,"SleepMode":"Normal","Sleep":10,"LoadAvg":99,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":6,"RSSI":70,"LinkCount":24,"Downtime":"0T00:01:22"}}}' len 263`

The output from the log (over port 514, can't get serial unfortunately) sheds no light either other than that this is clearly a disconnect of WiFi and not a restart of the esp8266. Excerpt is:

`Jul 17 15:31:56 sonoff-kitchen ESP-WIF: Checking connection...
Jul 17 15:31:56 sonoff-kitchen ESP-WIF: Connected
Jul 17 15:32:16 sonoff-kitchen ESP-WIF: Checking connection...
Jul 17 15:32:16 sonoff-kitchen ESP-WIF: Connected
Jul 17 15:32:36 sonoff-kitchen ESP-WIF: Checking connection...
Jul 17 15:32:36 sonoff-kitchen ESP-WIF: Connected
Jul 17 15:32:49 sonoff-kitchen ESP-MQT: TLS connected in 4189 ms, max ThunkStack used 2708
Jul 17 15:32:49 sonoff-kitchen ESP-MQT: MFLN not supported by TLS server
Jul 17 15:32:49 sonoff-kitchen ESP-MQT: Server fingerprint: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX
Jul 17 15:32:49 sonoff-kitchen ESP-MQT: Connected
Jul 17 15:32:49 sonoff-kitchen ESP-MQT: tele/kitchen/LWT = Online (retained)
Jul 17 15:32:49 sonoff-kitchen ESP-MQT: cmnd/kitchen/POWER =`

### SUGGESTED NEXT STEPS / APPROACH

So my thinking is........

1. Clearly other people are still having this issue; several have tried to raise this and get help but it's still occurring.

2. Going back to 2.3.0 is no longer a viable workaround. 

3. It may be well be an issue with the arduino core, but things in the Tasmota code seem to significantly affect it so I'm hoping we can find a solution within Tasmota as the core itself still seems unable to find one other than disabling sleep entirely - and that doesn't work for me at least with Tasmota.

Looking for as many comments as possible please, preferably with some hard data on what works or doesn't work for you especially with different settings of sleep and whether any of them solve the problem.

Jason2866, 2019-07-18:
It is NOT under the control of Tasmota as you wrote correct it is Arduino Core related.
Tasmota just uses the core. Tasmota gives you the choice to use the core that works best for you.
See https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#what-is-this-arduino-core
Sorry more cant be done from Tasmota side.

Jason2866, 2019-07-18:
You can try core pre 2.6.x with nonos SDK SDK22y. There are many good feedbacks.
Just enable in platformio.ini

avrhack, 2019-07-18:
> It is NOT under the control of Tasmota as you wrote correct it is Arduino Core related.
> Tasmota just uses the core

I understand that - as I said in my original post - the question I posed is whether this community, by doing some testing across a range of devices and settings, can find a workaround for what is a pretty serious flaw that the core developers have been unable to resolve for many, many releases over an extended period of time.

> You can try core pre 2.6.x with nonos SDK SDK22y. There are many good feedbacks.
> Just enable in platformio.ini

That's certainly something I hadn't considered thanks - will give it a try now and report back here on whether it seems to finally resolved this.

Jason2866, 2019-07-18:
> I understand that - as I said in my original post - the question I posed is whether this community, by doing some testing across a range of devices and settings, can find a workaround for what is a [...]

This has been done. Result is, there is NO general solution. For some works 2.4.2. for other do 2.5.2 work. There are too many parameters which differs. Buying new wifi devices (which are known to work) is definitly not a solution...

Jason2866, 2019-07-18:
My experience with pre 2.6.  on >20 devices  are similar to this

```
6:16:39 CMD: status 0
16:16:39 MQT: stat/sonoff-13D92F/STATUS = {"Status":{"Module":8,"FriendlyName":["Fluter"],"Topic":"sonoff-13D92F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:16:39 MQT: stat/sonoff-13D92F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"7T04:09:13","StartupUTC":"2019-07-11T10:07:26","Sleep":50,"CfgHolder":4617,"BootCount":63,"SaveCount":839,"SaveAddress":"F4000"}}
16:16:39 MQT: stat/sonoff-13D92F/STATUS2 = {"StatusFWR":{"Version":"6.6.0.1(sonoff)","BuildDateTime":"2019-07-11T11:46:00","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
16:16:39 MQT: stat/sonoff-13D92F/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","280500000100060000005A00000000000000","00000000"]}}
16:16:39 MQT: stat/sonoff-13D92F/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0F8AE386","040003A1","00441497","001000C0"]}}
16:16:39 MQT: stat/sonoff-13D92F/STATUS5 = {"StatusNET":{"Hostname":"sonoff-13D92F-6447","IPAddress":"192.168.2.119","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"EC:FA:BC:13:D9:2F","Webserver":2,"WifiConfig":5}}
16:16:39 MQT: stat/sonoff-13D92F/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-13D92F","MqttClient":"sonoff-13D92F","MqttUser":"openhabian","MqttCount":15,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:16:39 MQT: stat/sonoff-13D92F/STATUS7 = {"StatusTIM":{"UTC":"Thu Jul 18 14:16:39 2019","Local":"Thu Jul 18 16:16:39 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"05:29","Sunset":"21:09"}}
16:16:39 MQT: stat/sonoff-13D92F/STATUS10 = {"StatusSNS":{"Time":"2019-07-18T16:16:39"}}
16:16:39 MQT: stat/sonoff-13D92F/STATUS11 = {"StatusSTS":{"Time":"2019-07-18T16:16:39","Uptime":"7T04:09:13","Vcc":3.410,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

avrhack, 2019-07-18:
> My experience with pre 2.6. on >20 devices are similar to this
[snip]
> 

So in short are you saying you don't get **any** dropouts now on pre2.6 whereas you were on 2.5.2? If so that's a very good sign that the problem has finally been identified and fixed - very difficult to do of course when it's intermittent like this and people are doing it as volunteers in their spare time....

avrhack, 2019-07-18:
I'm going to close this as an issue as it seems there is little appetite for solving it within Tasmota as a workaround and given the views above that it's not possible to do so anyway. I will continue my own investigations separately alone.

closed 2019-07-18 by avrhack

ascillato, 2019-07-18:
>it seems there is little appetite for solving it within Tasmota

Sorry that your impression of all the hard work we and also the arduino team are doing for free and all our free time spent on this, is not enough for you.

The esp8266 chip doesn't have its low level radio routines as open source, so the bug for mqtt disconnections on core 2.4.2 and 2.5.2 can't be solved by us. The issue is in the SDK as was several times explained in the wiki, in a lot of other issues and also in arduino esp8266 repository and in the sdk repository.
The manufacturer is no longer solve issues for nonoSDK. They are moving to another type of SDK. A real time approach that require a huge amount of work to make that new sdk to be arduino compatible.
Now the last version of the arduino esp8266 core (actual stage core) that will become core 2.6.0 on September, 2 this year (2019) uses the SDK called 2.2y (instead of 2.2x) and it is much much better than the previous cores. 

If your rssi is greater than 40% you are not going to have mqtt disconnections. If it is below that, you will have issues.

So, sorry but your last comment was very offensive and not helpful at all. 

I recommend you to buy a ready solution from any manufacturer instead of using DIY devices and softwares.