arendst/Tasmota #6112: I would like to manage the PowerOnState for each relay.

nivek88, 2019-07-18:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

nivek88, 2019-07-18:
I would like to manage the PowerOnState for each relay of my Sonoff Dual Independently. I do have a Sonoff dual on my ceiling lamp. One relay controls the lamp and the other one powers a 3d printer. But each time I accidentally hit the wall switch for the ceiling lamp, the printer relay will be on as well. I always need to switch it off manually. But the printer is on the attic and I can't see it. The lamp lights up the hallway below the attic.

Or is there another way to set this?

meingraham, 2019-07-19:
You can set up rules which trigger on System#Boot to set each relay as you wish.

If you have specific questions about rules, please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-22 by ascillato2