arendst/Tasmota #6113: New kinetic tuya device - Model ERC309

kvvoff, 2019-07-19:
Please add a new device to the wiki: AC dimmer ERC310. 

As well as [ERC309 ](https://github.com/arendst/Sonoff-Tasmota/issues/6056)supports kinetic switches without batteries. I just chose the configuration 54 Tuya Dimmer and can control the dimmer. 
The firmware process is described in [ERC309](https://github.com/arendst/Sonoff-Tasmota/issues/6056).

But I have a few problems, namely:
1. On the main page there are 6 switches. SetOption41 does not help. (SetOption41 0, SetOption41 1 .... SetOption41 8) nothing changes at all. They are always 6 switches

2. When I add this dimmer to the Home Assistant through mqtt autodiscovery (SetOption19 1) It also shows six different switches, among which only one turns this dimmer on and off. 
Ok, I make a command (SetOption30 1) [that is listed here](https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#add-tasmota-devices-to-home-assistant) and in the assistant the light behaves generally strange: The first light just turns the dimmer on and off:
[image]

 And 6 light only changes the brightness of the slider, but DOES NOT allow the dimmer to be turned on or off:
[image]

Very strange.

What am I doing wrong? :)

meingraham, 2019-07-19:
@kvvoff,

It is the ERC309 that has the ESP chip which is flashed. The ERC310 is a package of the EC309 plus the kinetic switch; yes?

Is [this flashing guide](https://blakadder.github.io/templates/ERC309.html) accurate?

Mike

kvvoff, 2019-07-19:
@meingraham It looks similar, but there is no relay
[image]

kvvoff, 2019-07-19:
And it also connects to kinetic switches and allows to adjust the brightness

meingraham, 2019-07-19:
I may have the wrong impression of this device. I thought the ERC309 was just a controller and that it communicated with the switches which would have the relay circuitry.

kvvoff, 2019-07-22:
309 is a relay, but it is not controlled by gpio, but through a separate MCU
310 is a dimmer, which is also (like all Tuya dimmers) controlled through a separate MCU

[image]

The kinetic switch is a wireless switch (which looks like an Xiaomi Aqara) and it connects directly to the MCU (no matter what firmware is on esp8266)

kinetic switch (no battery need) with erc*** modules: 
[image]

Xiaomi Aqara(with battery):
[image]

ascillato2, 2019-07-29:
Hi,

Any news on this? Have you tried to make a template?

ascillato, 2019-07-30:
Hi,

Any news on this? Have you tried to make a template?

kvvoff, 2019-07-30:
@ascillato I have no changes, any help on this is welcome

ascillato, 2019-07-30:
If the device has a separated mcu to control the dimmer, you need to scan the communication between the esp8266 (with original firmware) and that mcu in order to have all required commands.

Do you have a oscilloscope or digital analyzer?

ascillato, 2019-07-30:
@kvvoff Have you tried latest Tasmota version from http://thehackbox.org/tasmota/020300/sonoff.bin

kvvoff, 2019-07-30:
@ascillato I'll try.  I will write the result

kvvoff, 2019-08-02:
@ascillato 
Thank you for help!
All wark with Tasmota **6.6.0.3** (http://thehackbox.org/tasmota/020300/sonoff.bin)

[image]

Please, add ERC310 Dimmer to blakadder.github.io

Thanks!

kvvoff, 2019-08-02:
I rushed to conclusions.  In HA, the light behaves strangely - when I turn it on, it turns on, but the switch status goes off.  brightness below about 30 percent shines brighter than brightness 70, 80, 90 and 100%.

ascillato, 2019-08-02:
That is not a Tasmota issue. That is the STATE topic you are looking. You have to put the same as you see in the console when you turn it on

kvvoff, 2019-08-02:
Ok, I try

ascillato2, 2019-08-04:
Hi, any news on this?

kvvoff, 2019-08-05:
@ascillato2 
The slider on a small level shines brightly. The slider at the maximum level shines medium. And it's in Tasmota 6.6.0.3 . I recorded a video:
https://www.youtube.com/watch?v=U8se0a50XDA

s-hadinger, 2019-08-05:
It looks like the Dimmer is inverted on this device. The fact that it's still medium when at maximum level is a side effect of Gamma correction. You can try temporarily setting `LedTable 0` and when at maximum level, it should be darker.

With PWM devices, you could choose PMWi to automatically invert the dimmer. On Tuya Serial device, I don't think we have this option. `pwm_inverted` is not available.

@arendst Theo any idea on how to do this properly?

kvvoff, 2019-08-05:
@s-hadinger The fact is that **on firmware version 6.6.0.0 the slider works as it should**, but **6 switches appear** (I wrote about this above). 

On firmware version 6.6.0.3 there is one switch, but it works like a video.

s-hadinger, 2019-08-05:
Ah ok. The thread is too long...

kvvoff, 2019-08-05:
Any thoughts on how to combine the 6 switches into 1?

ascillato2, 2019-08-12:
@kvvoff 

Hi, 

Have you tried latest version from development branch? (6.6.0.4)

kvvoff, 2019-08-13:
@ascillato2 Hi, I will try the new version, I will write the result

ascillato2, 2019-08-15:
Hi,

Any news on this?

kvvoff, 2019-08-16:
Hi, I'm sorry I didn’t answer for a long time.  I don’t have the opportunity to check now, I will check on Monday

ascillato2, 2019-08-16:
Ok, no problem. Let's wait until you have time. Thanks.

closed 2019-08-16 by ascillato2

kvvoff, 2019-08-20:
@ascillato 
Sonoff-Tasmota 6.6.0.6
Same. Slider to the maximum - the brightness is dim. Slider to the minimum - the brightness is even dimmer. A slider somewhere in the middle - the brightness is brighter. Is this exactly the case in the firmware? Maybe you need to make some settings?

kvvoff, 2019-08-20:
@ascillato At the moment, I have found only one way out. I am using the release version (with 6 switches). Next I added to Home Assistant lights.yaml:

> `- platform: group
>   name: ERC310 GROUP TEST
>   entities:
>     - light.erc310_dimmer_6
>     - light.erc310_dimmer`

[image]
[image]
[image]

That is, I use the 1 and 6 switch together. Since only one slider works in one, only a button in the other. I merged them and now I can press the switch and adjust the brightness. But one moment, after turning on, the slider is in the very left position, although the light shines brightly. That is, the status of the slider is not updated with respect to the actual brightness

ValiEne, 2020-07-07:
Hi guys, I've flashed an ERC310 with Tasmota 8.3.1. How can I use it as a dimmer? Does anyone have a Template for this device? Thanks

lucasdellasantina96, 2021-06-24:
Hello, I can not receive the code of 433 to some it happened to him?