arendst/Tasmota #6114: How get from Sonoff RF Bridge signal strength/quality (rssi)

fabiosoft, 2019-07-19:
I flashed the Sonoff bridge RF 433mhz with tasmota and my sensors feel reborn.

How can i know if the signal quality received from away sensors is good?
I have the bridge closed in a metallic closet, and i want to verify that it's able to receive the codes. 

Are there some command combinations to debug/show in console these informations?

Thanks

meingraham, 2019-07-19:
@fabiosoft 

Tasmota reports the Wi-Fi signal strength.

Are you using the Portisch RF firmware? The Portisch wiki doesn't not mention a means to see the RF signal strength, but it may be "embedded" in one of the RF messages. You may want to [post a question there](https://github.com/Portisch/RF-Bridge-EFM8BB1).

In any case, as your question seems to be related to the RF capabilities, that is not within the Tasmota realm. Please close this issue.

Mike

fabiosoft, 2019-07-19:
I know I can get WiFi signal strength, but I am interested to RF signal strength.

Unfortunately the json in console doesn’t have this field

meingraham, 2019-07-19:
Did you read the rest of my note?

Jason2866, 2019-07-19:
There is no signal strength for the RF 433 part in any message. It works or not...

fabiosoft, 2019-07-19:
@Jason2866 oh 😢 ...ok thanks.
@meingraham i use the standard tasmota firmware flashed on sonoff rf bridge.

closed 2019-07-19 by fabiosoft

meingraham, 2019-07-19:
@fabiosoft 

The RF bridge has the ESP chip and the RF MCU. BOTH of those have their OWN firmware. You flashed "standard Tasmota" onto the ESP chip. One is also able to flash a DIFFERENT binary altogether (Portisch) over the Sonoff RF firmware that comes with the Bridge. The Portisch RF firmware has its own set of commands.

fabiosoft, 2019-07-19:
Never used this firmware...does it work same way?

meingraham, 2019-07-19:
If the RF sensors you have work with the Bridge as is, I would not do anything at this time. The Portisch firmware expands on the RF sensors that are supported by the Bridge. For more information, refer to the Portisch GitHub I referred you to in my first comment.