arendst/Tasmota #6117: Teckin SP22 power meter showing voltage 30V to high

Matzebhv, 2019-07-20:
After flashing 2 of my teckin plugs it seems that measuring voltage is about 30V to high. Both plugs are showing above 260V, original firmware and measurement with an external voltage meter shows 231V. Firmware ist Sonoff-Tasmota 6.6.0, devicce configured as teckin.

Regards
Matze

Matzebhv, 2019-07-20:
[image]

meingraham, 2019-07-21:
https://github.com/arendst/Sonoff-Tasmota/wiki/Power-monitoring-calibration

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Matzebhv, 2019-07-21:
Thank you for that information. After calibration everthing works fine!

closed 2019-07-21 by Matzebhv