arendst/Tasmota #6122: set teleperiod via mqtt

prankousky, 2019-07-23:
Hi everybody,

I have read through `mqtt` and `commands`, but I cannot figure out how to do this...

`Information` states `MQTT full topic	home/tasmota/tasmota-4880/cmnd/`. So I tried changing `teleperiod` via mqtt by

* sending an empty message to `home/tasmota/tasmota-4880/cmnd/teleperiod 15`
* sending `15` to `home/tasmota/tasmota-4880/teleperiod`
* sending `teleperiod 15` to `home/tasmota/tasmota-4880/cmnd`

According to `console` the device also subscribes to `cmnd/tasmota-4880/#`, so I tried substituting this for the commands/topics mentioned above.

I had set `teleperiod 10` manually via console, so I can manually check `teleperiod` via console and receive `10`. What am I doing wrong? Thanks for your help :)

meingraham, 2019-07-23:
You have not provided the information requested. We cannot assist you without information to help us understand your current configuration and setup.

As this is usage rather than feature or bug, please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

prankousky, 2019-07-23:
@meingraham : I assumed because mqtt control ought to be universal, it would not matter what my actual hardware is. Anyway, sorry for leaving those details.

* device `sonoff basic`
* firmware `Sonoff-Tasmota 6.6.0.2 von Theo Arends`

There is absolutely no output that I can paste from the console, because all the console shows is what I'll just refer to as "regular stuff" (for example, current status etc.; these things will be displayed whether or not I send *anything* via mqtt, so they are not triggered by these commands).

This is my `information` pages content (as I am using the German IDE, unfortunately everything is German, but it should be graspable what is what (so if something is not the actual English translation I apologize, I just translated this on the go))

```
Tasmota Version	6.6.0.2(basic)
Build-Date & Time	2019.07.22 14:06:55
Core-/SDK-Version	2_5_2/2.2.2-dev(c0eb301)
Uptime	0T00:27:02
Amount Flash Writes	26 at 0xFA000
Amount Start Ups	3
Reason for Restart	Power on
Name [friendly name] 1	tasmota
	 
AP1 SSID (RSSI)	<ssid> (50%)
Hostname	tasmota-2896-2896
IP-Adresse	<ip>
Gateway	<gateway>
Subnet Mask	255.255.255.0
DNS-Server	<dns>
MAC-Adresse	<mac>
	 
MQTT Host	mqtt.local
MQTT Port	1883
MQTT-User	<user>
MQTT client	tasmota-2896
MQTT topic	tasmota-2896
MQTT group topic	tasmota
MQTT full topic	cmnd/tasmota-2896/
MQTT Fallback-Topic	cmnd/tasmota-2896_fb/
	 
Emulation	none
mDNS-discovery	deactivated
	 
ESP Chip ID	6613840
Flash Chip ID	0x144051
Actual Flash Size	1024kB
Total Flash Size	1024kB
Flash Size Used	457kB
Flash Size Available	544kB
Free Flash Space	23kB
```

If we should not discuss this further, please close the issue and then I'll register to Discord and ask there; but perhaps this might be of interest for other users and/or be helpful (as in feature request) because -at least from my point of view- the documentation does not clearly state how one can submit these changes via mqtt.

It states 

> Commands over MQTT are issued by using `cmnd/%topic%/<command> <parameter>`. If there is no <parameter> (an empty MQTT message/payload), a query is sent for current status of the <command>.

So I would think that sending `15` to `cmnd/tasmota-2896/teleperiod` ought to work. This is just following the example `cmnd/%topic%/<command> <parameter>`.

closed 2019-07-23 by prankousky

meingraham, 2019-07-23:
Actually, `Status  0` has a lot of information we can use - MQTT topic strings, SetOption (19, 59, etc.) settings, etc.

I'm confident that the contributors on Discord can resolve your questions.