arendst/Tasmota #6129: Shelly 1 PM exposed pins

Fibrino, 2019-07-24:
Hello,
I have a Shelly 1 PM device with Tasmota v.6.6.0 firmware.
I would like to know if it is possible to use the pins exposed on the programming/debug header (3V3, GND, GPIO3...) to connect external sensors.

Thanks.

Jason2866, 2019-07-24:
NO, because it is DANGEROUS. You have AC mains on the GPIO pins!
Same as Sonoff POW https://blakadder.github.io/templates/sonoff_Pow.html

ascillato2, 2019-07-24:
Closing this issue as it is **duplicated**. Sorry.

Never connect sensors in power measurement capable devices. You can be electrocuted due to the DC voltage is connected to mains AC. Please, read the warnings of the wiki. 

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-24 by ascillato2