arendst/Tasmota #6134: Ability to use POWER<x> for each configured PWM channel instead of POWER for the whole device

jziolkowski, 2019-07-26:
Currently for multi-PWM devices like RGB controllers the POWER command - as expected - toggles the light. But there are scenarios where per-channel control would be useful, such as (like myself) an H801 controlling 5 separate white LED strips. Having a SetOption that alters this behavior would be really nice. 

The requested behavior can be done using rules and events as a workaround, but "native" command support per each channel would be truly useful.

thedexboy, 2019-07-26:
This would also be useful for custom made boards running Tasmota.

s-hadinger, 2019-07-26:
This is what Channel<x> is supposed to do, with a range of 0..100. 

Don't forget to 'SetOption37 128' to ensure that all 5 pwm can be independently controlled. 

Is it what you are looking for?

jziolkowski, 2019-07-26:
I don't want to re-set the Channel value each time.
I tried SetOption37 but either I don't know how to use it correctly or it doesn't work the way I think it does.

After giving it some more thought maybe it would be enough to allow Channel commands to accept values "ON/OFF" (which currently are converted to 1/0) or "TOGGLE"

jziolkowski, 2019-07-26:
as explained by @Jason2866 on discord, SetOption37 applies only to specific brand of Tuya-based lightbulbs.

s-hadinger, 2019-07-26:
Correct but that's only part of the story. By default when you change Channel1..3 it resets channels 4 & 5. This way you automatically switch between RGB and White.  

Unless SetOption37 >= 128, then channels 1..3 and 4-5 are independent. SetOption37 128 activates this mode without changing anything else.

s-hadinger, 2019-07-26:
Ah I understand better your request, you want Power On/Off on each channel. It should be doable but we need another SetOption boolean to signal you want <n> independent channels. 

@arendst is there a boolean SetOption I could use or could you add one please?

Aldo, I didn't find where states of multiple Relays are stored in Settings. Any hint?

meingraham, 2019-07-26:
@s-hadinger 

https://github.com/arendst/Sonoff-Tasmota/issues/5594#issuecomment-484901489

```
Tasmota keeps score of the relays state in a masked variable called "power". A 1 in a bit location always means a relay turned ON (depending on configured relay as relay1 or relay1i the GPIO is turned low or high). Every change in relay state is only executed after a change in "power". When "power" changes a copy is saved in flash.
```

arendst, 2019-07-26:
Just take the next boolean setoption one as  documented in settings.h

s-hadinger, 2019-07-30:
@jziolkowski You can try this branch to see work in progress.
https://github.com/s-hadinger/Sonoff-Tasmota/tree/temp_multi_chan

`SetOption68 1` enables multiple independent PWM channels.

Numbering of Power<x> is the same as currently, i.e. PWM numbers come just above the relays. This can however lead to inconsistent numbering.

For example, if you have 2 relays and 3 PWM, relays are controlled via `Power1` and `Power2` while PWM are controlled with `Power3`, `Power4`and `Power5`. However levels are controlled via `Channel1`, `Channel2`and `Channel3`. If you don't have relays, numbering is the same between `Power<x>`and `Channel<x>`.

It's still unfinished and not ready for PR:

- `Channel<x>`with non-zero value should power up the PWM if SetOption15
- SetOption15 is currently ignored and will not be completely supported, it would incur too much refactoring
- `Color` command trigger an error, to be fixed
- Philips Hue not yet supported

s-hadinger, 2019-07-31:
New version. I actually aligned numbering of `Power<x>` and `Channel<x>` otherwise it's too confusing.

I completely gave up supporting SetOption20 in multi-channels. There are far too many side effects and it would require to refactor some parts of code. My goal is still to have minimal code impact for this feature.

`Color` is fixed. You can either use `Color` to change all values at once or `Channel<x>` to change one channel only.

Philips Hue still not supported;

meingraham, 2019-07-31:
... and, I'm back... to say... what does the documentation in the wiki need to say with respect to updates/new information, SetOption20, SetOption68, ...

jziolkowski, 2019-07-31:
and what happens when you have both relay and pwm?

s-hadinger, 2019-08-04:
@meingraham It's still a bit too early for wiki update. It should be ok in a few days.

@jziolkowski If you have both relay and pwm, it's basically similar to currently. Relays will get `Power1` - `Power<n>`, and PWM will start after the last Relay: `Power<n+>`

I have a problem though. The Philips Hue emulation is currently limited to `MAX_FRIENDLYNAMES` i.e. `4`. I don't think we should limit to 4 devices, since we could have 2 relays and 5 PWM... The hard limit is 16 devices per Tasmota.

I propose to first use the 4 FriendlyNames as usual, then use the last FriendlyName(4) and append an index (5-16).
@arendst do you have any advice on how to name virtual devices beyond Friendly Name 4?

jziolkowski, 2019-08-04:
I'm confused. You said:
> New version. I actually aligned numbering of Power<x> and Channel<x> otherwise it's too confusing.

but now you say that PWM are added after last relay.

s-hadinger, 2019-08-04:
Let's take an example. Let's say you have 2 Relays and 3 PWMs.

Relay1: `Power1`
Relay2: `Power2`
PWM1: `Power3` and `Channel3`
PWM2: `Power4` and `Channel4`
PWM3: `Power5` and `Channel5`

In the previous approach it would have been: PWM1: `Power3` and `Channel1` which would have been confusing

jziolkowski, 2019-08-04:
Ahhhh ok. Now it makes sense. Thanks.

s-hadinger, 2019-08-05:
By the way, I just found a bug in the current implementation of multiple relays with Hue Emulation, they all inherit the dimmers from the light, instead of being seen as simple on/off bulbs. Will correct it at the same time.

s-hadinger, 2019-08-05:
Release candidate, still in the same branch: https://github.com/s-hadinger/Sonoff-Tasmota/tree/temp_multi_chan

I will test further but it seems to react correctly, including Hue emulation.

@jziolkowski If you can please test.

s-hadinger, 2019-08-06:
All tests are good now. I will push the PR later in the day.

PR #6182 (2019-08-06): Add SetOption68 to enable multi-channel PWM instead of a single light (#6134)
## Description:

Introduced `SetOption68 1` to treat PWM as separate channels instead of a single light. In this mode use `Power<x>` to turn on/off lights, and `Channel<x>` (0..100) to change the value of each channel. `Color <hhhhhh>` still works to set all channels at once.

Numbering of devices is: 1/ First Relays, 2/ then PWMs. `Channel<x>` is aligned to `Power<x>` numbering. For example. Let's say you have 2 Relays and 3 PWMs.

Relay1: `Power1`
Relay2: `Power2`
PWM1: `Power3` and `Channel3`
PWM2: `Power4` and `C [... cut]
closed 2019-08-06 by arendst