arendst/Tasmota #6136: Using a cheap Geeklink IR blaster

kobuki, 2019-07-26:
It would be nice to add support for the "Geeklink Smart Home WIFI+IR+4G Universal Intelligent Remote Controller" ([aliexpress](https://www.aliexpress.com/item/32958748327.html?spm=a2g0s.9042311.0.0.26064c4dBL9SGw)). I have on arriving soon and if possible I plan to use Tasmota on it.

https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster already has a solution, and since Tasmota supports several ESP826x based IR blaster devices, I though it would be able to easily support it.

ascillato2, 2019-07-26:
Hi,

With the template feature of Tasmota, you can configure your device. You will have to test each GPIO to match with your device.

If you need help on this, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-26 by ascillato2

kobuki, 2019-07-26:
It was a feature request that belongs here. Though if you say it's just a matter of finding the proper pins and making a template, I'll do that.

s-hadinger, 2019-07-26:
This one works fine with Tasmota and has a great look 

https://s.click.aliexpress.com/e/AYLhNT6

meingraham, 2019-07-26:
https://blakadder.github.io/templates/geeklink.html

A feature request to add function to the software. To add devices, the Template feature is there... for users to do the work.

kobuki, 2019-07-26:
If I have that bit of information of being able to add the device to Tasmota by simply using templates, I look for that solution without posting here. However it's my mistake I didn't look for an existing template, and about that, my apologies. I just didn't think it's so simple I can use this device with a template. Too bad issues can't be deleted on Github.

@s-hadinger thanks, there are quite a few similar models from China, I chose the cheapest to test it for my needs.

meingraham, 2019-07-26:
Yeah, if you follow installation steps provided on the wiki, it describes how to set up a device, existing (> 350) or never before seen.

[image]

kobuki, 2019-07-26:
I know how to set up a device, I just didn't think it can be done using a template. That's all to it. I compile Tasmota for myself anyway so I'm not exactly a very novice user either, but it was indeed an overlook.

s-hadinger, 2019-07-26:
For all IR blasters, just start with the generic template (18) and find the IRsend and IRrcv pins. Unless you're very unlucky, it should work.

kobuki, 2019-07-26:
@s-hadinger thanks for the hint.

meingraham, 2019-07-26:
`I know how to set up a device` I misunderstood your original post. `It would be nice to add support for the "Geeklink Smart Home WIFI+IR+4G Universal Intelligent Remote Controller"`

- https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#irremote
- https://github.com/arendst/Sonoff-Tasmota/wiki/IRSend-RAW-Encoding
- https://github.com/arendst/Sonoff-Tasmota/wiki/YTF-IR-Bridge
- https://github.com/arendst/Sonoff-Tasmota/wiki/Components#components-list
  - Component 8 & 51

kobuki, 2019-07-26:
I basically needed https://github.com/arendst/Sonoff-Tasmota/issues/6136#issuecomment-515484886 and https://blakadder.github.io/templates/geeklink.html, thanks. Again, it was my overlook.

meingraham, 2019-07-26:
Sounds good.

For future reference, providing the information requested in the GirHub Issue template will aid in readers understanding your request. When you provide no details, it leaves the reader with very little to go on. We do ask for that information consciously.

Having said that, as stated in the issue template, Discord is where we're directing a vast majority of questions... even if eventually a bug or a feature request result (very very small fraction). These kinds of questions are understood more easily and answered more quickly on Discord.

Mike

kobuki, 2019-07-26:
I provided all required information - that is, adding support for a specific device. I chose Feature Request because of that. Filling all the other questionnaire would have just added bloat without actual useful information. OTOH, for a bug report or some request that does require clarification and explanation I usually try to provide that baseline and a lot more I find it contributes to my post.

I do appreciate said guidelines - next time I'll be more careful to look where I forgot in a hurry. Please let us close the discussion, I have nothing to add. Thanks for all the ones helping out with patience. This is a great community with a lot of effort and helpful people. I've been notified of responses of my request within an hour and that in itself is a small wonder. Almost no professional service comes close!

s-hadinger, 2019-07-26:
This should help: https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/236

Pin 5 GPIO14 is the IR Receiver
Pin 14 GPIO6 ? is the IR Transmitter(s)
Pin 12 GPIO10 ? is the red status LED on the front of the unit
Pin 13 GPIO8 ? is the orange L:ED on the front of the unit

meingraham, 2019-07-26:
GPIO8 is reserved for the flash. Pin 13 has to be a different GPIO.

kobuki, 2019-08-01:
I just wanted to report that the device is working fine with my relatively old AC, using my old LIRC config (hex codes converted to bin). I used the template mentioned here. There's some weirdness with received random IR codes that are spammed on the console, I'll look into that later. I'm not sure where the fault lies with that.

A working example:
`IRSend raw,0,8951,4543,593,542,1654,110000010110000000000000010001100000000000000000
`

nagyrobi, 2019-11-03:
On Geeklink IR Blaster GPIO0 Button1 - is actually the pin named "KEY" on the board.
Since Tasmota v7.0.0.2 (tasmota-ir.bin) it is possible to connect a DS18x20 temp sensor to the IR blaster. 
It also seems like the circuit already has a pull-up resistor built-in.
Connect the temp sensor as follows:
- VCC to the 5V pin on the board (this gives 5V when plugged normally to the microUSB source)
- data to the KEY pin on the board
- GND to GND
In Tasmota Module Config just select GPIO0 Button1 as DS18x20(4) and you're done.
You can also try to SetOption74 1 in order to enable ESP chip's internal pull-up resistors.
If not, you can always connect a resistor between 5V and KEY (I tested with 2,2k and also works fine).

https://groups.google.com/forum/#!topic/sonoffusers/9k0D4klj3f8