arendst/Tasmota #6137: problem with static IP command

apas2, 2019-07-26:
hello;
i try to get static IP with HTTP & COMMAND:
> 15:56:40 CMD: IPAddress1 "192.168.1.151"
i get:
15:56:40 MQT: stat/sonoff/RESULT = {"IPAddress1":"0.168.1.151 (192.168.1.63)"}
i did not notice the ZERO starting ;}
I have made : config > restart > my sonoff has disappeared !!!
(i precise CMD are real & direct copies of HTTP)
is this a bug ?
is this misuse ?
regards

ascillato2, 2019-07-26:
Hi,

At commands explanation in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#wi-fi), it shows that the command does not require the ""

try the command as:

```
IPAddress1 192.168.1.151
```

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-26 by ascillato2

apas2, 2019-07-26:
hi
i am not sure, but i think i have try first:
IPAddress1 192.168.1.151
with answer:
unknown command..
I'll get back the sonoff soon my android will be refilled, with direct HTTP.
thanks for answering
may be the command syntax should be corrected ?
(this is not so important, may be i am the only one to make the mistake)
regards

ascillato, 2019-07-26:
It is not going to be changed due to backwards compatibility. It does not require the quotes in the sintax as explained in the wiki. Thanks

apas2, 2019-07-28:
it is not my meaning.
keep the syntax as is (coherent with others rules)
but may be  a change in the interpreter to by pass the " " or reject this kind of bad syntax user (should be safer)
may be there are others commands which fails with the same mistake (unexpected "")
TASMOTA is already a very very smart program, 
many thanks to all the developers and contributors.
regards.