arendst/Tasmota #6138: MQTT flouding with TELE messages

epandi, 2019-07-27:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
Since updating to the latest Tasmota 6.6.0.3 my MQTT broker gets flooded with a lot of tele/+/SENSOR messages from a BW-SHP6 plug. My Teleperiod is set to 300, but I get messages like 4 times a minute.  

09:15:48 CMD: teleperiod
09:15:48 MQT: stat/fleurcam/RESULT = {"TelePeriod":"300"}

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Blitzwolf BW-SHP6
- [x] Tasmota binary firmware version number used : 6.6.0.3 / (thehackbox)
- [x] Development IDE - Compiler / Upload tools used : Thehackbox 2_4_2/2.2.1(cfd48f3)
--

- [x] Provide the output of command ``status 0`` :
```
09:16:24 CMD: status 0
09:16:24 MQT: stat/fleurcam/STATUS = {"Status":{"Module":45,"FriendlyName":["FleurCam"],"Topic":"fleurcam","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:16:24 MQT: stat/fleurcam/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:35:42","StartupUTC":"2019-07-27T05:40:42","Sleep":0,"CfgHolder":4617,"BootCount":8,"SaveCount":29,"SaveAddress":"FA000"}}
09:16:24 MQT: stat/fleurcam/STATUS2 = {"StatusFWR":{"Version":"6.6.0.3(ff909ba-sonoff)","BuildDateTime":"2019-07-26T11:01:54","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
09:16:24 MQT: stat/fleurcam/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","280500000100060000005A00000000000000","000000C9"]}}
09:16:24 MQT: stat/fleurcam/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0FDAE396","001683A0","22B617CD","01001BC0"]}}
09:16:24 MQT: stat/fleurcam/STATUS5 = {"StatusNET":{"Hostname":"fleurcam-1787","IPAddress":"192.168.1.12","Gateway":"192.168.0.1","Subnetmask":"255.255.252.0","DNSServer":"192.168.0.222","Mac":"BC:DD:C2:23:26:FB","Webserver":2,"WifiConfig":2}}
09:16:24 MQT: stat/fleurcam/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.222","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2326FB","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
09:16:24 MQT: stat/fleurcam/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 27 07:16:24 2019","Local":"Sat Jul 27 09:16:24 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:16","Sunset":"21:36"}}
09:16:24 MQT: stat/fleurcam/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:16:24 MQT: stat/fleurcam/STATUS10 = {"StatusSNS":{"Time":"2019-07-27T09:16:24","Epoch":1564211784,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":4,"ApparentPower":32,"ReactivePower":32,"Factor":0.11,"Voltage":237,"Current":0.136}}}
09:16:24 MQT: stat/fleurcam/STATUS11 = {"StatusSTS":{"Time":"2019-07-27T09:16:24","Epoch":1564211784,"Uptime":"0T01:35:42","UptimeSec":5742,"Heap":18,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":866,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxx","BSSId":"28:6C:07:98:xx:xx","Channel":9,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
09:12:36 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:12:36","Epoch":1564211556,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":2,"ApparentPower":33,"ReactivePower":33,"Factor":0.07,"Voltage":238,"Current":0.138}}
09:12:37 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:12:37","Epoch":1564211557,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":5,"ApparentPower":33,"ReactivePower":32,"Factor":0.15,"Voltage":237,"Current":0.138}}
09:12:50 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:12:50","Epoch":1564211570,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":4,"ApparentPower":33,"ReactivePower":32,"Factor":0.12,"Voltage":237,"Current":0.138}}
09:13:13 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:13:13","Epoch":1564211593,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":2,"ApparentPower":31,"ReactivePower":31,"Factor":0.07,"Voltage":237,"Current":0.131}}
09:13:21 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:13:21","Epoch":1564211601,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":2,"ApparentPower":31,"ReactivePower":31,"Factor":0.06,"Voltage":237,"Current":0.129}}
09:13:23 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:13:23","Epoch":1564211603,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":6,"ApparentPower":33,"ReactivePower":32,"Factor":0.17,"Voltage":237,"Current":0.139}}
09:13:51 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:13:51","Epoch":1564211631,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":2,"ApparentPower":33,"ReactivePower":33,"Factor":0.06,"Voltage":238,"Current":0.139}}
09:14:10 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:14:10","Epoch":1564211650,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":6,"ApparentPower":33,"ReactivePower":33,"Factor":0.17,"Voltage":237,"Current":0.140}}
09:14:11 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:14:11","Epoch":1564211651,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":2,"ApparentPower":31,"ReactivePower":31,"Factor":0.07,"Voltage":237,"Current":0.130}}
09:14:12 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:14:12","Epoch":1564211652,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":5,"ApparentPower":31,"ReactivePower":31,"Factor":0.16,"Voltage":237,"Current":0.130}}
09:14:15 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:14:15","Epoch":1564211655,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":2,"ApparentPower":32,"ReactivePower":32,"Factor":0.06,"Voltage":238,"Current":0.135}}
[... cut ...]
09:14:51 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:14:51","Epoch":1564211691,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":2,"ApparentPower":35,"ReactivePower":35,"Factor":0.07,"Voltage":238,"Current":0.147}}
09:14:58 MQT: tele/fleurcam/SENSOR = {"Time":"2019-07-27T09:14:58","Epoch":1564211698,"ENERGY":{"TotalStartTime":"2018-10-30T20:03:26","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":2,"ApparentPower":34,"ReactivePower":34,"Factor":0.05,"Voltage":238,"Current":0.145}}

```
**(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2019-07-27:
Duplicate issue #5559, #5028, #4813, #4130, #4145, #3795, #3778, #3660, #3648...

epandi, 2019-07-27:
Thank you. Didnt find the proper search terms for the issue I guess.

closed 2019-07-27 by epandi