arendst/Tasmota #6140: Amazon echo not working for about 1 day now

ameel, 2019-07-28:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Amazon echo cannot speak to any sonoff-tasmota flashed devices (light bulbs) since yesterday. I have rebooted router and all light bulbs, but amazon echo cannot find the devices. All other smart devices (non sonoff-tasmota) work fine with amazon echo
 

- [Yes ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [Yes ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [Yes ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [Yes ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ No] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [Yes ] Device used (i.e. Sonoff Basic) : Tuya lights flashed with sonoff-tasmota 6.6.0(release-classic)
- [ Yes] Tasmota binary firmware version number used : 6.6.0(release-classic) / (pre-compiled)
- [ Yes] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

02:17:39 CMD: status 0
02:17:39 SRC: WebConsole from 192.168.0.48
02:17:39 RSL: Received Topic /status, Data Size 1, Data 0
02:17:39 RSL: Group 0, Index 1, Command STATUS, Data 0
02:17:39 RSL: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Lounge 2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
02:17:39 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:20","StartupUTC":"2019-07-28T01:17:19","Sleep":50,"CfgHolder":4617,"BootCount":111,"SaveCount":393,"SaveAddress":"FB000"}}
02:17:39 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-classic)","BuildDateTime":"2019-07-06T13:11:57","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
02:17:39 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["The Forbidden Zone",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00028009","280500000100060000000000000000000000","00000000"]}}
02:17:39 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","6010A392","009003A8","200217C6","010000C0"]}}
02:17:39 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1049","IPAddress":"192.168.0.107","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"CC:50:E3:4F:84:19","Webserver":2,"WifiConfig":4}}
02:17:39 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4F8419","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
02:17:39 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Jul 28 01:17:39 2019","Local":"Sun Jul 28 02:17:39 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00"}}
02:17:39 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-07-28T02:17:39"}}
02:17:39 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-07-28T02:17:39","Uptime":"0T00:00:20","Vcc":3.066,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"POWER":"OFF","Dimmer":100,"Color":"255,175,0,100","HSBColor":"41,100,100","Channel":[100,69,0,39],"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"The Forbidden Zone","BSSId":"0C:9D:92:CF:51:30","Channel":2,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
02:17:15 HTP: Restart
02:17:16 SRC: WebGui from 0.0.0.0
02:17:16 UDP: Packet (199)
02:17:16 UDP: Packet (199)
00:00:00 CFG: Loaded from flash at F4, Count 392
00:00:00 SRC: Restart
00:00:00 Project sonoff Lounge 2 Version 6.6.0(release-classic)-2_3_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 The Forbidden Zone in mode 11N as sonoff-1049...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-1049 with IP address 192.168.0.107
00:00:04 UPP: Multicast (re)joined
00:00:04 UDP: Packet (199)
00:00:04 UDP: Packet (199)
00:00:06 UDP: Packet (199)
00:00:06 UDP: Packet (199)
00:00:06 HTP: Main Menu
02:17:26 UDP: Packet (199)
02:17:26 UDP: Packet (199)
02:17:26 UDP: Packet (101)
02:17:26 NTP: Drift 0, (UTC) Sun Jul 28 01:17:26 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
02:17:28 UPP: Hue 3 response packets sent to 192.168.0.3:57281
02:17:29 APP: Boot Count 111
02:17:29 UDP: Packet (199)
02:17:29 CFG: Saved to flash at FB, Count 393, Bytes 3584
02:17:30 UDP: Packet (174)
[... 136 lines cut ...]
02:19:09 UDP: Packet (199)
02:19:09 UDP: Packet (199)
02:19:10 UDP: Packet (199)
02:19:10 UDP: Packet (199)
02:19:10 UDP: Packet (199)
02:19:12 UDP: Packet (199)
02:19:12 UDP: Packet (199)
02:19:13 UDP: Packet (199)
02:19:13 UDP: Packet (199)
02:19:13 UDP: Packet (199)
02:19:18 UDP: Packet (199)
02:19:19 UDP: Packet (199)
02:19:19 UDP: Packet (199)
02:19:21 UDP: Packet (137)
02:19:22 UDP: Packet (199)
02:19:22 UDP: Packet (199)
02:19:22 UDP: Packet (199)
02:19:23 UDP: Packet (174)
02:19:25 UDP: Packet (199)
02:19:25 WIF: Checking connection...
02:19:25 WIF: Connected
02:19:26 UDP: Packet (101)

```
**(Please, remember to close the issue when the problem has been addressed)**

ameel, 2019-07-28:
Solved as follows:
Router was configured with 192.168.0.3 as DMZ. Disabled DMZ and all devices are now detected by amazon echo.

ameel, 2019-07-28:
Closed too soon. Devices aren't responding anymore. Log shows:

03:41:08 UPP: Hue 3 response packets sent to 192.168.0.3:57281
03:41:09 UDP: Packet (199)
03:41:09 UDP: Packet (199)
03:41:12 UDP: Packet (199)
03:41:16 UDP: Packet (199)
03:41:16 UDP: Packet (101)
03:41:17 WIF: Checking connection...
03:41:17 WIF: Connected
03:41:18 UPP: Hue 3 response packets sent to 192.168.0.3:57281
03:41:19 UDP: Packet (199)
03:41:22 UDP: Packet (199)
03:41:22 UDP: Packet (199)
etc

s-hadinger, 2019-07-28:
Your logs look correct. Upnp is rece correctly and answered accordingly. 

Do you confirm 192.168.0.3 is your echo device? What type of Echo device is it?

s-hadinger, 2019-07-28:
I just tried home and Hue emulation works fine via Echo device

ameel, 2019-07-28:
> Your logs look correct. Upnp is rece correctly and answered accordingly.
> 
> Do you confirm 192.168.0.3 is your echo device? What type of Echo device is it?

Hi, thanks for the prompt reply. 192.168.0.3 is a linux computer on my server. I have 2 amazon echo dot version 2 on my network. I monitored the UPP logs for the devices and they're all going to the linux computer instead of the amazon echo. The lights aren't detected at all (maybe detected once every few hours for a minute or so).

Any way to force the devices to the actual echo IP?

s-hadinger, 2019-07-29:
Your network configuration looks uncommon. Tasmota responds to the IP it received the request from. This means that your Linux box is kind of routing upnp packets which it shouldn't be. Are the Tasmota devices and Echo on the same network? What are the IP addresses of the Echo devices?

Keep in mind upnp request is broadcast UDP, response is unicast UDP. The response seems to be dropped.

ameel, 2019-07-29:
Not sure why 192.168.0.3 is sending requests. I'll try to turn it off.

In the mean time, all devices have been working again since last night (15 hours or so). Only change was  the following on my router:
Enable multicast routing (IGMP Proxy) - enabled
Enable efficient multicast forwarding (IGMP Snooping) - enabled.

I can't think of anything else that has changed

s-hadinger, 2019-07-30:
It's very difficult to help you when you give so little information about your network configuration.

This is obviously a local networking problem, not linked to Tasmota. I suggest you close the issue, and reopen in case it is a Tasmota bug or issue.

ameel, 2019-07-30:
thanks for trying to help. I'll close for now until I can narrow down the issue. cheers

closed 2019-07-30 by ameel