arendst/Tasmota #6142: Modify webinterface elements through module (WS2812)

ChristianWaechter, 2019-07-28:
I’m controlling a strip of WS2812 LEDs with a Wemos D1 Mini running Tasmota. This works fine so far. When I select the according WS2812 module in the configuration, a brightness slider and Toggle-Button appears on the tasmote website of the device. I also can change the led pattern/color with some console commands. I now would like to add only a few more buttons to the slider on the tasmota webinterface, so I can quickly change the pattern/color or similar from there. I do not want to add an extra server, if possible… Is there any way to modify this WS2812 module? If so, could you please give me a direction where to edit the source code, as I was not yet able to find the correct spot.

ascillato2, 2019-07-29:
Hi,

The sliders are defined in _xdrv_01_webserver.ino_

Sliders shows up if the device has brightness property or pwm, etc, supported by the hardware selected.

All those other options you ask for are not displayed in the webinterface so as to save flash and ram in order to support more sensors and features, and also because Tasmota is meant to be a MQTT device for home automation and not a standalone device.

closed 2019-07-29 by ascillato2

Chemita23, 2023-07-02:
For you:
https://github.com/Chemita23/WS2812FX_web_GRBW