arendst/Tasmota #6147: BlitzWolf SHP6 EU resets after given setOption32 time has expired

RickPennings, 2019-07-29:
### BUG DESCRIPTION
When a BlitzWolf SHP6 EU is flashed with Tasmota, there are no problems. The moment you change the Configuration -> Configure Module to BlitzWolf SHP (45), after resetting, the module keeps working for 48 seconds, then resets to firmware defaults. Figuring this is the same result as holding a button I changed the setOption32 to a different time setting, and what'd you know, it firmware resets to defaults with the given time.

I'm assuming there is a hold button registered with this configuration module preset, but I can't seem to figure out how to resolve this issue.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : **BlitzWolf SHP6 EU**
- [X] Tasmota binary firmware version number used : Latest release of sonoff.bin / (pre-compiled or self-compiled ?)
- [X] Development IDE - Compiler / Upload tools used : n.a. / esptool.py
- [X] Provide the output of command ``status 0`` : n.a.
```
Not available due to firmware resetting to defaults.

```
- [X] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
No console output available for given issue.

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
BlitzWolf SHP6 EU, flash with sonoff.bin. Register on wifi. Modify Configuration -> Configure Module to BlitzWolf SHP (45). Wait for period of SetOption32 to see firmware reset to default.

### EXPECTED BEHAVIOUR
After changing the Configuration -> Configure Module to BlitzWolf SHP (45), i'd like the module not to reset.

### SCREENSHOTS
n.a.

### ADDITIONAL CONTEXT
n.a.

Jason2866, 2019-07-29:
Please add `status 0` output before you try to change config

RickPennings, 2019-07-29:
@Jason2866 Updated the report.

meingraham, 2019-07-29:
Please add Status 0 of the device in the state where it stops after the reset. Also provide the out of Template and GPIO. 

Thanks

RickPennings, 2019-07-29:
@meingraham Could you explain what you mean with "out of template"? GPIO is configured like the tutorial https://github.com/arendst/Sonoff-Tasmota/wiki/BlitzWolf-SHP6. Do note, I followed the tutorial by the letter, and used the http://thehackbox.org/tasmota/release/sonoff.bin binary.

plampix, 2019-07-29:
The module resets like this after setting the module to the correct one (45, Blitzwolf SHP):

14:06:26 RSL: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
14:06:27 CFG: Use defaults
14:06:27 MQT: Attempting connection...
14:06:28 DNS: Query done. MQTT services found 0
14:06:29 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

00:00:00 Project sonoff Sonoff Version 6.4.1(sonoff)-2_4_2
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-7888 with IP address 192.168.4.1
Status 0
00:00:06 CMD: Status 0
00:00:06 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0
,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:06 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:06","StartupUTC
":"","Sleep":50,"BootCount":0,"SaveCount":4,"SaveAddress":"F8000"}}
00:00:06 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:00:06 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
00:00:06 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00
003BC0"]}}
00:00:06 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7888","IPAddress":"0.0.0.0","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"BC:DD:C2:23:1E:D0","Webserver":2,"WifiConfi
g":4}}
00:00:06 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_231ED0","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:06 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:06 1970","Local":"Thu Jan 01 00:00:06 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:
43","Sunset":"16:03"}}
00:00:06 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:06"}}
00:00:06 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:06","Uptime":"0T00:00:06","Vcc":3.328,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"","BSSId":"FF:FF:FF:FF:FF:FF",
"Channel":1,"RSSI":100}}}

plampix, 2019-07-29:
Oops, I just realized we use an ancient version of the firmware, retrying with version 6.6.0...

plampix, 2019-07-29:
It appears this module does not have a pullup resistor, configuring gpio13 as button1n makes a lot less noise in the logs. This could also be the cause of random resets to default configs because of a false long press detection that takes 10 * (hold time) deciseconds.

meingraham, 2019-07-29:
@plampix 

We ask for `Status 0` because it helps highlight firmware/Core versions, certain settings, etc. The firmware version would have been one of the first red flags to address ;-)

`Template` and `GPIOS` are also Console commands. They display additional device configurations. These also can help highlight strange GPIO assignments.

It seems that prodding for the information has at least given you an aha moment of things to investigate.

RickPennings, 2019-07-29:
@meingraham I tested with the latest version, and reproduced the problem with it. Unfortunately the remark of plampix clouds this report. If I need to present more data I will update the issue stated on top. I will add an attachment with the requested logs and requested command outputs with the states clearly stated during the output of the log.

meingraham, 2019-07-29:
Not doubting your issue exists. We just need more information.

Another possibility is that device models stay the same but the manufacturer actually changes the electronics. Looks exactly the same on the outside and very similar inside, but they alter the electronics to use different energy monitoring chips, controls MCU, even the ESP (8285 vs. 8266). What you may be encountering is related to this.

You may want to start with a "blank" template. First go with no GPIO assigned and choose BASE 45. Then follow the "new device" GPIO setup to "find" the functions of each GPIO. perhaps this will eliminate the bootloops that are resetting the device to defaults.

Mike

RickPennings, 2019-07-29:
@meingraham After flashing it over 40 times to reproduce this problem @plampix tried the following. Before changing to module type 45, he used 0 and changed the GPIO13 to Button1 and GPIO15 to relay1. This reproduced the same issue. After erase_flash and write_flash of sonoff.bin the problem did not occur anymore. This would mean that either the modification of @plampix actually has been stored somewhere, where erase_flash and write_flash have not been writing, or it might have shorted something in the plug itself.

I don't know how, but the problem is not reproducible anymore. I'm going to order another plug to check if I can reproduce it on that one. For the time being, I'm closing the issue.

Thanks all for the support!

closed 2019-07-29 by RickPennings

meingraham, 2019-07-29:
Ah, good old remnants in the flash.

BTW, you have used the (45) `Module` which sets the device GPIO "in code". And it appears that is still correct for the SHP6 now that you've swept out any gremlins from the flash. What I was suggesting was using the Template to assign ALL GPIO yet still make use of the special energy monitoring code for (45). See the wiki for Templates. Since 6.4.1.16 most devices will not be added as Modules unless they are truly unique. Templates offer much more flexibility and re-usability. There are over 350 devices supported using templates.

Anyway, glad it's all working now.

Mike

RickPennings, 2019-07-30:
Problem occurred again. Going to try your last comment @meingraham, I'm guessing that it'll solve the problem. Thanks for the tip. When I'll get it working I'll send an addition for the guide for this device.