arendst/Tasmota #6149: Will this switch work with Tasmota?

ShonP40, 2019-07-30:
Will this switch work with Tasmota?
https://www.amazon.com/dp/B07P2KMNQM/ref=cm_sw_r_cp_api_i_s-cqDbNY41GDS

ascillato2, 2019-07-30:
It is very difficult to say if it will or not without seeing the internal board. Sorry.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-07-30 by ascillato2

ShonP40, 2019-07-30:
So can someone suggest me a smart touch WiFi switch that supports Tasmota and can control electric blinds? (up and down).