arendst/Tasmota #6152: Cannot set Width2 for seconds to 0

robin13, 2019-07-31:
### BUG DESCRIPTION

Cannot disable the second hand of the clock.  I tried this by setting `Width2 0` which should be possible and should disable based on the documentation.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Wemos D1 mini
- [x] Tasmota binary firmware version number used : 6.6.0.3 self-compiled
- [x] Development IDE - Compiler / Upload tools used : platformio / espytool
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
08:44:39 CMD: status 0
08:44:39 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:44:39 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:01:07","StartupUTC":"2019-07-31T06:43:32","Sleep":50,"CfgHolder":4617,"BootCount":4,"SaveCount":26,"SaveAddress":"FA000"}}
08:44:39 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.3(sonoff)","BuildDateTime":"2019-07-30T09:48:11","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
08:44:39 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":5,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["cundr_heim",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005A00000000000000","00000000"]}}
08:44:39 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":547,"Free":456,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FCAE396","041683A1","22B617CD","01001BC0"]}}
08:44:39 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2193","IPAddress":"10.10.10.124","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer":"10.10.10.1","Mac":"2C:F4:32:12:C8:91","Webserver":2,"WifiConfig":4}}
08:44:39 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"hassio.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_12C891","MqttUser":"hassio","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
08:44:39 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Jul 31 06:44:39 2019","Local":"Wed Jul 31 08:44:39 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+02:00","Sunrise":"06:22","Sunset":"21:31"}}
08:44:39 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-07-31T08:44:39","Epoch":1564555479}}
08:44:39 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-07-31T08:44:39","Epoch":1564555479,"Uptime":"0T00:01:07","UptimeSec":67,"Heap":23,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":565,"POWER":"ON","Dimmer":100,"Color":"FFFFFF","HSBColor":"0,0,100","Channel":[100,100,100],"Scheme":5,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"cundr_heim","BSSId":"F0:9F:C2:F7:C1:63","Channel":6,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [x] Provide the output of console when you experience your issue if apply :

```
15:27:25 CMD: Width2 0
15:27:25 MQT: stat/sonoff/RESULT = {"Width2":1}
```

### TO REPRODUCE

When trying to disable the second "hand" of the clock by setting the pixel width to zero:

```
15:27:25 CMD: Width2 0
15:27:25 MQT: stat/sonoff/RESULT = {"Width2":1}
```

It is ignored and the current width is echoed.

Setting to a number larger than 1 works:

```
15:26:28 CMD: Width2 3
15:26:28 MQT: stat/sonoff/RESULT = {"Width2":3}
```

### EXPECTED BEHAVIOUR

I would expect:

```
15:27:25 CMD: Width2 0
15:27:25 MQT: stat/sonoff/RESULT = {"Width2":0}
```

arendst, 2019-07-31:
From the code I see only Width1 could be set to 0 which in turn would mean the width equals all leds.

Other widths start at 1 until 32.

I'll look for a way to add your expectation which btw will need more coding than just adding the test for zero as input...

robin13, 2019-07-31:
Should we change the wiki documentation in the mean time then so that the documentation matches the code?

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

meingraham, 2019-07-31:
wiki - done

ascillato2, 2019-07-31:
Closing issue as the enhancement was added by Theo. Thanks :+1:

closed 2019-07-31 by ascillato2

robin13, 2019-07-31:
Sweet - I just tested and it works.  Thank you for the phenomenally fast fix @arendst :) :)

robin13, 2019-07-31:
I also just updated the wiki to reflect the new `0..32` range.