arendst/Tasmota #6156: Gosund RGB+W configuration changes back to Sonoff Basic

holgerlembke, 2019-07-31:
I'm fishing a little bit in the dark.... I have no idea, what is happening beside the effects I see.

I have a Gosund RGB+W LED lamp and use this template: 

https://blakadder.github.io/templates/gosund_e27_rgbw.html

I use two mqtt commands 
''' 
// javascript
var poweronoff = "cmnd/%name%/power"; //   on|off
var ledcolor = "cmnd/%name%/color";
''' 
to set, well, power and LED colors... Works nicely. 

But sometimes at random (I don't see a pattern, next day when I continue to work on my html/js code or so) the configuration changed to Sonoff basic.

Any ideas?

Updated from 6.6.0 to 6.6.03, no change.

meingraham, 2019-07-31:
Start here:
- Fill in the requested template
- Address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-07-31:
Going back to sonoff basic could be for:

* bootloop protection (if the device hungs at boot before having 10 seconds of uptime and that is repeated several times in a row - or the device was powered and let without power in less than 10 seconds and that was done several times in a row)
* your button was pressed for more than 40 seconds
* flash problem

closed 2019-07-31 by ascillato2