arendst/Tasmota #6160: Shelly 2.5 - dual power metering

tobby88, 2019-08-01:
I stumbled upon this: https://shelly.cloud/shelly-25-wifi-smart-relay-roller-shutter-home-automation/

"DUAL POWER METERING
Shelly 2.5 has two integrated precise power meters. You can measure the consumption for each channel separately."

With Tasmota it only shows one power metering (which seems to be a sum of both outputs?).

It would be nice if Tasmota supports both measurings seperately.

arendst, 2019-09-14:
How about this:

[image]

```
16:24:49 MQT: tele/shelly25/SENSOR = {"Time":"2019-09-14T16:24:49","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":52.5},"ENERGY":{"TotalStartTime":"2019-04-13T18:21:03","Total":3.262,"Yesterday":0.000,"Today":0.005,"Period":1.34,"Power":[23.63,67.17],"ApparentPower":[23.63,67.17],"ReactivePower":[0.00,0.00],"Factor":[1.00,1.00],"Voltage":229.3,"Current":[0.103,0.293]},"TempUnit":"C"}
```

KaiserSoft, 2019-09-14:
Very nice, thanks!

Is it possible to get the energy totals per channel as well or does the device only track the combined total?
Sry, already flashed Tasmota on mine so I can't check the original Shelly software anymore.

arendst, 2019-09-15:
For now with consolidated Energy totals.

digiblur, 2019-09-15:
Nice!  Looking forward to it with the Shelly EM as well (same power chip)

arendst, 2019-09-16:
Did you try Shelly EM with template of Shelly 2.5?
```
{"NAME":"Shelly 2.5","GPIO":[56,255,17,255,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
```

digiblur, 2019-09-16:
I haven't yet but definitely bumping it up in my list for this week to test.

meingraham, 2019-09-16:
@arendst 

How does Domoticz handle JSON "arrays"? In this `SENSOR` message, 
```
{"Time":"2019-09-14T16:24:49","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":52.5},"ENERGY":{"TotalStartTime":"2019-04-13T18:21:03","Total":3.262,"Yesterday":0.000,"Today":0.005,"Period":1.34,"Power":[23.63,67.17],"ApparentPower":[23.63,67.17],"ReactivePower":[0.00,0.00],"Factor":[1.00,1.00],"Voltage":229.3,"Current":[0.103,0.293]},"TempUnit":"C"}
```

I'm having trouble setting up a JSON function to return a separate value for each channel of any particular metric. For example, `"Power":[23.63,67.17]`, I want to define an Item to extract Power1 and another to extract Power2. Right now the only thing I can return directly is `[23.63,67.17]` which I then have to parse and convert to extract each value.

I grasp how I would do it if it were formatted like  
`"Power":{"Channel1":23.63 ,"Channel2":67.17}` but `[23.63,67.17]` now requires my own logic in my hub (openHAB) to process.

Your insights much appreciated ;-)

Mike

arendst, 2019-09-16:
- Domoticz default returns only phase/channel 1 info using the inbuilt Domoticz parameters.
- As an alternative a rule could be used to send the info to Domoticz.

I haven't tested rules using a JSON list but if it doesn't work I/someone have/has to find a solution.

meingraham, 2019-09-16:
@digiblur 

When you get around to testing this, can you see if HA can extract the individual values from the JSON payload?

Otherwise, @arendst, might I suggest this `"Power":{"Channel1":23.63 ,"Channel2":67.17}` or this `"Power":{"Power1":23.63 ,"Power2":67.17}` rather than this `"Power":[23.63,67.17]`? This way, I think most JSON functions can grab each value using an expression such as `$.ENERGY.Power.Power1` or `$.StatusSNS.ENERGY.Power.Power1` (e.g., from a `STATUS8`/`STATUS10` payload).

Mike

arendst, 2019-09-16:
Nice suggestion but it takes too much space. When three phase devices are used it needs three value pairs for each power, voltage and current item. This makes the JSON string too long for both MQTT response and rule/script handling.

JSON lists are handled succesfully in other tools so I guess we have to find a way to make it work with rules too.

meingraham, 2019-09-16:
Time for the Rules/Scripting caped crusaders (@arendst @laurentdong @gemu2015) to come to the rescue :wink:

effelle, 2019-09-16:
Quick test with `value_json` under Home Assistant show me is possible to split the value using an index. Not sure if I can do the same on OH `Jsonpath`.
```
{% set value_json = {"Time":"2019-09-14T16:24:49","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":52.5},"ENERGY":{"TotalStartTime":"2019-04-13T18:21:03","Total":3.262,"Yesterday":0.000,"Today":0.005,"Period":1.34,"Power":[23.63,67.17],"ApparentPower":[23.63,67.17],"ReactivePower":[0.00,0.00],"Factor":[1.00,1.00],"Voltage":229.3,"Current":[0.103,0.293]},"TempUnit":"C"} %}

{{ value_json.ENERGY.Current[0] } = 0.103
{{ value_json.ENERGY.Current[1] }} = 0.293
```
No idea how to filter the values with rules.

effelle, 2019-09-16:
Jsonpath can use index too.
`$.ENERGY.Current[0]` = 0.103 
`$.ENERGY.Power[1]` = 0.293

gemu2015, 2019-09-16:
should already work in scripter e.g.

var=ENERGY#Power[1]

effelle, 2019-09-16:
So what we miss is handling with normal rules?

arendst, 2019-09-16:
So cut (scripter) and paste (rule) should solve it.

meingraham, 2019-09-16:
Yeah, stupid me used [1] and [2]. I thought it was unsupported syntax in OH when it was PEBCAK!

tobby88, 2019-09-18:
> How about this:
>
> [image]
>
> ```
> 16:24:49 MQT: tele/shelly25/SENSOR = [...]

Looks nice :) Small improvement: I would suggest to add the unit after every value, not just at the end, but maybe that's a matter of taste.

To seperate the energy as suggested by @KaiserSoft would be very nice to :)

ascillato2, 2019-10-02:
Closing this issue as the feature has been added. Thanks :)

closed 2019-10-02 by ascillato2

cmiguelcabral, 2020-03-06:
I know this issue is already closed, but I'm looking for a similar thing that is having Totals also split.
Home Assistant's Utility Meter uses the total energy, giving that, I'm unable to know how much energy each machine (connected to a different output) is consuming.
Any plat to add that??
Thanks

umairyounus, 2021-08-16:
Any way we can get a split total?

Y-V-L, 2021-08-28:
What about the Split of the Totals. This is also needed for the Shelly EM and Shelly EM3 if you will be able to count the imports and exports separately ...

ascillato, 2021-09-29:
Splitted totals added to Tasmota by Theo.

https://github.com/arendst/Tasmota/commit/d3fc03d3980c25cd8b66063405bcab12adadaff5