arendst/Tasmota #6166: %var% Not working with Scale

bronzeblood, 2019-08-01:
### BUG DESCRIPTION
Using %var1% with Scale is not working.  e.g. SCALE2 %var1%, 85, 100, 0, 100

%var1% is ignored and 0 is used.

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Device used (i.e. Sonoff Basic) : D1 Mini Pro
- [ x] Tasmota binary firmware version number used : _6.6.0.2 self-compiled ?)
- [ x] Development IDE - Compiler / Upload tools used : Vscode /Platformio
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

23:53:18 CMD: status 0
23:53:18 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:53:18 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:26","StartupUTC":"2019-08-01T22:51:52","Sleep":50,"CfgHolder":4617,"BootCount":2,"SaveCount":6,"SaveAddress":"F6000"}}
23:53:18 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019-07-06T14:26:16","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
23:53:18 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
23:53:18 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":557,"Free":444,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
23:53:18 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6867","IPAddress":"192.168.4.186","Gateway":"192.168.4.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"BC:DD:C2:24:9A:D3","Webserver":2,"WifiConfig":4}}
23:53:18 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_249AD3","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:53:18 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Aug 01 22:53:18 2019","Local":"Thu Aug 01 23:53:18 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:23","Sunset":"20:29"}}
23:53:18 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-08-01T23:53:18"}}
23:53:18 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-08-01T23:53:18","Uptime":"0T00:01:26","Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IOT","BSSId":"A2:2A:A8:D4:BB:91","Channel":11,"RSSI":64,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```

### TO REPRODUCE

23:52:07 CMD: var1 89
23:52:07 RSL: stat/sonoff/RESULT = {"Var1":"89"}
23:52:12 CMD: SCALE2 89, 85, 100, 0, 100
23:52:12 RSL: stat/sonoff/RESULT = {"Scale2":"26.667"}
23:52:21 CMD: SCALE2 %var1%, 85, 100, 0, 100
23:52:21 RSL: stat/sonoff/RESULT = {"Scale2":"-566.667"}

meingraham, 2019-08-01:
I can confirm the same behavior with 6.6.0.3

ascillato, 2019-08-01:
Sorry, %var1% is for rules. Using in that way will not work.

ascillato2, 2019-08-01:
Example of use inside a rule:

```
rule1 1
rule1 on var1#state do SCALE2 %var1%, 85, 100, 0, 100 endon

```
everytime you use the command VAR1 89 in the console, that rule will be executed storing the result in VAR2

closed 2019-08-01 by ascillato2