arendst/Tasmota #6169: A timer is sometimes not executed.

Michael33X, 2019-08-02:
Last night a switch-off timer was not triggered.

I use
- Tasmota Version 6.6.0(release-sonoff)
- Build 2019.07.06 13:17:19
- binary download: - sonoff-DE.bin, 517.008 Bytes
- on an OBI Socket 2 Modul

I use the module for lighting with random function as a presence simulation. Last night a timer (switch-off) was not executed.

Here's what I did:

I have programmed a timer with the following parameters:
{"Timer2":{"Arm":1,"Mode":0,"Time":"00:00","Window":15,"Days":"1111111","Repeat":1,"Output":1,"Action":0}}
This means the light should be switched off at midnight +/- 15 minutes.

During the last few days the timer event was triggered twice. Once with an offset of -13 minutes and once with 0 minutes. But last night the timer **was not triggered** and the light stayed on all night.

Could it be that the problem is related to the date change or the positive offset or the combination of both? Maybe, the problem will not occur if I program the timer to 00:20, for example?

I found a location in the source code that might be related to the problem.
**..\sonoff\xdrv_09_timers.ino:**

`void TimerEverySecond(void)`
`{`
`...`
`if (xtimer.arm) {`
`set_time += timer_window[i];                // Add random time offset`
`if (set_time < 0) { set_time = 0; }         // Stay today;`
`if (set_time > 1439) { set_time = 1439; }`

I think the last line should be instead:
`if (set_time > 1439) { set_time = set_time - 1440; }`

But if `set_time > 1439` never occurs, then there is no error here.

If someone has a solution to the problem, thank you in advance.

ascillato2, 2019-08-02:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Also, you will need to set a SYSLOG with loglevel 4 in order to catch what is really going on. Could be that your device have restarted just before the trigger or something like that. Also, depending on your WIFICONFIG, Tasmota could restart in case of wifi lost.

I use the exact same config than you for more than a year and never had a missed trigger.

As the timers are triggers, in the case that your device loose power BEFORE the trigger and recover power AFTER the trigger, Tasmota will miss it. For that case, you will need a rule to check time at boot and do the proper automations like:

```
rule1 1
rule1 on time#initialized do backlog event checksunrise=%time%; event checksunset=%time% endon on event#checksunset>%sunset% do power1 1 endon on event#checksunrise<%sunrise% do power1 1 endon
```

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-02 by ascillato2

arendst, 2019-08-02:
I think the problem is more in the definition of set_time. It's currently defined as uint16_t which doesn't allow for negative values while there may be negative values due to random window.

In the next update this will be changed to int32_t

Michael33X, 2019-08-04:
Hello,

thank you so much for taking care of my request.

@ascillato2: Sorry, I'm new here at Github and didn't know how to handle templates.

@arendst:
In the meantime I didn't change the setup of my ESP and continued to watch.
Here is the result: Timer2 was not triggered last night.

Since I have observed that the timer is triggered at negative offset and offset=0, I suspect that the error occurs only at positive offset. There are no peculiarities in the log: no reset and no wifi reconnect, as ascillato2 suspected. Everything runs ok only the timer2 is missing.

For your information here the complete setup of my module as well as a summery from the weblog Level4..:

**// Parameters that differ from original default settings.**

Device: OBI Soket 2 (61)

PowerOnState 0
FriendlyName1 Port02
NtpServer1 192.168.14.254  // router, own ntpserver, internettraffic has been blocked by firewall.
NtpServer2 0
NtpServer3 0
SerialLog 0
Timezone 99
WebLog 4
Hostname Port02
TelePeriod 3600      // too many log entries
Latitude XX.XXXX    // my grographic coordinates
Longitude Y.YYYY
SetOption0 0
SetOption3 0

{"Timer1":{"Arm":1,"Mode":2,"Time":"00:45","Window":0,"Days":"1111111","Repeat":1,"Output":1,"Action":1}}
{"Timer2":{"Arm":1,"Mode":0,"Time":"00:00","Window":15,"Days":"1111111","Repeat":1,"Output":1,"Action":0}}

//Timer3 to Timer16 are default unset:

**// Weblog 03.08.2019**

21:58:44 WIF: Prüfe Verbindung...
21:58:44 WIF: verbunden
21:59:00 SRC: Timer
21:59:00 RSL: RESULT = {"POWER":"ON"}  // sunset = 21:14 + 45 min, ok
21:59:00 RSL: POWER = ON
21:59:04 WIF: Prüfe Verbindung...
21:59:04 WIF: verbunden
...
22:28:48 WIF: verbunden
22:28:49 RSL: STATE = {"Time":"2019-08-03T22:28:49","Uptime":"0T22:03:18","Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"K....U","BSSId":"E0:...:53","Channel":13,"RSSI":98,"LinkCount":4,"Downtime":"0T05:47:38"}}
22:29:08 WIF: Prüfe Verbindung...
...
23:28:48 WIF: verbunden
23:28:49 RSL: STATE = {"Time":"2019-08-03T23:28:49","Uptime":"0T23:03:18","Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"K....U","BSSId":"E0:...:53","Channel":13,"RSSI":98,"LinkCount":4,"Downtime":"0T05:47:38"}}
23:29:08 WIF: Prüfe Verbindung...
...
23:41:48 WIF: Prüfe Verbindung...
23:41:48 WIF: verbunden
...
00:01:46 NTP: Drift 0, (UTC) Sat Aug 03 22:01:46 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
00:01:48 WIF: Prüfe Verbindung...
00:01:48 WIF: verbunden
...
00:14:28 WIF: Prüfe Verbindung...
00:14:28 WIF: verbunden
00:14:48 WIF: Prüfe Verbindung...
00:14:48 WIF: verbunden
00:15:08 WIF: Prüfe Verbindung...
00:15:08 WIF: verbunden
00:15:28 WIF: Prüfe Verbindung...
00:15:28 WIF: verbunden

Michael33X, 2019-08-04:
Additional note.
Timer2 has triggerd with offset -1 minute.
04.08.2019
23:59:00 RSL: RESULT = {"POWER":"OFF"}
23:59:00 RSL: POWER = OFF

s-hadinger, 2019-08-05:
Are you using the latest dev version. It should not be possible to trigger anymore at 23:59 with Theo’s latest patch, at least not with your timer. 

Keep in mind that the current implementation cannot cross the midnight boundary, or it would require substantial refactoring. Any negative value will be replaced with 00:00 and any value above 23:59 will be replaced with 23:59. 

You should rather have something like “Time”:”00:10” and “Window”:10

Michael33X, 2019-08-05:
@s-hadinger 

I use Tasmota Version 6.6.0(release-sonoff) Build 2019.07.06 13:17:19, see above.

> "Any negative value will be replaced with 00:00 and any value above 23:59 will be replaced with 23:59."

That's not right.

According to my observations, with Timer2: "Time":"00:00" and "Windows":15, that's the way it is:

**Offset / Action**

**negativ** / The timer is executed correctly, at 23:51 for example
**Zero** / The timer is executed correctly at 00:00.
**positiv** / The timer **is not executed** at all !!

**Only positive offsets do not work.** But I've posted that countless times now. Didn't you read those?

s-hadinger, 2019-08-05:
Precisely. Theo fixed 3 days ago the issue with negative offsets that would trigger **the wrong day**. This is why sometimes it is triggered twice and sometimes not.

Since your build is 2019.07.06, it does not include this patch.

Please install the latest version.

Michael33X, 2019-08-07:
@s-hadinger 

Thank you very much for your information. May I ask two more questions?

1. "Please install the latest version". Does it mean compiling the develepment code my myself? This is not possible for me without errors using arduino IDE as described. Or where can I get the latest bin?

2. I am in the process of reading and understanding the source code. Which IDE would you recommend: Visual Source Code, Visual Studio? Especially I would like to have a working debugger like JTAG. Maybe Visual GDB or Visual Micro is a febug solution? Or do you debug via serial console like Serial.print() for example.?

Which programming environment do you use for develepment?

ascillato, 2019-08-07:
@Michael33X 

From wiki:

1- https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisites#tasmota-firmware

Latest precompiled versions are at: http://thehackbox.org/tasmota/release/020502/sonoff.bin

Flashing precompiled bins: https://github.com/arendst/Sonoff-Tasmota/wiki/Flashing#esptoolpy

2- https://github.com/arendst/Sonoff-Tasmota/wiki/Flashing#flashing-and-compiling-from-source

Visual Studio Code with Platformio is the best (https://platformio.org/platformio-ide)

Help and general questions are better to be issued in the Tasmota Support Chat (https://discord.gg/Ks2Kzd4)

meingraham, 2019-08-07:
[Gitpod](https://github.com/arendst/Sonoff-Tasmota/wiki/Compiling-Tasmota-on-Gitpod) uses VSC & PlatformIO

Michael33X, 2019-08-07:
Thanks a lot,

That means, there is no debugger for the ESP8266 or ESP-WROOM-02? No JTAG or anything? How do you debug your code? How the hell are you debugging your code?

ascillato, 2019-08-07:
> How the hell are you debugging your code?

Please calm down.

Just add logs to the console as all drivers has.