arendst/Tasmota #6170: Is WiFi password essential in my_user_config.h file

eugeneshortall, 2019-08-02:
I am trying to flash Sonoff devices, they flash ok but I do not get the follow-on access.
I use and open network in my house as we are far from anything. I know that certain systems require a network password. 

Is a password essential in  the  my_user_config.h file, if not how do I indicate "no password"

Thanks,

ascillato2, 2019-08-02:
As Tasmota is meant for home automation, blank passwords are not allowed. Sorry.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-02 by ascillato2

eugeneshortall, 2019-08-02:
Thank you for the clarification
Eugene