arendst/Tasmota #6171: `DisplayMode 0` shows 128...

foxsam21, 2019-08-02:
### BUG DESCRIPTION
`DisplayMode 0` shows 128...
I brought this up on chat and was helped (alot.. Thanks guys) by @mike2nl  and @meingraham and I was sent here to report this issue.

I ran esptool and erased flash before anything. Then I flashed a precompiled binary from: `http://thehackbox.org/tasmota/020502/sonoff-display.bin` and the same with`2.4.2` core.
I set up the module like this:
```
friendlyname oled
topic oled_test
ssid1 XXXXXXXXXX
password1 XXXXXXXXXX
mqtthost XXXXXXXXXX

module 18
gpio4 06
gpio5 05
gpio16 162

power1 on

displaymodel 2
displaymode 1
```
Like such the correct time and date is displayed.
But when I try `displaymode 1` it does not work and does not restart. I get this:
```
14:41:46 CMD: displaymode 0
14:41:46 MQT: stat/oled_test/RESULT = {"DisplayMode":128}
```
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : `wemos D1`
- [x] Tasmota binary firmware version number used : `6.6.0.3 / pre-compiled`
- [x] Development IDE - Compiler / Upload tools used : `esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 <binary>`
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

14:46:06 MQT: stat/oled_test/STATUS = {"Status":{"Module":18,"FriendlyName":["oled"],"Topic":"oled_test","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:46:06 MQT: stat/oled_test/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:13","StartupUTC":"2019-08-02T13:39:53","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":22,"SaveAddress":"F6000"}}
14:46:06 MQT: stat/oled_test/STATUS2 = {"StatusFWR":{"Version":"6.6.0.3(bea3174-display)","BuildDateTime":"2019-08-02T00:07:24","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
14:46:06 MQT: stat/oled_test/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXXXXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005A00000000000000","00000000"]}}
14:46:06 MQT: stat/oled_test/STATUS4 = {"StatusMEM":{"ProgramSize":548,"Free":452,"Heap":16,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"XXXXXX","FlashMode":3,"Features":["00000809","8FAAA39F","00368FA0","22B617C1","010008C0"]}}
14:46:06 MQT: stat/oled_test/STATUS5 = {"StatusNET":{"Hostname":"oled_test-4468","IPAddress":"XXXXXXXXXX","Gateway":"XXXXXXXXXX","Subnetmask":"255.255.255.0","DNSServer":"XXXXXXXXXX","Mac":"XXXXXXXXXX","Webserver":2,"WifiConfig":4}}
14:46:06 MQT: stat/oled_test/STATUS6 = {"StatusMQT":{"MqttHost":"XXXXXXXXXX","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_72B174","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
14:46:06 MQT: stat/oled_test/STATUS7 = {"StatusTIM":{"UTC":"Fri Aug 02 13:46:06 2019","Local":"Fri Aug 02 14:46:06 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:24","Sunset":"20:28"}}
14:46:06 MQT: stat/oled_test/STATUS10 = {"StatusSNS":{"Time":"2019-08-02T14:46:06","Epoch":1564753566}}
14:46:06 MQT: stat/oled_test/STATUS11 = {"StatusSTS":{"Time":"2019-08-02T14:46:06","Epoch":1564753566,"Uptime":"0T00:06:13","UptimeSec":373,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"XXXXXXXXXX","BSSId":"XXXXXXXXXX","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
It's in the bug report on top.
```

### TO REPRODUCE
Reflash, setup module and try `displaymode 0`

### EXPECTED BEHAVIOUR
`displaymode 0` should return:
```
14:51:06 CMD: displaymode 0
14:51:06 MQT: stat/oled_test/RESULT = {"DisplayMode":0}
14:51:07 APP: Restarting
```

arendst, 2019-08-02:
Reproduced. Digging....

arendst, 2019-08-02:
Pls verify.

foxsam21, 2019-08-02:
OK. Thanks. it is fixed and works...
One thing missing from what I did that is needed for my particular model is to set `{"DisplayHeight":32}` If not the text will be unreadable.

closed 2019-08-02 by foxsam21