arendst/Tasmota #6172: Web command using user and password not working

PitySOFT, 2019-08-02:
Using Tasmota 6.5.0 on a device and I can't get over the user and password authentication to set a command via web url.

I keep getting the message WARNING: "Need user=\<username\>&password=\<password\>" even if the user and password are correct.

I see that on [`line 2157`](https://github.com/arendst/Sonoff-Tasmota/blob/4a8c7b347397ab6755a0f2c7486f3a75d53a673c/sonoff/xdrv_01_webserver.ino#L2157) for the user length is used the length of password.

I change that line to this code `char tmp1[sizeof(WEB_USERNAME)];` and I still get the same error.

What's the problem?

ascillato2, 2019-08-03:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

PitySOFT, 2019-08-04:
### BUG DESCRIPTION
I keep getting the message WARNING: "Need user=<username>&password=<password>" even if the user and password are correct.

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Basic
- [x] Tasmota binary firmware version number used : 6.5.0 / self-compiled 
- [x] Development IDE - Compiler / Upload tools used : AdruinoIDE / AdruinoIDE OTA
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
17:57:34 MQT: stat/sw/curtebunicu/STATUS = {"Status":{"Module":1,"FriendlyName":["sw-curtebunicu"],"Topic":"curtebunicu","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
17:57:34 MQT: stat/sw/curtebunicu/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"1T17:24:12","StartupUTC":"2019-08-02T21:33:22","Sleep":50,"CfgHolder":4620,"BootCount":5,"SaveCount":21,"SaveAddress":"F7000"}}
17:57:34 MQT: stat/sw/curtebunicu/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019-08-03T00:32:45","Boot":4,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
17:57:34 MQT: stat/sw/curtebunicu/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":2,"LogHost":"10.0.0.5","LogPort":514,"SSId":["IoT",""],"TelePeriod":600,"Resolution":"558180C0","SetOption":["0000A069","1E0500000100000000000000000000000000","00000001"]}}
17:57:34 MQT: stat/sw/curtebunicu/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","0F0000B0","00000000","00800096","000000C0"]}}
17:57:34 MQT: stat/sw/curtebunicu/STATUS5 = {"StatusNET":{"Hostname":"sw-curtebunicu","IPAddress":"10.0.10.21","Gateway":"10.0.10.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.10.1","Mac":"5C:CF:7F:95:EB:CC","Webserver":2,"WifiConfig":5}}
17:57:34 MQT: stat/sw/curtebunicu/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_95EBCC","MqttUser":"esp","MqttCount":6,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:57:34 MQT: stat/sw/curtebunicu/STATUS7 = {"StatusTIM":{"UTC":"Sun Aug 04 14:57:34 2019","Local":"Sun Aug 04 17:57:34 2019","StartDST":"Sun Mar 31 03:00:00 2019","EndDST":"Sun Oct 27 04:00:00 2019","Timezone":99,"Sunrise":"06:04","Sunset":"20:36"}}
17:57:34 MQT: stat/sw/curtebunicu/STATUS10 = {"StatusSNS":{"Time":"2019-08-04T17:57:34"}}
17:57:34 MQT: stat/sw/curtebunicu/STATUS11 = {"StatusSTS":{"Time":"2019-08-04T17:57:34","Uptime":"1T17:24:12","Vcc":3.513,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":56,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IoT","BSSId":"7A:8A:20:71:53:89","Channel":11,"RSSI":46,"LinkCount":6,"Downtime":"0T00:00:36"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
Try to execute a command using the web URL and have a web password set in Tasmota settings.

### EXPECTED BEHAVIOUR
See the results of the command

curzon01, 2019-08-04:
unfortunately you didn't provide info about how you get your URL (browser, curl, wget, others) and how you provide the credentials 

Anyway pls note to code special characters in password (eg %$#...) using URL encoding with % sign, see https://stackoverflow.com/questions/1856785/characters-allowed-in-a-url

PitySOFT, 2019-08-04:
The call is made from Crome using this url http://10.0.10.1/cm?user=admin&password=password%239&cmnd=status%207

My password is password#9

curzon01, 2019-08-05:
can not be reproduced:

http://devesp08/cm?user=admin&password=password%239&cmnd=status%207

returns a valid JSON on all my test retrieves (Chrome 73, Firefox 68.01, curl, wget) using pre-build Tasmota 6.5.0/ 6.6.0 sonoff.bin/sonoff-display.bin and own compiled 6.5.0 and 6.6.0 releases (same tested with sonoff.bin/sonoff-display.bin) whereas

http://devesp08/cm?user=admin&password=password%23**x**&cmnd=status%207

returns the expected {"WARNING":"Need user=<username>&password=<password>"}.

Pls use a pre-build release (http://thehackbox.org/tasmota/release/), then you can try compile your own (first without changes in source code, only change my_user_config.h if needed)

PitySOFT, 2019-08-05:
I flashed the device with version 6.6.0 same settings as before and now I confirm it works. No idea why was not working with 6.5.0

closed 2019-08-05 by PitySOFT