arendst/Tasmota #6173: DimmerOnState

briansmith7, 2019-08-03:
Hello, 

I think it would be helpful to have a DimmerOnState to instruct Tasmota to reset Dimmer variable to a certain default value (e.g. 100%) every time device is powered on (similar to PowerOnState). 
Right now Dimmer is set to last known value every time the device (PS-16-DZ) is powered on. 

Maybe I am missing something, but it seems that this feature is currently missing (based on what I have seen in the Commands wiki page). 

Thank you to all the contributors for this amazing software!

ascillato, 2019-08-03:
Hi, you can achieve that, using rules

briansmith7, 2019-08-03:
Awesome, this solved my issue! Thank you @ascillato!

closed 2019-08-03 by briansmith7

ascillato, 2019-08-03:
Hi,

If you want, please share the rules you made for other users. Thanks.