arendst/Tasmota #6174: Tuya Dimmer bug - 100% in tasmota not reals 100%

h4nc, 2019-08-03:
### BUG DESCRIPTION
With tasmota 6.6 master and even the newest dev version 100% in the tasmota UI (or set via mqtt) is not real 100% of the light. 
So if I press the Plus-hardware button in the device the light goes to real 100%.

I did not have this with 6.5. 
The reason why I went from 6.6 master to 6.6.0.2/3 is because there was another issue with the Tuya dimmer. It was showing six switches and the dimmer slider did not work. The dev version solved this, but it seems like there is still an issue.

ascillato2, 2019-08-04:
Hi,

Please, can you set your weblog to level 4 (weblog 4 command in the console) and try to do all the steps you have explained, and paste here the output of the console, in order to have more debug information? Thanks

h4nc, 2019-08-04:
Ok fine, this is what I did:

1) turn on via mqtt
2) dimm to 100% cia mqtt
3) press the plus hardware button to go to real 100%

Output:
```
17:45:16 MQT: empfangenes topic cmnd/sonoff_dimmer/Dimmer, Datengröße 2, Daten 42
17:45:16 SRC: MQTT
17:45:16 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "42"
17:45:16 SRC: Light
17:45:16 TYA: Send dim value=42 (id=2)
17:45:16 TYA: Send "55aa00060008020200040000002a3f"
17:45:16 TYA: Send "55aa0006000501010001010e"
17:45:17 MQT: stat/sonoff_dimmer/RESULT = {"POWER":"ON","Dimmer":42}
17:45:17 TYA: RX Packet: "55aa00070008020200040000002a40"
17:45:17 TYA: RX Dim State=42
17:45:17 TYA: RX Packet: "55aa0007000501010001010f"
17:45:17 TYA: RX Device-1 --> MCU State: On Current State:On
17:45:17 TYA: RX Packet: "55aa00070008020200040000002a40"
17:45:17 TYA: RX Dim State=42
17:45:17 TYA: RX Packet: "55aa0007000501010001010f"
17:45:17 TYA: RX Device-1 --> MCU State: On Current State:On
17:45:17 TYA: Send "55aa00000000ff"
17:45:17 TYA: RX Packet: "55aa000000010101"
17:45:17 TYA: Heartbeat
17:45:17 CFG: in Flash gespeichert am FA, zählen 2434, Bytes 3584
17:45:18 MQT: empfangenes topic cmnd/sonoff_dimmer/Dimmer, Datengröße 3, Daten 100
17:45:18 SRC: MQTT
17:45:18 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "100"
17:45:18 MQT: stat/sonoff_dimmer/RESULT = {"POWER":"ON","Dimmer":100}
17:45:18 TYA: Send dim value=100 (id=2)
17:45:18 TYA: Send "55aa00060008020200040000006479"
17:45:18 TYA: RX Packet: "55aa0007000802020004000000647a"
17:45:18 TYA: RX Dim State=100
17:45:18 TYA: RX Packet: "55aa0007000501010001010f"
17:45:18 TYA: RX Device-1 --> MCU State: On Current State:On
17:45:18 TYA: RX Packet: "55aa00070008020200040000006278"
17:45:18 TYA: RX Dim State=98
17:45:18 SRC: Switch
17:45:18 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "98"
17:45:19 MQT: stat/sonoff_dimmer/RESULT = {"POWER":"ON","Dimmer":98}
17:45:19 TYA: Send dim skipped value=98
17:45:19 CFG: in Flash gespeichert am F9, zählen 2435, Bytes 3584
17:45:25 TYA: RX Device-1 --> MCU State: On Current State:On
17:45:25 TYA: RX Packet: "55aa0007000802020004000000c9df"
17:45:25 TYA: RX Dim State=201
17:45:25 SRC: Switch
17:45:25 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "201"
17:45:25 MQT: stat/sonoff_dimmer/RESULT = {"Dimmer":100}
17:45:25 TYA: RX Packet: "55aa0007000501010001010f"
17:45:25 TYA: RX Device-1 --> MCU State: On Current State:On
17:45:25 TYA: RX Packet: "55aa0007000802020004000000cce2"
17:45:25 TYA: RX Dim State=204
17:45:25 SRC: Switch
17:45:25 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "204"
17:45:25 MQT: stat/sonoff_dimmer/RESULT = {"Dimmer":100}
17:45:25 TYA: RX Packet: "55aa0007000501010001010f"
17:45:25 TYA: RX Device-1 --> MCU State: On Current State:On
17:45:25 TYA: RX Packet: "55aa0007000802020004000000cfe5"
17:45:25 TYA: RX Dim State=207
17:45:25 SRC: Switch
17:45:25 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "207"
17:45:25 MQT: stat/sonoff_dimmer/RESULT = {"Dimmer":100}
17:45:25 TYA: RX Packet: "55aa0007000501010001010f"
17:45:25 TYA: RX Device-1 --> MCU State: On Current State:On
17:45:25 TYA: RX Packet: "55aa0007000802020004000000d2e8"
17:45:25 TYA: RX Dim State=210
17:45:25 SRC: Switch
17:45:25 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "210"
17:45:25 MQT: stat/sonoff_dimmer/RESULT = {"Dimmer":100}
17:45:25 TYA: RX Packet: "55aa0007000501010001010f"
17:45:25 TYA: RX Device-1 --> MCU State: On Current State:On
17:45:25 TYA: RX Packet: "55aa0007000802020004000000d4ea"
17:45:25 TYA: RX Dim State=212
17:45:25 SRC: Switch
17:45:25 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "212"
17:45:25 MQT: stat/sonoff_dimmer/RESULT = {"Dimmer":100}
17:45:25 TYA: RX Packet: "55aa0007000501010001010f"
17:45:25 TYA: RX Device-1 --> MCU State: On Current State:On
17:45:25 TYA: RX Packet: "55aa0007000802020004000000d7ed"
17:45:25 TYA: RX Dim State=215
17:45:25 SRC: Switch
17:45:25 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "215"
17:45:25 MQT: stat/sonoff_dimmer/RESULT = {"Dimmer":100}
17:45:25 TYA: RX Packet: "55aa0007000501010001010f"
17:45:25 TYA: RX Device-1 --> MCU State: On Current State:On
17:45:25 TYA: RX Packet: "55aa0007000802020004000000daf0"
17:45:25 TYA: RX Dim State=218
17:45:25 SRC: Switch
17:45:25 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "218"
17:45:25 MQT: stat/sonoff_dimmer/RESULT = {"Dimmer":100}
17:45:25 TYA: RX Packet: "55aa0007000501010001010f"
17:45:25 TYA: RX Device-1 --> MCU State: On Current State:On
17:45:25 TYA: RX Packet: "55aa0007000802020004000000ddf3"
17:45:25 TYA: RX Dim State=221
17:45:25 SRC: Switch
17:45:25 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "221"
17:45:25 MQT: stat/sonoff_dimmer/RESULT = {"Dimmer":100}
17:45:26 WIF: Prüfe Verbindung...
17:45:26 WIF: verbunden
17:45:28 TYA: Send "55aa00000000ff"
17:45:28 TYA: RX Packet: "55aa000000010101"
17:45:28 TYA: Heartbeat
```

s-hadinger, 2019-08-05:
Looks like there is an conversion issue between Tuya value (0..255) and Dimmer (0..100). I don’t own a Tuya device so I will not be able to debug, but I’ll have a look though.

s-hadinger, 2019-08-05:
Can you please check SetOption66?

h4nc, 2019-08-05:
Thank you. Now it works again like in 6.5.x 

100% = 100%

h4nc, 2019-10-25:
Just like to mention that I after upgrading to stable 6.7.0 I had to set `SetOption66 1` again, so this wasn't set persistent.

h4nc, 2019-10-25:
Updated to 6.7.0 and this issue came back. 100% light only at the physical switch.

Tried to set `SetOption66 1` again but it looks like this command is gone in the [commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands) section. 

How can I fix this in 6.7.0?

ascillato, 2019-10-25:
Yes, `setoption66 `has been replaced by `DimmerRange` command.

Please, check it in the commands page (https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

ascillato, 2019-10-25:
See related information in release page and in https://github.com/arendst/Sonoff-Tasmota/pull/6353

h4nc, 2019-10-25:
Thanks. Just to make sure I did not set something unknown with `SetOption66 1 ` I set it back to 0.

`DimmerRange 1 255` fixed it.

closed 2019-10-25 by h4nc