arendst/Tasmota #6177: Errors in compilation for NodeMCU (Arduino IDE)

fmarzocca, 2019-08-04:
### BUG DESCRIPTION
Trying to flash a NodeMCU (following this instructions: https://www.instructables.com/id/Flashing-SONOFF-Tasmota-Firmware-on-NodeMCU/) I am getting compilation errors.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ X Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X ] Device used (i.e. Sonoff Basic) : NodeMCU ESP 12E
- [X] Tasmota binary firmware version number used :  ? / (Arduino IDE)
- [X] Development IDE - Compiler / Upload tools used : Arduino IDE v.1.8.9
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
Can't run

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
Following the guide, I have downloaded the code from here: https://github.com/arendst/Sonoff-Tasmota
Then I have copied all the content of the `lib` folder into my Arduino/libraries. The started Arduino IDE, loaded sonoff.ino and launched compilation.
I got this error:
```
Arduino:1.8.9 (Mac OS X), Scheda:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

support:929:18: error: variable or field 'ModuleGpios' declared void
 void ModuleGpios(myio *gp)
                  ^
support:929:18: error: 'myio' was not declared in this scope
support:929:24: error: 'gp' was not declared in this scope
 void ModuleGpios(myio *gp)
                        ^
support:956:1: error: 'gpio_flag' does not name a type
 gpio_flag ModuleFlag()
 ^
support_rtc:212:37: error: 'TIME_T' has not been declared
 void BreakTime(uint32_t time_input, TIME_T &tm)
                                     ^
support_rtc:271:19: error: 'TIME_T' was not declared in this scope
 uint32_t MakeTime(TIME_T &tm)
                   ^
support_rtc:271:29: error: expected primary-expression before ')' token
 uint32_t MakeTime(TIME_T &tm)
                             ^
support_rtc:302:21: error: 'TimeRule' was not declared in this scope
 uint32_t RuleToTime(TimeRule r, int yr)
                     ^
support_rtc:302:33: error: expected primary-expression before 'int'
 uint32_t RuleToTime(TimeRule r, int yr)
                                 ^
support_rtc:302:39: error: expression list treated as compound expression in initializer [-fpermissive]
 uint32_t RuleToTime(TimeRule r, int yr)
                                       ^
xdrv_01_webserver:515:31: error: 'IPAddress' has not been declared
 void StartWebserver(int type, IPAddress ipweb)
                               ^
xdrv_09_timers:188:24: error: variable or field 'ApplyTimerOffsets' declared void
 void ApplyTimerOffsets(Timer *duskdawn)
                        ^
xdrv_09_timers:188:24: error: 'Timer' was not declared in this scope
xdrv_09_timers:188:31: error: 'duskdawn' was not declared in this scope
 void ApplyTimerOffsets(Timer *duskdawn)
                               ^
/Users/fabio/Downloads/Sonoff-Tasmota-development/sonoff/sonoff.ino: In function 'void GpioInit()':
sonoff:1231:22: error: 'ModuleGpios' was not declared in this scope
   ModuleGpios(&def_gp);
                      ^
sonoff:1249:31: error: 'ModuleFlag' was not declared in this scope
   my_module_flag = ModuleFlag();
                               ^
/Users/fabio/Downloads/Sonoff-Tasmota-development/sonoff/support_rtc.ino: In function 'String GetDuration(uint32_t)':
support_rtc:103:21: error: invalid initialization of reference of type 'int&' from expression of type 'TIME_T'
   BreakTime(time, ut);
                     ^
support_rtc:212:6: error: in passing argument 2 of 'void BreakTime(uint32_t, int&)'
 void BreakTime(uint32_t time_input, TIME_T &tm)
      ^
/Users/fabio/Downloads/Sonoff-Tasmota-development/sonoff/support_rtc.ino: In function 'String GetDT(uint32_t)':
support_rtc:122:26: error: invalid initialization of reference of type 'int&' from expression of type 'TIME_T'
   BreakTime(time, tmpTime);
                          ^
support_rtc:212:6: error: in passing argument 2 of 'void BreakTime(uint32_t, int&)'
 void BreakTime(uint32_t time_input, TIME_T &tm)
      ^
/Users/fabio/Downloads/Sonoff-Tasmota-development/sonoff/support_rtc.ino: In function 'uint32_t MakeTime(TIME_T&)':
support_rtc:271:29: error: 'uint32_t MakeTime(TIME_T&)' redeclared as different kind of symbol
 uint32_t MakeTime(TIME_T &tm)
                             ^
support_rtc:271:10: error: previous declaration of 'uint32_t MakeTime'
 uint32_t MakeTime(TIME_T &tm)
          ^
/Users/fabio/Downloads/Sonoff-Tasmota-development/sonoff/support_rtc.ino: In function 'uint32_t RuleToTime(TimeRule, int)':
support_rtc:302:39: error: 'uint32_t RuleToTime(TimeRule, int)' redeclared as different kind of symbol
 uint32_t RuleToTime(TimeRule r, int yr)
                                       ^
support_rtc:302:10: error: previous declaration of 'uint32_t RuleToTime'
 uint32_t RuleToTime(TimeRule r, int yr)
          ^
support_rtc:325:18: error: 'MakeTime' cannot be used as a function
   t = MakeTime(tm);         // First day of the month, or first day of next month for "Last" rules
                  ^
/Users/fabio/Downloads/Sonoff-Tasmota-development/sonoff/support_rtc.ino: In function 'void RtcSecond()':
support_rtc:375:74: error: 'RuleToTime' cannot be used as a function
         daylight_saving_time = RuleToTime(Settings.tflag[1], RtcTime.year);
                                                                          ^
support_rtc:376:67: error: 'RuleToTime' cannot be used as a function
         standard_time = RuleToTime(Settings.tflag[0], RtcTime.year);
                                                                   ^
Più di una libreria trovata per "NewPing.h"
Usata: /Users/fabio/Dropbox/codice/Arduino/libraries/NewPing-1.9.1
Non usata: /Users/fabio/Dropbox/codice/Arduino/libraries/NewPing
exit status 1
variable or field 'ModuleGpios' declared void
```

### EXPECTED BEHAVIOUR
Correct compilation

meingraham, 2019-08-04:
@fmarzocca 

**_From the Issue template preamble_**  
**&nbsp;&nbsp;\<Please DO NOT OPEN AN ISSUE:>**  
\<snip>  
**&nbsp;&nbsp;\<If your issue is _compilation problem_, please address that to the Tasmota Support Chat>**  
\<snip>  

Compilation errors are usually due to IDE installation errors. An alternative to installing an IDE is to use an online IDE such as [Gitpod](https://github.com/arendst/Sonoff-Tasmota/wiki/Compiling-Tasmota-on-Gitpod).

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

fmarzocca, 2019-08-04:
ok, thanks and sorry

closed 2019-08-04 by fmarzocca