arendst/Tasmota #6183: Sonoff T1EU3C - EU 3 gang switch (ESP8285) - enter flash mode

ivanovd, 2019-08-06:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Sonoff T1EU3C - EU 3 gang switch - How to flash
  
I have just bought the new Sonoff T1 EU 3 gang switch. It is based on the ESP8285 chip.

Can somebody help me with information how to put the chip in flash mode?

Here are some detailed pictures of the chip and the components around it.

I hope that somebody will be able to assist me to add this device to the database.

[image]
[image]
[image]
[image]
[image]

meingraham, 2019-08-06:
It appears that you did not perform any searches from any sources for your answer. The ESP8285 is put into programming mode by exactly the same means as the ESP8266.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-06 by ascillato2

ivanovd, 2019-08-06:
Thank you for the honest feedback @meingraham 

I did search and read throughout the issues and the forums, however, the layout of the board is different and shorting the R19 with GND as it is on T1 EU 2-gnag for example doesn't work. 

Therefore I was hoping to get an answer. 

Thanks anyway. I will try the chat.

toto79, 2019-08-07:
Try to find gpio0 in the button row, 2nd from the right corner. (Hope you can read pcb lines)
Pictures: https://github.com/arendst/Sonoff-Tasmota/issues/1249

meingraham, 2019-08-07:
Looks like the ESP pin 15 (GPIO0) is broken out to a via hole on the PCB. You may be able to get a small conductor (needle or small wire) into that hole. The you can verify with a meter that there is continuity between the hole and the pin. It may be easier to use that whole relying on friction to make the short between GPIO0 and GND. Alternatively, you can use said small conductor (needle) directly on pin 15 to short it to GND while powering up the device under 3.3v to get it into programming mode.

meingraham, 2019-08-07:
@toto79 subsequent Discord dialog...

```
[2:01 PM] ivanovd: I found that in the datasheet, I also noticed that there is a label GPIO0 on the back side of the board and on the other side it is R10
[2:02 PM] ivanovd: I checked with the multimeter and pin 15 is indeed connected to R10
```

ivanovd, 2019-08-07:
I managed to flash the switch with Tasmota and it was successful. However, after resetting it, it doesn't boot. The three buttons stay lit up just as when it is in flash mode. I tried flashing ESPHome with the same result. It flashes successfully, but after the reset it doesn't boot into the new firmware.

```
Command: esptool.py --port COM3 --baud 115200 --after no_reset write_flash --flash_mode dio 0x00000 C:\Users\techn\Desktop\bathroom_lights.bin --erase-all

esptool.py v2.6
Serial port COM3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
MAC: dc:4f:22:ac:78:e1
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 4.0s
Flash params set to 0x0220
Compressed 320048 bytes to 224665...
Wrote 320048 bytes (224665 compressed) at 0x00000000 in 20.2 seconds (effective 126.8 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.
```

Any ideas?

toto79, 2019-08-08:
Flashmode
Gpio0 is still connected to ground
VCC problems while or after flashing

(Gpio2 or gpio15 Arena in wrong state)
(Maby it helps to Flash a blank.bin)

meingraham, 2019-08-08:
@toto79 Perform an erase_flash using esptool.py as documented in the wiki installation articles.

toto79, 2019-08-08:
@meingraham: thx for support, but it's not my thread and i'm not in trouble, i'm answering to questions of invanovd :)

... he asked: "Any ideas?" (Device is still in flash mode After reset)

So i give my ideas, hat can couse gis issue ;)

meingraham, 2019-08-08:
@toto79 It wasn't clear that you were providing some steps to attempt. I thought you were having issues. We had already resolved his issue (see his EDIT). He was using the wrong flashing mode.

Flashing a blank binary is not the recommended method for clearing our the flash memory. It may not completely erase all of flash.

jimmypikachu, 2019-08-24:
> I managed to flash the switch with Tasmota and it was successful. However, after resetting it, it doesn't boot. The three buttons stay lit up just as when it is in flash mode. I tried flashing [...]

Hi ivanovd can you show an image of which pin did you pulled to ground? I had the same problem I used to use T1 UK, but now they stopped the production so I am trying out the T2EU1C / 2C / 3C, but cannot find the GPIO0 nor the said datasheet. I unboxed a 1C but now see nocomponent in the R10, R24 and Q3 Q4 connection, probably they were for 2nd and 3rd Channel?

ivanovd, 2019-08-24:
It is R10 that I shorter to GND

ivanovd, 2019-08-24:
Keep in mind that in 1C it might be a different component. Flip the board around and see if there is an engravement GPIO0 on the back of the component that is linked to GPIO0. The other way to look for it is with a multimeter set to measure resistance ... one end of the multimeter goes to pin 17 of the ESP8285 and with the other you check which resistor will measure ~0 ohm

jimmypikachu, 2019-08-24:
> Keep in mind that in 1C it might be a different component. Flip the board around and see if there is an engravement GPIO0 on the back of the component that is linked to GPIO0. The other way to look for it is with a multimeter set to measure resistance ... one end of the multimeter goes to pin 15 of the ESP8285 and with the other you check which resistor will measure ~0 ohm

Thank you very much ivanovd, I somehow also traced to that resistor, but the printing on the board is a bit unclear so I thought that also labeled R19. what I meant I saw R10 was the resistor 1cm directly under the "J" of "JCYD01", I thought that was the actual R10.

Anyway I will try again later, as my first attempt cannot put it to program mode and I thought I was facing the R19 problem. Thanks again!!