arendst/Tasmota #6188: Power Meter Reading with Schneider  EasyLogic™ PM2200 series

sensorthai, 2019-08-07:
Hi, 
  I try to create new Power Meter Sensor for Schneider  EasyLogic™ PM2200 
but it reboot after send 
  Any one investigate for me ?

http://www.schneider-electric.com/resources/sites/SCHNEIDER_ELECTRIC/content/live/FAQS/307000/FA307813/en_US/Public_PM2xxx_PMC%20Register%20List_v1001.xlsx

`
00:00:00 WIF: Connecting to AP2 4G-CPE-020331 in mode 11N as SONOFF_CB3D6F...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on SONOFF_CB3D6Fwith IP address 192.168.100.195
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:06 MQT: v1/devices/me/LWT = Online (retained)
00:00:06 MQT: v1/devices/me/cmnd/rpc/request/ =
00:00:06 MQT: v1/devices/me/attributes = {"Module":"WiFi-EZ001","Version":"6.5.0(sonoff)","FallbackTopic":"cmnd/DVES_CB3D6F_fb/","GroupTopic":"capteur"}
00:00:06 MQT: v1/devices/me/attributes = {"WebServerMode":"Admin","Hostname":"SONOFF_CB3D6F","IPAddress":"192.168.100.195"}
00:00:06 MQT: v1/devices/me/attributes = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40204900 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
11:06:09 MQT: v1/devices/me/attributes = {"Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"AP":2,"SSId":"4G-CPE-020331","BSSId":"00:03:7F:86:6B:40","Channel":7,"RSSI":98,"LinkCount":1,"Downtime":"0T00:00:04"}
11:06:09 MQT: v1/devices/me/telemetry = {"ts":1565150769000 ,"values":{"ANALOG_A0":9,"PM_Total":51.241,"PM_Frequency":50.01,"PM_Active_L1":0,"PM_Active_L2":0,"PM_Active_L3":0,"PM_ReActive_L1":-0,"PM_ReActive_L2":0,"PM_ReActive_L3":0,"PM_PF_L1":null,"PM_PF_L2":null,"PM_PF_L3":null,"PM_Volt_L1":119,"PM_Volt_L2":0,"PM_Volt_L3":120,"PM_Current_L1":0.000,"PM_Current_L2":0.000,"PM_Current_L3":0.000}}
11:06:10 MQT: Attempting connection...
11:06:10 MQT: Connected
11:06:10 MQT: v1/devices/me/LWT = Online (retained)
11:06:10 MQT: v1/devices/me/cmnd/rpc/request/ =

Exception (0):
epc1=0x40204900 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys
sp: 3ffffc30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdd0:  40107402 00dd9999 00000019 00000100
3ffffde0:  c0036035 401003ca 00000001 00000000
3ffffdf0:  00000000 40100197 00000000 00000022
3ffffe00:  3fffc200 401073c8 3fffc258 4000050c
3ffffe10:  4000437d 00000030 00000018 ffffffff
3ffffe20:  60000200 00000008 00021700 80000000
3ffffe30:  20000000 3fff7bf8 80000000 200fc280
3ffffe40:  80000000 3fffc6fc 3ffeef74 3fff7bfc
3ffffe50:  00000074 000fc280 60000600 00000030
3ffffe60:  ffffffff 0000000c 00000000 4020490f
3ffffe70:  4010746d 00080000 00000000 4010745c
3ffffe80:  c0036035 00bdf6ba 00000000 4000050c
3ffffe90:  00000000 00000000 0000001f 40104301
3ffffea0:  4000050c 401073c8 3fffc258 4000050c
3ffffeb0:  40000f68 00000030 00000014 ffffffff
3ffffec0:  40000f58 00000000 00000020 00000000
3ffffed0:  00000000 4023a558 00000000 bfffffff
3ffffee0:  ffffffff 3fffc6fc 3ffeef74 3fffdab0
3ffffef0:  00000000 3fffdcb0 3ffef030 00000030
3fffff00:  00000000 400042db 3ffead71 60000600
3fffff10:  40004b31 3fff797c 000002f4 000fc000
3fffff20:  40104866 3ffef020 3ffeeec0 401070c8
3fffff30:  4023a029 3ffeeec0 3ffef020 020649ff
3fffff40:  3fff797c 00001000 4023a4be 00000008
3fffff50:  3ffef020 3ffeef74 4023a56b 3ffeef74
3fffff60:  3ffef020 00a1ce43 4023a480 000000f2
3fffff70:  4023c791 3ffeef74 3ffef020 020623f8
3fffff80:  4023c7d6 3fffdab0 00000000 3fffdcb0
3fffff90:  3ffef038 00000000 40000f65 3fffdab0
3fffffa0:  40000f49 00003603 3fffdab0 40000f49
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
[... 365 lines cut ...]

  if (pm2200_type) {
    switch (function) {
      case FUNC_INIT:
        PM2200Init();
        break;
    case FUNC_EVERY_250_MSECOND:
    //case FUNC_EVERY_SECOND:
        PM2200250ms();
        break;
      case FUNC_JSON_APPEND:
        PM2200Show(1);
        break;
#ifdef USE_WEBSERVER
      case FUNC_WEB_APPEND:
        PM2200Show(0);
        break;
#endif  // USE_WEBSERVER
    }
  }
  return result;
}

#endif
`

ascillato2, 2019-08-07:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Also remember that it ks available the Tasmota support chat for general questions. 

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-08-07:
If you are using core 2.3.0, you must use the tx and rx gpios of the esp8266 (hardware serial)

If you need to use software serial in other gpios, please upgrade your core to 2.5.2

Also, exception(0) sometimes is led to insuficient power to your device.

meingraham, 2019-08-07:
I believe hardware serial is also supported on GPIO13/GPIO15 for some devices:

```
/* 6.3.0.3 20181105
 * Add optional hardware serial when GPIO13(Rx) and GPIO15(Tx) are selected removing hardware serial from GPIO01(Tx) and GPIO03(Rx) (#4288)
```

This was incorporated into the master branch as of 6.4.0.

sensorthai, 2019-08-07:
Sorry for Template missing,

after I change to core 2.5.2

it has no reboot as in log, Thank you for your support.

if I want to contribute this as new feature of Tasmota. how I can do that ?

00:00:00 Project capteur Sonoff Version 6.5.0(sonoff)-2_5_0
00:00:00 WIF: Connecting to AP1 wsa in mode 11N as sonoff...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff with IP address 192.168.1.146
14:39:36 LOG: SerialLog 2, WebLog 2, SysLog 0, LogHost , LogPort 514, TelePeriod 15
14:39:36 RSL: tele/sonoff/attributes = {"Uptime":"0T00:00:29","SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"AP":1,"SSId":"wsa","BSSId":"78:44:76:E4:4E:90","Channel":11,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:04"}
14:39:36 RSL: tele/sonoff/telemetry = {"ts":1565185176000 ,"values":{"ANALOG_A1":10,"PM_Total":51.241,"PM_Frequency":50.02,"PM_Active_L1":0,"PM_Active_L2":0,"PM_Active_L3":0,"PM_ReActive_L1":-0,"PM_ReActive_L2":0,"PM_ReActive_L3":0,"PM_PF_L1":null,"PM_PF_L2":null,"PM_PF_L3":null,"PM_Volt_L1":118,"PM_Volt_L2":0,"PM_Volt_L3":118,"PM_Current_L1":0.000,"PM_Current_L2":0.000,"PM_Current_L3":0.000}}
14:39:51 RSL: tele/sonoff/attributes = {"Uptime":"0T00:00:44","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"AP":1,"SSId":"wsa","BSSId":"78:44:76:E4:4E:90","Channel":11,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:04"}
14:39:51 RSL: tele/sonoff/telemetry = {"ts":1565185191000 ,"values":{"ANALOG_A1":3,"PM_Total":51.241,"PM_Frequency":50.01,"PM_Active_L1":0,"PM_Active_L2":0,"PM_Active_L3":0,"PM_ReActive_L1":-0,"PM_ReActive_L2":0,"PM_ReActive_L3":0,"PM_PF_L1":null,"PM_PF_L2":null,"PM_PF_L3":null,"PM_Volt_L1":118,"PM_Volt_L2":0,"PM_Volt_L3":118,"PM_Current_L1":0.000,"PM_Current_L2":0.000,"PM_Current_L3":0.000}}
14:40:06 RSL: tele/sonoff/attributes = {"Uptime":"0T00:00:59","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"AP":1,"SSId":"wsa","BSSId":"78:44:76:E4:4E:90","Channel":11,"RSSI":98,"LinkCount":1,"Downtime":"0T00:00:04"}
14:40:06 RSL: tele/sonoff/telemetry = {"ts":1565185206000 ,"values":{"ANALOG_A1":8,"PM_Total":51.241,"PM_Frequency":50.01,"PM_Active_L1":0,"PM_Active_L2":0,"PM_Active_L3":0,"PM_ReActive_L1":-0,"PM_ReActive_L2":0,"PM_ReActive_L3":0,"PM_PF_L1":null,"PM_PF_L2":null,"PM_PF_L3":null,"PM_Volt_L1":118,"PM_Volt_L2":0,"PM_Volt_L3":119,"PM_Current_L1":0.000,"PM_Current_L2":0.000,"PM_Current_L3":0.000}}
14:40:21 RSL: tele/sonoff/attributes = {"Uptime":"0T00:01:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"AP":1,"SSId":"wsa","BSSId":"78:44:76:E4:4E:90","Channel":11,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:04"}
14:40:21 RSL: tele/sonoff/telemetry = {"ts":1565185221000 ,"values":{"ANALOG_A1":3,"PM_Total":51.241,"PM_Frequency":50.04,"PM_Active_L1":0,"PM_Active_L2":0,"PM_Active_L3":0,"PM_ReActive_L1":-0,"PM_ReActive_L2":0,"PM_ReActive_L3":0,"PM_PF_L1":null,"PM_PF_L2":null,"PM_PF_L3":null,"PM_Volt_L1":118,"PM_Volt_L2":0,"PM_Volt_L3":118,"PM_Current_L1":0.000,"PM_Current_L2":0.000,"PM_Current_L3":0.000}}
14:40:36 RSL: tele/sonoff/attributes = {"Uptime":"0T00:01:29","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"AP":1,"SSId":"wsa","BSSId":"78:44:76:E4:4E:90","Channel":11,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:04"}
14:40:36 RSL: tele/sonoff/telemetry = {"ts":1565185236000 ,"values":{"ANALOG_A1":3,"PM_Total":51.241,"PM_Frequency":50.04,"PM_Active_L1":0,"PM_Active_L2":0,"PM_Active_L3":0,"PM_ReActive_L1":-0,"PM_ReActive_L2":0,"PM_ReActive_L3":0,"PM_PF_L1":null,"PM_PF_L2":null,"PM_PF_L3":null,"PM_Volt_L1":118,"PM_Volt_L2":0,"PM_Volt_L3":118,"PM_Current_L1":0.000,"PM_Current_L2":0.000,"PM_Current_L3":0.000}}
14:40:51 RSL: tele/sonoff/attributes = {"Uptime":"0T00:01:44","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"AP":1,"SSId":"wsa","BSSId":"78:44:76:E4:4E:90","Channel":11,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:04"}

ascillato, 2019-08-07:
> if I want to contribute this as new feature of Tasmota. how I can do that ?

Please, do a pull request adding this as a new driver. Thanks

s-hadinger, 2019-08-07:
Thanks for this feature.

Maybe a hint about your crash, you are allocating a lot of data on the stack in `PM2200Show()`: 17x33 bytes, with 4 bytes alignment that makes at least 612 bytes on the stack. Please keep in mind the ESP8266 stack is 4k (actually it's a bit more as there is a continuation stack as well).

Can you please split the `snprintf_P()` and add each value one after the other reusing the same buffer.

Here is an example from `xdrv_08_serial_bridge.ino`:

```
      Response_P(PSTR("{\"" D_JSON_SSERIALRECEIVED "\":\""));
      for (uint32_t i = 0; i < serial_bridge_in_byte_counter; i++) {
        ResponseAppend_P(PSTR("%02x"), serial_bridge_buffer[i]);
      }
      ResponseAppend_P(PSTR("\"}"));
```

ascillato2, 2019-08-17:
@sensorthai 

Thanks for your interest on this. We would love to have that added to tasmota. If you need assistance on making a Pull Request, please just ask.

closed 2019-08-17 by ascillato2