arendst/Tasmota #6193: iFan03 PowerOnState only for light

crispy78, 2019-08-08:
Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Not a problem, just an inconvenience.
I would like the light to turn on when I power up the fan (iFan03 powered). When I use PowerOnState on it will also turn up the fan to max. power. I haven’t made rule (yet) that runs on boot, turning on the light at boot.

**Describe the solution you'd like**
I could think of several ways to fix this, PowerOnState1 for the first relay, PowerOnState2 for the second, etc., etc. FanPowerOnState (on/off/speed 1, speed 2, speed 3...).

**Describe alternatives you've considered**
Making a rule that turns on the light at boot and turns off the fan.

**Additional context**
-

**(Please, remember to close the issue when the problem has been addressed)**

crispy78, 2019-08-10:
I’ve created a rule that works for me

‘’’
on system#boot do backlog power1 on; fanspeed 0 endon
‘’’

ascillato2, 2019-08-10:
A faster rule could be:

```
Rule1 1 
Rule1 on power1#boot do backlog power1 on; fanspeed 0 endon
```

`Power1#boot` is triggered before system#boot

crispy78, 2019-08-10:
Thanks 👍 
I'm going to go for the first suggestion.

closed 2019-08-11 by ascillato2

byrnes-patrick, 2019-08-13:
@ascillato2 

When I use your first suggestion and boot the device, the light flashes on for a second, turns off, then turns on and stays on. Is there something I am doing wrong or is there another command I need to use to avoid the light flashing?

crispy78, 2019-08-13:
Same here, just noticed the same behavior.
Reverting back to:
``` PowerOnState 3
Rule1 on system#boot do backlog power1 on; fan speed 0 endon
Rule1 1 ```

gnasser76, 2019-09-03:
> A faster rule could be:
> 
> ```
> Poweronstate 0
> Rule1 1 
> Rule1 on power1#boot do backlog fanspeed 0; power1 1 endon
> ```
> 
> `Power1#boot` is triggered before system#boot
> 
> Also, if you want the light to be as its previous state but the fan to start always turned off:
> 
> ```
> Poweronstate 3
> Rule1 1 
> Rule1 on power1#boot do backlog fanspeed 0 endon
> ```

what if i wanted the fan and light to keep previous state ?

rafiw, 2024-12-18:
> what if i wanted the fan and light to keep previous state ?
try this
```
Rule1
  ON System#Boot DO
    Power1 %mem1%
    FanSpeed %mem2%
  ENDON
  ON Power1#State DO Mem1 %value% ENDON
  ON FanSpeed#State DO Mem2 %value% ENDON
Rule1 1
```