arendst/Tasmota #6206: TTP223 crash sonoff and nodemcu board

pcpaulo, 2019-08-10:
I am try to use sonoff configurated as button on GPIO14 and after few minutes the system crash and clean all configuration. I need to do all configuration again. I have tested with another board nodemcu with the same problem. there is not trace in the console.
 i am using version 6.5

meingraham, 2019-08-10:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

You have something connected incorrectly which is causing Tasmota to reboot repeatedly. To get this to stop, Tasmota resets settings so that it ca quit rebooting.

Please close this issue address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

pcpaulo, 2019-08-10:
i will follow Tasmota Support Discord Chat

closed 2019-08-10 by pcpaulo