arendst/Tasmota #6207: TCP/IP over serial, not wifi

dmit2k, 2019-08-11:
Hello,

I’m pretty new to Tasmota but really like it so far, thanks for the great product!

I’m trying to use generic ESP8266 board as an external controller for various sensors and relays, which should be directly attached to an embedded linux board over serial / usb. This linux board has no wireless connectivity, but I would still really like to stay with Tasmota web interface and MQTT client.

So here are my two questions / feature requests:

1) Is there any way to provide the ESP Tasmota board with TCP/IP over serial, like PPP or something like this?

2) if not, then is is possible to control / get status of Tasmota board from the serial console commands?

s-hadinger, 2019-08-11:
I'm not sure to understand what you try to achieve. Tasmota is primarily designed for MQTT, it does not directly provide Serial over TCP, though you may achieve similar results purely with MQTT.

All commands are available through Serial interface and Web interface as well. If you want to connect to a Linux box, I recommend a device like Wemos D1 mini, they offer native USB connectivity directly connected to the Serial interface. You can use it from linux with serial emulation over USB.

ascillato2, 2019-08-11:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-11 by ascillato2

dmit2k, 2019-08-11:
Sorry for continuing, but I don't understand why this topic was closed so fast without being answered. I guess @s-hadinger has not tried to dig into the question and understand what exactly the question is about.

> Tasmota is primarily designed for MQTT, it does not directly provide Serial over TCP, though you may achieve similar results purely with MQTT.

Which exactly results I can achieve with MQTT? Connectivity through a serial port? Then this is what I'm asking for. Please point me to the corresponding documentation page.

> All commands are available through Serial interface and Web interface as well.

OK, thank you, if that's true then I have missed something about this functionality as when I connect through the serial interface to the ESP board I can see all the debug info, but can't control the device and peripherials.

> If you want to connect to a Linux box, I recommend a device like Wemos D1 mini, they offer native USB connectivity directly connected to the Serial interface. You can use it from linux with serial emulation over USB.

I don't have a problem connecting ESP to USB. I was just asking if it is (or can be) possible to provide TCP/IP connectivity to the ESP via the serial port like PPP, as I don't have WiFi network at this place.

ascillato, 2019-08-11:
>I was just asking if it is (or can be) possible to provide TCP/IP connectivity to the ESP via the serial port like PPP, as I don't have WiFi network at this place.

Sorry, not supported

ascillato, 2019-08-11:
From serial interface you can control Tasmota with the commands explained in the wiki: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

dmit2k, 2019-08-11:
> From serial interface you can control Tasmota with the commands explained in the wiki: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

Thank you for the answer. Probably I have some misunderstanding of the serial console implementation - I can see all the debug info, but it does not react on typing any commands :( OK

For the PPP over serial part of the question - is it technically possible to implement with existing Tasmota build or a code modification? Does it worth providing a kind of bounty?

arendst, 2019-08-11:
All commands are available from serial console.

If you don't see responses it's because your serial terminal is not configured correctly. It needs to send cr and lf with every users request.

Once configured correctly execute command seriallog 3 and you should be able to execute commands as documented in the wiki.

s-hadinger, 2019-08-11:
Also note that you need to enable local echo if you want to see what you're typing in the serial console.

dmit2k, 2019-08-11:
@arendst thanks for the reply, for the product, and sorry for the buzzing!

@s-hadinger:
> Also note that you need to enable local echo if you want to see what you're typing in the serial console.

Yes I have. I have also tried both from my Mac with 'screen /dev/tty.usbserial-1410 115200' and minicom, and also from my RPi device with all possible serial terminals I have (picocomd, minicom, screen, etc).

The result is the same: I don't see any reaction on typing, but still see all the logs generated by ESP. Checked with another firmwares and it works like a charm.....

I have tried

dmit2k, 2019-08-11:
> Any more ideas?

serial console issue resolved, Ctrl+J is our best friend.... Though this should be somehow pointed out in the docs.......

Sorry for bothering

meingraham, 2019-08-11:
@dmit2k 

@arendts [above](https://github.com/arendst/Sonoff-Tasmota/issues/6207#issuecomment-520259833) - `It needs to send cr and lf with every users request`

\<CR> == Ctrl-M == Enter
\<LF> == Ctrl-J