arendst/Tasmota #6208: Erratic behaviour on Blitzwolf BHP-6

ilgrank, 2019-08-11:
Hi!
I've flashed several devices today, among which two Sonoff S20 and a Blitzwolf BHP6
While the Sonoffs are working perfectly, I have a strange issue with the BHP6.

Basically, when pressing slowly (ie waiting at least 3-4 seconds between "Toggle" presses) it works. Anything faster, and it either stops responding, 'slows down' a lot (ie, the buttonpresses do nothing, 'queue up', and then I hear several relay togglings sequentially) or reports wrong info (says on when off)

I'm on Tasmota 6.60 w/SDK 2.30, and I've configured it for Blitzwolf SHP series.
Voltage and power reading are working correctly.
Nothing interesting in the logs.  
[here ](https://cdn1.imggmi.com/uploads/2019/8/11/e4b70076c26da5516ce4f5aff3c85f98-full.jpg) is a screenshot of the info screen.

I've searched around, but there seems to be very few people with BHP6..

ascillato2, 2019-08-11:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you? The information you provided is not enough to search for any trouble.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-11 by ascillato2

ascillato, 2019-08-11:
If you are not using MQTT, you should turn it off.

ilgrank, 2019-08-11:
- [yes] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [yes] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [yes] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [yes] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [no] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [v] Device used (i.e. Sonoff Basic) : BlitzWolf BW-SHP6
- [v] Tasmota binary firmware version number used : 6.60 basic pre-compiled
- [v] Development IDE - Compiler / Upload tools used : tried 2.30 and 2.52, flashed with NodeMCU-PyFlasher-4.0-x64
- [ ] Provide the output of command ``status 0`` :
```
21:20:50 CMD: status 0
21:20:50 RSL: STATUS = {"Status":{"Module":45,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:20:50 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/020502//sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:10","StartupUTC":"2019-08-11T20:20:40","Sleep":50,"CfgHolder":4617,"BootCount":36,"SaveCount":114,"SaveAddress":"F9000"}}
21:20:50 RSL: STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019-07-06T14:26:16","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
21:20:50 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","280500000100060000000000000000000000","00000000"]}}
21:20:50 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":557,"Free":444,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
21:20:50 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-0128","IPAddress":"192.168.0.235","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.102","Mac":"CC:50:E3:26:xx:xx","Webserver":2,"WifiConfig":4}}
21:20:50 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun Aug 11 20:20:50 2019","Local":"Sun Aug 11 21:20:50 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:37","Sunset":"20:13"}}
21:20:50 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:20:50 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-08-11T21:20:50","ENERGY":{"TotalStartTime":"2019-08-11T20:06:27","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":27,"ApparentPower":78,"ReactivePower":73,"Factor":0.35,"Voltage":237,"Current":0.327}}}
21:20:50 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-08-11T21:20:50","Uptime":"0T00:00:10","Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Home","BSSId":"44:4E:6D:DD:xx:xx","Channel":11,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:04"}}}
21:20:53 RSL: STATE = {"Time":"2019-08-11T21:20:53","Uptime":"0T00:00:13","Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"POWER":"ON","Wifi":{"AP":1,"SSId":"Home","BSSId":"44:4E:6D:DD:xx:xx","Channel":11,"RSSI":70,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:20:53 RSL: SENSOR = {"Time":"2019-08-11T21:20:53","ENERGY":{"TotalStartTime":"2019-08-11T20:06:27","Total":0.001,"Yesterday":0.000,"Today":0.001,"Period":0,"Power":27,"ApparentPower":72,"ReactivePower":66,"Factor":0.38,"Voltage":237,"Current":0.302}}
```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
21:27:30 RSL: INFO1 = {"Module":"BlitzWolf SHP","Version":"6.6.0(sonoff)","FallbackTopic":"cmnd/DVES_26E080_fb/","GroupTopic":"sonoffs"}
21:27:30 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-0128","IPAddress":"192.168.0.235"}
21:27:30 RSL: INFO3 = {"RestartReason":"External System"}
21:27:30 RSL: RESULT = {"POWER":"ON"}
21:27:30 RSL: POWER = ON
21:27:30 NTP: Drift 0, (UTC) Sun Aug 11 20:27:30 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
21:27:35 APP: Boot Count 38
21:27:35 CFG: Saved to flash at F6, Count 117, Bytes 3584
21:27:38 RSL: STATE = {"Time":"2019-08-11T21:27:38","Uptime":"0T00:00:13","Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"POWER":"ON","Wifi":{"AP":1,"SSId":"Amber2","BSSId":"44:4E:6D:DD:0E:6F","Channel":11,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:27:38 RSL: SENSOR = {"Time":"2019-08-11T21:27:38","ENERGY":{"TotalStartTime":"2019-08-11T20:06:27","Total":0.004,"Yesterday":0.000,"Today":0.004,"Period":0,"Power":30,"ApparentPower":85,"ReactivePower":79,"Factor":0.35,"Voltage":239,"Current":0.355}}
21:27:49 WIF: Checking connection...
21:27:49 WIF: Connected

```
Additional info: I tried both with SDK 2.30 and SDK 2.52, same behaviour.
A brief video of me clicking once on the toggle button and what happens is attached.

[Blitzwolf.zip](https://github.com/arendst/Sonoff-Tasmota/files/3490428/Blitzwolf.zip)