arendst/Tasmota #6211: EnergyReset3 xx.xx to preset the energy total value does not work.

wongnam, 2019-08-12:
### BUG DESCRIPTION
Today i try to set a value for the Energy Total but it does not work, it's still the current value.
eg: type or sent by MQTT **EnergyReset3 25.7** in console but the value will not change.

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : NodeMCU + PZEM004T + LCDLCD1602
- [x] Tasmota binary firmware version number used : Ver 6.6.0.2(sonoff)-core2.5.2 (self-compiled)
- [x] Development IDE - Compiler / Upload tools used : VSC / PlatformIO IDE
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
15:35:07 CMD: status 0
15:35:07 MQT: stat/energy1/STATUS = {"Status":{"Module":0,"FriendlyName":["Energy Monitor"],"Topic":"energy1","ButtonTopic":"","Power":1,"PowerOnState":3,"LedState":7,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:35:07 MQT: stat/energy1/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"21T16:53:20","StartupUTC":"2019-07-21T15:41:47","Sleep":50,"CfgHolder":4617,"BootCount":33,"SaveCount":400,"SaveAddress":"F8000"}}
15:35:07 MQT: stat/energy1/STATUS2 = {"StatusFWR":{"Version":"6.6.0.2(sonoff)","BuildDateTime":"2019-07-21T22:40:27","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
15:35:07 MQT: stat/energy1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":1,"LogHost":"192.168.12.155","LogPort":514,"SSId":["iot-1","Wong"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00208009","280500000100060000005A00000000000000","00000000"]}}
15:35:07 MQT: stat/energy1/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1625C2","FlashMode":3,"Features":["00000809","0F28A386","041001A0","020000CD","000008C0"]}}
15:35:07 MQT: stat/energy1/STATUS5 = {"StatusNET":{"Hostname":"energy1-1856","IPAddress":"192.168.12.116","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"BC:DD:C2:82:47:40","Webserver":2,"WifiConfig":4}}
15:35:07 MQT: stat/energy1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_824740","MqttUser":"admin","MqttCount":19,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:35:07 MQT: stat/energy1/STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 12 08:35:07 2019","Local":"Mon Aug 12 15:35:07 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+07:00","Sunrise":"05:42","Sunset":"18:13"}}
15:35:07 MQT: stat/energy1/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:35:07 MQT: stat/energy1/STATUS10 = {"StatusSNS":{"Time":"2019-08-12T15:35:07","Epoch":1565598907,"ENERGY":{"TotalStartTime":"2019-08-12T00:00:00","Total":11.552,"Yesterday":17.748,"Today":11.532,"Power":1242,"ApparentPower":1286,"ReactivePower":332,"Factor":0.97,"Voltage":230,"Current":5.590}}}
15:35:07 MQT: stat/energy1/STATUS11 = {"StatusSTS":{"Time":"2019-08-12T15:35:07","Epoch":1565598907,"Uptime":"21T16:53:20","UptimeSec":1875200,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-1","BSSId":"B0:C7:45:36:85:E0","Channel":7,"RSSI":100,"LinkCount":13,"Downtime":"0T00:00:47"}}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
15:38:46 CMD: weblog 4
15:38:46 MQT: stat/energy1/RESULT = {"WebLog":4}
15:38:46 DMP: A3 38 11 54 00 00 40
15:38:46 CFG: Saved to flash at FA, Count 406, Bytes 3584
15:38:47 DMP: A0 00 E4 00 00 00 84
15:38:47 DMP: A1 00 05 37 00 00 DD
15:38:48 DMP: A2 04 CA 00 00 00 70
15:38:48 DMP: A3 38 11 54 00 00 40
15:38:49 DMP: A0 00 E4 00 00 00 84
15:38:49 DMP: A1 00 05 34 00 00 DA
15:38:49 WIF: Checking connection...
15:38:49 WIF: Connected
15:38:50 DMP: A2 04 CC 00 00 00 72
15:38:50 DMP: A3 38 11 55 00 00 41
15:38:50 CMD: EnergyReset3 25.7
15:38:50 SRC: WebConsole from 192.168.12.130
15:38:50 RSL: Received Topic /EnergyReset3, Data Size 4, Data 25.7
15:38:50 RSL: Group 0, Index 3, Command ENERGYRESET, Data 25.7
15:38:50 MQT: stat/energy1/RESULT = {"EnergyReset":{"Total":11.634,"Yesterday":17.748,"Today":11.609}}
15:38:51 DMP: A0 00 E4 02 00 00 86
15:38:51 DMP: A1 00 05 34 00 00 DA
15:38:52 DMP: A2 04 C6 00 00 00 6C
15:38:52 DMP: A3 38 11 56 00 00 42
15:38:53 DMP: A0 00 E4 02 00 00 86
15:38:53 DMP: A1 00 05 35 00 00 DB
15:38:54 DMP: A2 04 CA 00 00 00 70
15:38:54 DMP: A3 38 11 57 00 00 43
15:38:55 MQT: tele/energy1/STATE = {"Time":"2019-08-12T15:38:55","Epoch":1565599135,"Uptime":"21T16:57:08","UptimeSec":1875428,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-1","BSSId":"B0:C7:45:36:85:E0","Channel":7,"RSSI":100,"LinkCount":13,"Downtime":"0T00:00:47"}}
15:38:55 MQT: tele/energy1/SENSOR = {"Time":"2019-08-12T15:38:55","Epoch":1565599135,"ENERGY":{"TotalStartTime":"2019-08-12T00:00:00","Total":11.636,"Yesterday":17.748,"Today":11.611,"Period":4,"Power":1226,"ApparentPower":1262,"ReactivePower":299,"Factor":0.97,"Voltage":228,"Current":5.530}}
15:38:55 DBG: [Total   11.636kW]
15:38:55 DMP: A0 00 E4 02 00 00 86
15:38:55 DMP: A1 00 05 35 00 00 DB
15:38:56 DMP: A2 04 C7 00 00 00 6D
15:38:56 DMP: A3 38 11 57 00 00 43
15:38:57 DBG: [Yesterd 17.748kW]
15:38:57 DMP: A0 00 E4 02 00 00 86
15:38:57 DMP: A1 00 05 38 00 00 DE
15:38:58 DMP: A2 04 CE 00 00 00 74
15:38:58 DMP: A3 38 11 58 00 00 44
15:38:59 DBG: [Today   11.611kW]
15:38:59 DMP: A0 00 E5 02 00 00 87
15:38:59 DMP: A1 00 05 38 00 00 DE
15:39:00 DMP: A2 04 CF 00 00 00 75
15:39:00 DMP: A3 38 11 59 00 00 45
15:39:01 DBG: [Period  4Wh     ]
15:39:01 DMP: A0 00 E5 02 00 00 87
15:39:01 DMP: A1 00 05 35 00 00 DB
15:39:02 DMP: A2 04 CF 00 00 00 75
15:39:02 DMP: A3 38 11 59 00 00 45
15:39:03 DBG: [Power   1226W   ]
15:39:03 DMP: A0 00 E5 01 00 00 86
15:39:03 DMP: A1 00 05 35 00 00 DB
15:39:04 DMP: A2 04 CE 00 00 00 74
15:39:04 DMP: A3 38 11 5A 00 00 46
15:39:05 MQT: tele/energy1/STATE = {"Time":"2019-08-12T15:39:05","Epoch":1565599145,"Uptime":"21T16:57:18","UptimeSec":1875438,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-1","BSSId":"B0:C7:45:36:85:E0","Channel":7,"RSSI":100,"LinkCount":13,"Downtime":"0T00:00:47"}}
15:39:05 MQT: tele/energy1/SENSOR = {"Time":"2019-08-12T15:39:05","Epoch":1565599145,"ENERGY":{"TotalStartTime":"2019-08-12T00:00:00","Total":11.639,"Yesterday":17.748,"Today":11.614,"Period":3,"Power":1230,"ApparentPower":1267,"ReactivePower":304,"Factor":0.97,"Voltage":229,"Current":5.530}}
15:39:05 DBG: [Factor  0.97    ]
15:39:05 DMP: A0 00 E5 01 00 00 86
15:39:05 DMP: A1 00 05 36 00 00 DC
15:39:06 DMP: A2 04 D1 00 00 00 77
15:39:06 DMP: A3 38 11 5B 00 00 47
15:39:07 DBG: [Voltage 228V    ]
15:39:07 DMP: A0 00 E5 03 00 00 88
15:39:07 DMP: A1 00 05 36 00 00 DC
15:39:08 DMP: A2 04 D4 00 00 00 7A
15:39:08 DMP: A3 38 11 5B 00 00 47
15:39:09 DBG: [Current 5.530A  ]
15:39:09 DMP: A0 00 E5 03 00 00 88
15:39:09 DMP: A1 00 05 38 00 00 DE
15:39:09 WIF: Checking connection...
15:39:09 WIF: Connected
15:39:10 DMP: A2 04 CE 00 00 00 74
15:39:10 DMP: A3 38 11 5C 00 00 48
15:39:11 DBG: [Total   11.639kW]
15:39:11 DMP: A0 00 E5 02 00 00 87
15:39:11 DMP: A1 00 05 38 00 00 DE
15:39:12 DMP: A2 04 D0 00 00 00 76
15:39:12 DMP: A3 38 11 5D 00 00 49
15:39:13 DBG: [Yesterd 17.748kW]
15:39:13 DMP: A0 00 E5 02 00 00 87
15:39:13 DMP: A1 00 05 35 00 00 DB

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
eg: type EnergyReset3 25.7 in console but the value will not change.
or send by MQTT to topic **cmnd/energy1/EnergyReset3**

### EXPECTED BEHAVIOUR
It will save the new value as its designed feature.

wongnam, 2019-08-12:
Sorry, because I was mistaken.
I should not add the decimal point to the value.

closed 2019-08-12 by wongnam

arendst, 2019-08-12:
It's working as designed...

You'll need to enter the requested energy value in Wh so without the decimal dot. The code does not process floating point entry.

NOTE: Just saw your solutinon while typing. Thx.

wongnam, 2019-08-12:
Thanks @arendst I realized my mistake. yes, It is working as designed. :)