arendst/Tasmota #6215: Tuya Dimmer : Slider not working

shantur, 2019-08-12:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
Tuya Dimmer switch web ui dimmer slider doesn't work on latest 6.6.0.4

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Tuya Dimmer
- [x] Tasmota binary firmware version number used : 6.6.0.4 / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

00:05:03 CMD: status 0
00:05:03 SRC: WebConsole from 192.168.10.22
00:05:03 RSL: stat/sonoff/STATUS = {"Status":{"Module":54,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:05:03 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:47","StartupUTC":"2019-08-12T22:59:16","Sleep":50,"CfgHolder":4617,"BootCount":47,"SaveCount":667,"SaveAddress":"F4000"}}
00:05:03 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.4(efb1908-sonoff)","BuildDateTime":"2019-08-12T19:02:17","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:05:03 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SKY91125",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280503000100060000005A00000000000000","00050000"]}}
00:05:03 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8FDAE397","003683A0","22B617CD","01001BC0","00000001"]}}
00:05:03 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6994","IPAddress":"192.168.10.21","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"EC:FA:BC:32:9B:52","Webserver":2,"WifiConfig":2}}
00:05:03 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_329B52","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:05:03 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 12 23:05:03 2019","Local":"Tue Aug 13 00:05:03 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:40","Sunset":"20:10"}}
00:05:03 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-08-13T00:05:03","Epoch":1565651103}}
00:05:03 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-08-13T00:05:03","Epoch":1565651103,"Uptime":"0T00:05:47","UptimeSec":347,"Heap":20,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":605,"MqttCount":0,"POWER":"ON","Dimmer":41,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SKY91125","BSSId":"A0:BD:CD:5E:19:1A","Channel":1,"RSSI":36,"LinkCount":1,"Downtime":"0T00:00:08"}}}
00:05:04 WIF: Checking connection...

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

00:00:00 CFG: Loaded from flash at F7, Count 624
00:00:00 Project sonoff Sonoff Version 6.6.0.4(efb1908-sonoff)-2_4_2
00:00:00 SNS: Hardware Serial
00:00:00 WIF: Connecting to AP1 SKY91125 in mode 11N as sonoff-6994...
00:00:00 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":50}
00:00:00 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":3}
00:00:00 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":42}
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on sonoff-6994 with IP address 192.168.10.21
23:59:26 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":37}
23:59:27 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":1}
23:59:27 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100}
23:59:31 CMD: weblog 4
23:59:31 RSL: stat/sonoff/RESULT = {"WebLog":4}
23:59:31 CFG: Saved to flash at F9, Count 630, Bytes 3584
23:59:32 HTP: Main Menu
23:59:34 TYA: RX Packet: "55aa03070005010100010112"
[... 391 lines cut ...]
00:00:03 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":42}
00:00:03 TYA: Send dim skipped value=23
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

- Open Web UI
- Try to use Tuya Dimmer slider

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Slider works.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

shantur, 2019-08-14:
For anyone who is in same boat should look at this post : https://github.com/arendst/Sonoff-Tasmota/issues/5545#issuecomment-493599301
What i was missing was `SetOption38 3`

shantur, 2019-08-14:
```11:26:50 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":80}
11:26:50 TYA: RX Packet: "55aa03070008060200040000097ea4"
11:26:50 TYA: RX Dim State=126
11:26:50 SRC: Switch
11:26:50 RSL: stat/sonoff/RESULT = {"Dimmer":80}
11:26:50 TYA: Send dim skipped value=80
11:26:56 TYA: Send "55aa00000000ff"
11:26:56 TYA: RX Packet: "55aa030000010104"
11:26:56 TYA: Heartbeat
11:26:56 WIF: Checking connection...
11:26:56 WIF: Connected
11:26:59 TYA: RX Packet: "55aa03070005010100010112"
11:26:59 TYA: RX Device-1 --> MCU State: On Current State:On
11:26:59 TYA: RX Packet: "55aa03070008020200040000001831"
11:26:59 TYA: RX Dim State=24
11:26:59 SRC: Switch
11:26:59 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":24}
11:26:59 TYA: RX Packet: "55aa0307000803020004000000001a"
11:26:59 TYA: RX Dim State=0
11:26:59 TYA: RX Packet: "55aa03070008040200040000002742"
11:26:59 TYA: RX Dim State=39
11:26:59 SRC: Switch
11:26:59 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":39}
11:26:59 TYA: Send dim skipped value=39
11:26:59 TYA: RX Packet: "55aa0307000805020004000000516d"
11:26:59 TYA: RX Dim State=81
11:26:59 SRC: Switch
11:26:59 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":81}
11:26:59 TYA: RX Packet: "55aa03070008060200040000097ea4"
11:26:59 TYA: RX Dim State=126
11:26:59 SRC: Switch
11:26:59 RSL: stat/sonoff/RESULT = {"Dimmer":81}
11:26:59 TYA: Send dim skipped value=81
11:26:59 CFG: Saved to flash at F5, Count 711, Bytes 3584
11:27:07 TYA: Send "55aa00000000ff"
11:27:07 TYA: RX Packet: "55aa030000010104"
11:27:07 TYA: Heartbeat
11:27:08 TYA: RX Packet: "55aa03070005010100010112"
11:27:08 TYA: RX Device-1 --> MCU State: On Current State:On
11:27:08 TYA: RX Packet: "55aa03070008020200040000001831"
11:27:08 TYA: RX Dim State=24
11:27:08 SRC: Switch
11:27:08 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":24}
11:27:08 TYA: RX Packet: "55aa0307000803020004000000011b"
11:27:08 TYA: RX Dim State=1
11:27:08 SRC: Switch
11:27:08 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":1}
11:27:08 TYA: Send dim skipped value=1
11:27:08 TYA: RX Packet: "55aa03070008040200040000002742"
11:27:08 TYA: RX Dim State=39
11:27:08 SRC: Switch
11:27:08 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":39}
11:27:08 TYA: RX Packet: "55aa0307000805020004000000506c"
11:27:08 TYA: RX Dim State=80
11:27:08 SRC: Switch
11:27:08 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":80}
11:27:08 TYA: RX Packet: "55aa03070008060200040000097fa5"
11:27:08 TYA: RX Dim State=127
11:27:08 SRC: Switch
11:27:08 RSL: stat/sonoff/RESULT = {"Dimmer":80}
11:27:08 TYA: Send dim skipped value=80
11:27:08 CFG: Saved to flash at F4, Count 712, Bytes 3584
11:27:16 WIF: Checking connection...
11:27:16 WIF: Connected
11:27:17 CMD: clear
11:27:17 SRC: WebConsole from 192.168.10.22
11:27:17 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
11:27:17 TYA: RX Packet: "55aa03070005010100010112"
11:27:17 TYA: RX Device-1 --> MCU State: On Current State:On
11:27:17 TYA: RX Packet: "55aa03070008020200040000001831"
[... 352 lines cut ...]
11:27:58 TYA: Send dim skipped value=0
11:27:58 TYA: RX Packet: "55aa03070008050200040000002844"
11:27:58 TYA: RX Dim State=40
11:27:58 SRC: Switch
11:27:58 RSL: stat/sonoff/RESULT = {"POWER":"OFF","Dimmer":40}
11:27:58 TYA: RX Packet: "55aa03070008060200040000097ea4"
11:27:58 TYA: RX Dim State=126
11:27:58 SRC: Switch
11:27:58 RSL: stat/sonoff/RESULT = {"Dimmer":40}
11:27:58 TYA: Send dim skipped value=0
11:27:58 CMD: power on
11:27:58 SRC: WebConsole from 192.168.10.22
11:27:58 TYA: Send dim value=40 (id=2)
11:27:58 TYA: Send "55aa0006000802020004000000283d"
11:27:58 TYA: Send "55aa0006000501010001010e"
11:27:58 RSL: stat/sonoff/RESULT = {"POWER":"ON"}
11:27:58 RSL: stat/sonoff/POWER = ON
11:27:58 TYA: RX Packet: "55aa03070008020200040000002841"
11:27:58 TYA: RX Dim State=40
11:27:59 CFG: Saved to flash at F8, Count 724, Bytes 3584
11:28:02 TYA: Send "55aa00000000ff"
11:28:02 TYA: RX Packet: "55aa030000010104"
11:28:02 TYA: Heartbeat```

Last Power on didn't work

shantur, 2019-08-14:
Re-opening issue as its still present as per https://github.com/arendst/Sonoff-Tasmota/issues/6215#issuecomment-521191828

closed 2019-08-14 by arendst

buzurk26, 2020-03-17:
> ### BUG DESCRIPTION
> Tuya Dimmer switch web ui dimmer slider doesn't work on latest 6.6.0.4
> 
> ### REQUESTED INFORMATION
> _Make sure these boxes are checked before submitting your issue. Thank you_
> 
> **FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**
> 
> * [x]  Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
> * [x]  Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
> * [x]  Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
> * [x]  Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
> * [x]  Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
> * [x]  Device used (i.e. Sonoff Basic) : Tuya Dimmer (http://szmakegood.com/ukpro.html)
> * [x]  Tasmota binary firmware version number used : 6.6.0.4 / (pre-compiled or self-compiled ?)
> * [ ]  Development IDE - Compiler / Upload tools used : ____ / ____
> * [x]  Provide the output of command `status 0` :
> 
> ```
> STATUS 0 OUTPUT HERE:
> 
> 
> 00:05:03 CMD: status 0
> 00:05:03 SRC: WebConsole from 192.168.10.22
> 00:05:03 RSL: stat/sonoff/STATUS = {"Status":{"Module":54,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
> 00:05:03 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:47","StartupUTC":"2019-08-12T22:59:16","Sleep":50,"CfgHolder":4617,"BootCount":47,"SaveCount":667,"SaveAddress":"F4000"}}
> 00:05:03 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.4(efb1908-sonoff)","BuildDateTime":"2019-08-12T19:02:17","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
> 00:05:03 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SKY91125",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280503000100060000005A00000000000000","00050000"]}}
> 00:05:03 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8FDAE397","003683A0","22B617CD","01001BC0","00000001"]}}
> 00:05:03 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6994","IPAddress":"192.168.10.21","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"EC:FA:BC:32:9B:52","Webserver":2,"WifiConfig":2}}
> 00:05:03 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_329B52","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
> 00:05:03 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 12 23:05:03 2019","Local":"Tue Aug 13 00:05:03 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:40","Sunset":"20:10"}}
> 00:05:03 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-08-13T00:05:03","Epoch":1565651103}}
> 00:05:03 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-08-13T00:05:03","Epoch":1565651103,"Uptime":"0T00:05:47","UptimeSec":347,"Heap":20,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":605,"MqttCount":0,"POWER":"ON","Dimmer":41,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SKY91125","BSSId":"A0:BD:CD:5E:19:1A","Channel":1,"RSSI":36,"LinkCount":1,"Downtime":"0T00:00:08"}}}
> 00:05:04 WIF: Checking connection...
> ```
> 
> * [x]  Provide the output of console when you experience your issue if apply :
>   _(Please use_ `weblog 4` _for more debug information)_
> 
> ```
> CONSOLE OUTPUT HERE:
> 
> 00:00:00 CFG: Loaded from flash at F7, Count 624
> 00:00:00 Project sonoff Sonoff Version 6.6.0.4(efb1908-sonoff)-2_4_2
> 00:00:00 SNS: Hardware Serial
> 00:00:00 WIF: Connecting to AP1 SKY91125 in mode 11N as sonoff-6994...
> 00:00:00 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":50}
> 00:00:00 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":3}
> 00:00:00 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":42}
> 00:00:08 WIF: Connected
> 00:00:08 HTP: Web server active on sonoff-6994 with IP address 192.168.10.21
> 23:59:26 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":37}
> 23:59:27 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":1}
> 23:59:27 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100}
> 23:59:31 CMD: weblog 4
> 23:59:31 RSL: stat/sonoff/RESULT = {"WebLog":4}
> 23:59:31 CFG: Saved to flash at F9, Count 630, Bytes 3584
> 23:59:32 HTP: Main Menu
> 23:59:34 TYA: RX Packet: "55aa03070005010100010112"
> 23:59:34 TYA: RX Device-1 --> MCU State: On Current State:On
> 23:59:34 TYA: RX Packet: "55aa0307000802020004000000011a"
> 23:59:34 TYA: RX Dim State=1
> 23:59:34 TYA: RX Packet: "55aa0307000803020004000000001a"
> 23:59:34 TYA: RX Dim State=0
> 23:59:34 TYA: RX Packet: "55aa0307000804020004000000031e"
> 23:59:34 TYA: RX Dim State=3
> 23:59:34 SRC: Switch
> 23:59:34 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":1}
> 23:59:34 TYA: Send dim skipped value=1
[... 387 lines cut ...]
[... cut ...]
> _Add any other context about the problem here._
> 
> Tried `SetOption34` with `2` and `3`
> `SetOptionn66` with `1` and `0`
> `SetOption41` with `0` to `8`
> 
> **(Please, remember to close the issue when the problem has been addressed)**

Hi Shantur, Did you ever get this working???? I have the following (assuming its the same) flashed with Tasmota, but cannot get it to function
https://www.alibaba.com/product-detail/SAA-Approval-AU-US-Standard-Wifi_62476608176.html?spm=a2700.icbuShop.41413.21.d60012d9mcebtg&fullFirstScreen=true

shantur, 2020-03-18:
@buzurk26 

I re-implemented Tuya functionality in Tasmota and I have got many MakeGood dimmers. All work perfectly on Tasmota.
You need to follow https://tasmota.github.io/docs/#/TuyaMCU

buzurk26, 2020-03-19:
> @buzurk26
> 
> I re-implemented Tuya functionality in Tasmota and I have got many MakeGood dimmers. All work perfectly on Tasmota.
> You need to follow https://tasmota.github.io/docs/#/TuyaMCU

Thanks for your reply

I did follow that, but could not get the log to list dpids like the guide shows, as below
```
TYA: Heartbeat
TYA: RX Packet: "55aa0107000501010001000f"
TYA: FnId=0 is set for dpId=1
TYA: RX Packet: "55aa01070005020100010010"
TYA: FnId=0 is set for dpId=2
```

Instead, all i get was
```
00:01:04 {"TuyaReceived":{"Data":"55AA030E000010","Cmnd":14}}
00:01:04 TYA: RX unknown command
00:01:04 TYA: Send "55aa00000000ff"
00:01:04 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
00:01:04 TYA: Heartbeat
00:01:06 WIF: Checking connection...
00:01:06 WIF: Connected
00:01:09 {"TuyaReceived":{"Data":"55AA030E000010","Cmnd":14}}
00:01:09 TYA: RX unknown command
00:01:14 {"TuyaReceived":{"Data":"55AA030E000010","Cmnd":14}}
00:01:14 TYA: RX unknown command
00:01:15 TYA: Send "55aa00000000ff"
00:01:15 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
00:01:15 TYA: Heartbeat
```

I did set module type as Tuya (54), I did set pins 1 and 3 as the tx and RX, no luck at all

I've spent countless hours on this, really hoping you can assist.

PR #6227 (2019-08-14): Tuya Dimmer fixes
## Description:

**Related issue (if applicable):** fixes #6215

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [ ] The code change is tested and works on core 2.3.0, 2.4.2 and 2.5.2
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).